espressif/esp-zigbee-sdk #420: esp32-h2 light sleep example and watchdog problem

sbusceti, 2024-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-2306-gdbce23f8a4

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

The ESP32-H2 starts without watchdog errors.

### What is the actual behavior?

The ESP32-H2 starts with following errors

`E (8430) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (8430) task_wdt:  - IDLE (CPU 0)
E (8430) task_wdt: Tasks currently running:
E (8430) task_wdt: CPU 0: Zigbee_main
E (8430) task_wdt: Print CPU 0 (current core) backtrace`

### Steps to reproduce.

When esp32-h2 starts with light-sleep example flashed, if you hold down boot button during esp32.h2 boot phase(or any other pin associated with a wakeup interrupts) the esp32-h2 shows a watchdog error.

1. build and flash ligh-sleep example
2. open monitor tab
3. hold down boot button
4. see watchdog error in monitor tab

xieqinan, 2024-09-06:
@sbusceti,

The watchdog issue is triggered by a pending state generated during the boot button [initializing process](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/main/esp_zb_sleepy_end_device.c#L108) when it is an abnormal state(hold down). I believe this is not a Zigbee-related issue and will likely get a quicker resolution if it is reported under the [esp-idf issue tracker](https://github.com/espressif/esp-idf/issues).

sbusceti, 2024-09-06:
@xieqinan 
I took the example code here, in the esp-zigbee-sdk.
The issue is not related to boot button only, same happens if you bind another pin to wakeup process.
I think the problem is in the task of switch_driver.c. How can I solve it?
My goal is to create a battery powered binary sensor that stays in light sleep, wakes up upon a pin interrupt, send pin state to coordinator and go back sleep. 
But with this issue, if pin is low during boot, the code will never executed.

xieqinan, 2024-10-28:
@sbusceti ,

The following code changes the GPIO interrupt type from `input low level trigger` to `falling edge`. Please copy it to the `switch_driver.c` and test again.

```
static bool switch_driver_gpio_init(switch_func_pair_t *button_func_pair, uint8_t button_num)
{
    gpio_config_t io_conf = {};
    switch_func_pair = button_func_pair;
    switch_num = button_num;
    uint64_t pin_bit_mask = 0;

    /* set up button func pair pin mask */
    for (int i = 0; i < button_num; ++i) {
        pin_bit_mask |= (1ULL << (button_func_pair + i)->pin);
    }
    /* interrupt of falling edge */
    io_conf.intr_type = GPIO_INTR_NEGEDGE;
    io_conf.pin_bit_mask = pin_bit_mask;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 1;
    /* configure GPIO with the given settings */
    gpio_config(&io_conf);
    /* create a queue to handle gpio event from isr */
    gpio_evt_queue = xQueueCreate(10, sizeof(switch_func_pair_t));
    if ( gpio_evt_queue == 0) {
        ESP_LOGE(TAG, "Queue was not created and must not be used");
        return false;
    }
    /* start gpio task */
    xTaskCreate(switch_driver_button_detected, "button_detected", 4096, NULL, 10, NULL);
    /* install gpio isr service */
    gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
    for (int i = 0; i < button_num; ++i) {
        gpio_isr_handler_add((button_func_pair + i)->pin, gpio_isr_handler, (void *) (button_func_pair + i));
    }
    return true;
}
```

closed 2024-12-20 by chshu