espressif/esp-zigbee-sdk #422: GPIO Wakeup

sbusceti, 2024-09-08:
### Question

Good morning everyone,
I have a doubt regarding the GPIO wakeup mode combined with Zigbee light sleep using the command esp_zb_sleep_now();.
Is it possible to wake up the ESP32-H2 with GPIO (without using EXT1)?
I have done some tests, but I couldn't get it to work, while I have no issues with EXT1.
Reading the documentation, I found this:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32h2/api-reference/system/sleep_modes.html#gpio-wakeup
My goal is to wake up the ESP32-H2 when one of N pins changes state and then invert the condition for wakeup.
For example:

- pin12 and pin13 are configured to wake up on a low level.
- pin12 goes low, pin13 remains high, so the ESP wakes up and inverts the wakeup condition only for pin12.

So the new configuration for the next wakeup will be:

- pin12 wakes up with HIGH, pin13 wakes up with LOW, and so on...

Is this possible?
Thanks!

sbusceti, 2024-09-08:
Another problem I'm encountering is putting the device to sleep and waking it up on state changes, rather than on a specific level.
For example:
Wake up on a low pin state, go back to sleep, and wake up again when the state goes high.
Is this possible?

lpy4105, 2024-09-20:
Hi @sbusceti 

It sounds like a question that is not related to the zigbee stack. Could please raise a question to [IDF project](https://github.com/espressif/esp-idf/issues)?

Personally, it seems to be feasible, you could find some examples for light sleep/wakeup here:https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep

closed 2024-12-20 by chshu