espressif/esp-zigbee-sdk #423: esp_zb_zcl_set_attribute_val aborts on large strings over ~100 characters. (ESP_ZB_ZCL_ATTR_TYPE_LONG_OCTET_STRING)

remenyo, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I want to set large string to a large octet string type attribut in my custom cluster.

### What is the actual behavior?

Strings over ~100 triggers abort inside the library.
Sometimes, it fails event at 90-93 characters.

### Steps to reproduce.

Hello, I want to send binary data to my coordinator from my end device. I have a manufacturer specific cluster, that has a Long octet string attribute, which if too long, esp_zb_zcl_set_attribute_val aborts. On smaller strings  (around 90) it works ok.

```c++
std::string new_report = "...data...";
size_t datasize = new_report.length() + 1 + 2; // len + nul terminator + array length.
LOGD("Allocating %u bytes. (%u free)", datasize, heap_caps_get_free_size(MALLOC_CAP_DEFAULT));
void *dataholder = malloc(datasize);
if (dataholder == NULL)
{
    LOGE("dataholder allocation failed.");
}
else
{
    reinterpret_cast<uint16_t *>(dataholder)[0] = new_report.length();
    LOGD("memcpy");
    strncpy(static_cast<char *>(dataholder) + 2, new_report.c_str(), datasize);
    LOGD("memcpy ok");
    if (esp_zb_lock_acquire(pdMS_TO_TICKS(10000)))
    {
        LOGD("set attr val (%u)", (*static_cast<uint16_t *>(dataholder)));
        LOGD("dataholder at [datasize-1] is: %d", static_cast<int>(*static_cast<uint8_t *>(static_cast<unsigned char *>(dataholder) + datasize - 1)));
        esp_zb_zcl_set_attribute_val(1, ZIGBEE::MANUFACTURER_CUSTOM_CLUSTER_ID, esp_zb_zcl_cluster_role_t::ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ZIGBEE::SERIAL_OUTPUT_ATTR_ID, dataholder, false);
        esp_zb_lock_release();
    }
    else
    {
        LOGE("esp_zb_lock_acquire fail");
    }

    if (esp_zb_lock_acquire(pdMS_TO_TICKS(10000)))
    {
        esp_zb_zcl_report_attr_cmd_t report_attr_cmd2;
        report_attr_cmd2.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;
        report_attr_cmd2.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_attr_cmd2.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
        report_attr_cmd2.zcl_basic_cmd.src_endpoint = 1;
        report_attr_cmd2.attributeID = ZIGBEE::SERIAL_OUTPUT_ATTR_ID;
        report_attr_cmd2.clusterID = ZIGBEE::MANUFACTURER_CUSTOM_CLUSTER_ID;
        esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd2);
        esp_zb_lock_release();
    }
    else
    {
        LOGE("esp_zb_lock_acquire fail");
    }
    free(dataholder);
}
```

```txt
D state_sender_task:    Allocating 713 bytes. (22512 free)
D state_sender_task:    memcpy
D state_sender_task:    memcpy ok
D state_sender_task:    set attr val (710)
D state_sender_task:    dataholder at [datasize-1] is: 0
ZBOSS: common/zb_debug.c:64   Assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:325
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:325

abort() was called at PC 0x4203c9d5 on core 0
```

Working at smaller length:
```txt
D state_sender_task:    Allocating 93 bytes. (22552 free)
D state_sender_task:    strncpy
D state_sender_task:    strncpy finished.
D state_sender_task:    set attr val (len = 90)
D state_sender_task:    dataholder at [datasize-1] is: 0
```

not working at smaller length:
```txt
ZBOSS: common/zb_debug.c:64   Assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:325
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:325
```

If I set really small size, like 6 characters, it sends without fragmenting, but for example at 93 bytes, it fragments into 72 + the rest = 2 framents. My coordinator (0x0000 support sizes up to 128 bytes, according to its description that the esp requests before sending.)

Thank you for reading this question.

### More Information.

Note, I do not reach report when setting large string, the esp_zb_zcl_set_attribute_val aborts. That is another thing, it reports to coordinator without bind. (Or it is acceptable in this case?)

The logs show that the format should be ok, but feel free to correct me.

There are a lot of issues around setting large strings. Do you know a bulletproof way to achieve kB level storage for custom cluster custom attributes, that are sendable from an end device to a coordinator anytime?

xieqinan, 2024-09-11:
@remenyo ,

Could you please add the code showing how the `ZIGBEE::MANUFACTURER_CUSTOM_CLUSTER_ID` is created? If the length of `ZIGBEE::SERIAL_OUTPUT_ATTR_ID` is registered as less than 100 during `esp_zb_device_register()`, it will trigger a memory leak issue.

remenyo, 2024-09-11:
@xieqinan ,

I hope this is what you requested. I do not set size anywhere so this may very could be the reason for my problem. Where should I set it?
I tried to filter the not needed parts from the following code, but I am also able to share the full file with little edits if needed.

```cpp
namespace ZIGBEE
{
    const uint16_t MANUFACTURER_CUSTOM_CLUSTER_ID = 0xff01;
    const uint16_t SERIAL_OUTPUT_ATTR_ID = 0x0201;
}

namespace ZIGBEE
{
    namespace ATTRIBUTE_STORE
    {

        //... other variables for other attributes

        uint8_t serialOutput[4096];
    }

    static void esp_zb_task(void *pvParameters)
    {
        ESP_ERROR_CHECK(nvs_flash_init());

        esp_zb_init(&zb_nwk_cfg); // it is a native radio router with zgp functionality enabled (0xFFFFFF, 0xFFFFFF)

        esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();

        // ...

        // Custom cluster creation

        esp_zb_attribute_list_t *custom_cluster_attr_list = esp_zb_zcl_attr_list_create(MANUFACTURER_CUSTOM_CLUSTER_ID);

        // Many esp_zb_custom_cluster_add_custom_attr calls to populate MANUFACTURER_CUSTOM_CLUSTER

        // Serial output
        esp_zb_custom_cluster_add_custom_attr(custom_cluster_attr_list, SERIAL_OUTPUT_ATTR_ID, esp_zb_zcl_attr_type_t::ESP_ZB_ZCL_ATTR_TYPE_LONG_OCTET_STRING, esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &ATTRIBUTE_STORE::serialOutput);

        ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_cluster_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
        // ADD clusters end
        
         // debug: list clusters and attributes
        for (esp_zb_cluster_list_t *c = cluster_list; c != NULL; c = c->next)
        {
            for (esp_zb_attribute_list_t *a = c->cluster.attr_list; a != NULL; a = a->next)
            {
                if (a->attribute.access != 0 && a->attribute.id != 0xfffd)
                {
                    LOGD("CLUSTER_ID 0x%04x ATTRIBUTE_ID 0x%04x TYPE 0x%02x ACCESS 0x%02x", c->cluster.cluster_id, a->attribute.id, a->attribute.type, a->attribute.access);
                }
            }
        }

        esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
        esp_zb_endpoint_config_t endpoint_config = {
            .endpoint = 1,
            .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
            .app_device_id = ESP_ZB_HA_THERMOSTAT_DEVICE_ID,
            .app_device_version = 1,
        };
        esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
        esp_zb_device_register(ep_list);
    }
}
```

I have quite a few attributes:
```txt
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0000 TYPE 0x29 ACCESS 0x05
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0011 TYPE 0x29 ACCESS 0x13
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0012 TYPE 0x29 ACCESS 0x13
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x001b TYPE 0x30 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x001c TYPE 0x30 ACCESS 0x13
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0010 TYPE 0x28 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0013 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0014 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0015 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0016 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0017 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x0018 TYPE 0x29 ACCESS 0x03
D: CLUSTER_ID 0x0201 ATTRIBUTE_ID 0x001e TYPE 0x30 ACCESS 0x01
D: CLUSTER_ID 0x0204 ATTRIBUTE_ID 0x0000 TYPE 0x30 ACCESS 0x03
D: CLUSTER_ID 0x0204 ATTRIBUTE_ID 0x0001 TYPE 0x30 ACCESS 0x03
D: CLUSTER_ID 0x0b05 ATTRIBUTE_ID 0x0000 TYPE 0x21 ACCESS 0x01
D: CLUSTER_ID 0x0b05 ATTRIBUTE_ID 0x0001 TYPE 0x21 ACCESS 0x01
D: CLUSTER_ID 0x0000 ATTRIBUTE_ID 0x0000 TYPE 0x20 ACCESS 0x01
D: CLUSTER_ID 0x0000 ATTRIBUTE_ID 0x0007 TYPE 0x30 ACCESS 0x01
D: CLUSTER_ID 0x0000 ATTRIBUTE_ID 0x0004 TYPE 0x42 ACCESS 0x01
D: CLUSTER_ID 0x0000 ATTRIBUTE_ID 0x0005 TYPE 0x42 ACCESS 0x01
D: CLUSTER_ID 0x0202 ATTRIBUTE_ID 0x0000 TYPE 0x30 ACCESS 0x03
D: CLUSTER_ID 0x0202 ATTRIBUTE_ID 0x0001 TYPE 0x30 ACCESS 0x03
D: CLUSTER_ID 0x0406 ATTRIBUTE_ID 0x0000 TYPE 0x18 ACCESS 0x05
D: CLUSTER_ID 0x0406 ATTRIBUTE_ID 0x0001 TYPE 0x30 ACCESS 0x01
D: CLUSTER_ID 0x0406 ATTRIBUTE_ID 0x0002 TYPE 0x18 ACCESS 0x01
D: CLUSTER_ID 0x0003 ATTRIBUTE_ID 0x0000 TYPE 0x21 ACCESS 0x03
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0001 TYPE 0x29 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0002 TYPE 0x30 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0003 TYPE 0x30 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0004 TYPE 0x30 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0005 TYPE 0x18 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0010 TYPE 0x49 ACCESS 0x03
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0011 TYPE 0x30 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0012 TYPE 0x29 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0020 TYPE 0x48 ACCESS 0x03
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0200 TYPE 0x43 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0201 TYPE 0x43 ACCESS 0x07
D: CLUSTER_ID 0xff01 ATTRIBUTE_ID 0x0fac TYPE 0x00 ACCESS 0x03
```

remenyo, 2024-09-16:
I'm also struggling with setting a plain char string. (43 + 1 characters max)

```cpp
namespace ZIGBEE
{
    const uint16_t MANUFACTURER_CUSTOM_CLUSTER_ID = 0xff01;
    const uint16_t STATUS_ATTR_ID = 0x0201;

    namespace ATTRIBUTE_STORE
    {

        //... other variables for other attributes

        uint8_t serialOutput[255];
    }

    static void esp_zb_task(void *pvParameters)
    {
        ESP_ERROR_CHECK(nvs_flash_init());

        esp_zb_init(&zb_nwk_cfg); // it is a native radio router with zgp functionality enabled (0xFFFFFF, 0xFFFFFF)

        esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();

        // ...

        // Custom cluster creation

        esp_zb_attribute_list_t *custom_cluster_attr_list = esp_zb_zcl_attr_list_create(MANUFACTURER_CUSTOM_CLUSTER_ID);

        // Many esp_zb_custom_cluster_add_custom_attr calls to populate MANUFACTURER_CUSTOM_CLUSTER

        // output
        esp_zb_custom_cluster_add_custom_attr(custom_cluster_attr_list, STATUS_ATTR_ID, esp_zb_zcl_attr_type_t::ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &ATTRIBUTE_STORE::serialOutput);

        ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_cluster_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
        // ADD clusters end

        esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
        esp_zb_endpoint_config_t endpoint_config = {
            .endpoint = 1,
            .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
            .app_device_id = ESP_ZB_HA_THERMOSTAT_DEVICE_ID,
            .app_device_version = 1,
        };
        esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
        esp_zb_device_register(ep_list);
    }
}
```

```cpp
namespace SENDER
{
    void debugPrintBuffer(const uint8_t *buffer, size_t length)
    {
        printf("Buffer contents:\n");
        for (size_t i = 0; i < length; ++i)
        {
            printf("%02X ", buffer[i]); // Print each byte as a two-digit hexadecimal number
            if ((i + 1) % 16 == 0)
            {
                printf("\n"); // New line every 16 bytes for readability
            }
        }
        printf("\n");
    }

    namespace SENDER_INTERNAL
    {
[... 175 lines cut ...]
4085d9e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4085da00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4085da20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4085da40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4085da60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4085da80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

---

Interesting addition:

`LOGD("%.0fkB free, largest free block: %.0fkB", static_cast<float>(heap_caps_get_free_size(MALLOC_CAP_DEFAULT)) / 1024.0, static_cast<float>(heap_caps_get_largest_free_block(MALLOC_CAP_DEFAULT)) / 1024.0);` in a different task causes.

```txt
Stack memory:
4086b3b0: 0x420d2000 0x00000000 0x00002710 0x40880000 0x40021046 0x4081be00 0x4086b3f4 0x400227f4
0x40021046: tlsf_walk_pool in ROM
0x400227f4: multi_heap_get_info_impl in ROM
```

but only if i set the strings. if I do not set the strings, it does work fine.
also, disabling this memory checking line, allows the string setting and reporting to work...

@xieqinan ?

lpy4105, 2024-09-20:
Hi @remenyo 

As @xieqinan has pointed, it might be caused by a known issue. Please try the following code when you are creating the string attribute:

```c
ATTRIBUTE_STORE::serialOutput[0] = sizeof(ATTRIBUTE_STORE::serialOutput);
esp_zb_custom_cluster_add_custom_attr(
    custom_cluster_attr_list, 
    STATUS_ATTR_ID, 
    esp_zb_zcl_attr_type_t::ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING, 
    esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | 
    esp_zb_zcl_attr_access_t::ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, 
    &ATTRIBUTE_STORE::serialOutput);
```

The stack will decide the actual space used to store the attribute by its initial value when creation.

remenyo, 2024-09-20:
Thank you for your response, I did not understood what [xieqinan](https://github.com/xieqinan) pointed to, but now I understand.

It indeed works if I set it initialy to the size in the first byte. I will test uint16 (longer) byte sizes and other types later with this method. (Aside from this, for some reason it hangs if i set over ~100 characters inside the zigbee loop.)

diazmanuel, 2024-09-30:
Hello @lpy4105 , we were having the same problem and we solved it by initializing the variable with a string of the maximum length that we needed, I understand that it is because it needs to be initialized as you suggest, but this should be fixed. One of the problems that we mention in the issue #319 refers to this problem.

closed 2025-05-30 by xieqinan