espressif/esp-zigbee-sdk #424: Custom attribute of type ESP_ZB_ZCL_ATTR_TYPE_LONG_CHAR_STRING results in "Failed to get attribute size" on startup

MartinJato, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### esp-zigbee-lib version.

1.0.9

### esp-zboss-lib version.

1.0.9

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

To set the value of a custom attribute of type ESP_ZB_ZCL_ATTR_TYPE_LONG_CHAR_STRING to "\x0c\x00""hello world" and not receive E (474) ESP_ZIGBEE_ATTRIBUTE: esp_zb_custom_cluster_add_custom_attr(3325): Failed to get attribute size.

__

### What is the actual behavior?

VS shows error "E (474) ESP_ZIGBEE_ATTRIBUTE: esp_zb_custom_cluster_add_custom_attr(3325): Failed to get attribute size" on the ESP-IDF monitor.

### Steps to reproduce.

Hello,

I want to create a simple custom cluster following this guideline [link](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html#custom-cluster-creation). For long string attribute i modified the given string attribute to fit the format of zigbee long string types. I tried out several ESP_ZB_ZCL_ATTR_TYPE_LONG_CHAR_STRING  attribute values:

```

uint8_t custom_attr1_value[] = "\x0b\x00""hello world"; //this one..
uint8_t custom_attr2_value[] = "\x0b\x00hello world"; //..or this one is expected to work
uint8_t custom_attr3_value[] = "\x00\x0b""hello world";
uint8_t custom_attr4_value[] = "\x0b\x0b""hello world";
uint8_t custom_attr5_value[] = "\x00\x00""hello world";
uint8_t custom_attr6_value[] = "  hello world";

```

None of these works, as I receive all the times the "failed to get attribute size" error in the IDF monitor.

Thanks for helping and best regards

xieqinan, 2024-09-11:
Hi,

Could you please use the `ZB_ZCL_ATTR_TYPE_LONG_OCTET_STRING` instead ?

chshu, 2024-09-29:
Please reopen if any follow up quetions.

closed 2024-09-29 by chshu