espressif/esp-zigbee-sdk #426: End device binding table

akira215, 2024-09-11:
### Question

I'm working on zigbee end devices and I cannot figure out how to access the device binding table:
I'm able to send `esp_zb_zdo_binding_table_req` command that return table but I understood that it is the reading of remote device (coordinator), am I correct ? Moreover, the returned structures linked by the next pointer are not located at the same memory address than the one returned by continuously requesting the next record using index.

I was guessing that the end device shall have it's own binding table, ideally using NVS. Is it the case in the sdk implementation ? If yes how could I retrieve this table ?

Additionally, I try to get the signal from the stack when the coordinator send a binding request to the end device (on end device side). I saw the request using a sniffer, I see some traffic on the `esp_zb_aps_data_indication_handler`, but the `esp_zb_app_signal_handler` don't trigger anything (i was guessing ESP_ZB_BDB_SIGNAL_FINDING_AND_BINDING_INITIATOR_FINISHED or ESP_ZB_BDB_SIGNAL_FINDING_AND_BINDING_TARGET_FINISHED). How could i set up a call back to intercept the binding request from coordinator on end device side ?

Thanks ahead

lpy4105, 2024-09-19:
Hi @akira215 

> I'm able to send esp_zb_zdo_binding_table_req command that return table but I understood that it is the reading of remote device (coordinator), am I correct ?

It's partially true. ZDO commands are sent through the network, you can send the request to the local device if you populate the correct address.

> I was guessing that the end device shall have it's own binding table, ideally using NVS. Is it the case in the sdk implementation ? If yes how could I retrieve this table ?

The SDK implements binding table for end device and stores it in NVS. You can send `esp_zb_zdo_binding_table_req` to local device to retrieve the binding table.

> Additionally, I try to get the signal from the stack when the coordinator send a binding request to the end device (on end device side).

Bindings are processed by the APS layer of the stack, users are not expected to deal with it themselves. Any special use cases?

akira215, 2024-09-30:
Hi sorry for the late reply, was abroad. 

The thing is that the `esp_zb_zdo_binding_table_req` return every bindings it cannot be filtered just for the end device itself. 

I understood that bindings are low level dealed by APS, my app would work without checking the device table, but i was wondering if i can check the table stored in the device, to be able to clean it if required and deal with custom bindings attributes.

xieqinan, 2024-10-17:
@akira215 ,

> but i was wondering if i can check the table stored in the device, to be able to clean it if required and deal with custom bindings attributes.

The `esp_zb_zdo_binding_table_req()` provides the capability to inquire about the binding table of both remote and local devices. If you need to clean up unexpected bindings, you can use the `esp_zb_zdo_device_unbind_req()` function to achieve this.

jkano, 2024-10-29:
Hi @xieqinan I have a use case for this but I'm not able to get a workaround. Whenever a binding request from the coordinator is received, I want to trigger an internal event to get the bind information, send it to another ESP32 which will send the info via MQTT to our cloud. This way, the information about bindings in our cloud are updated and we dont have to create like a periodic task to ask the binding table and update it.

But now, whenever a bind request is sent by the coordinator, the device doesn't call `esp_zb_app_signal_handler()` so I don't know when the binding table is changed.

Is there any way to get at least a callback when the binding table on the end device changed?

akira215, 2024-10-30:
Hi jkano,

I will let @xieqinan  answering, but what I could say is that during my investigations, I note that some signals are sent on the APS layer. You can try to catch them via `esp_zb_aps_data_indication_handler`. I don't know if it does you any good, but I made a basic c++ implementation here :
[https://github.com/akira215/esp-ash-components/blob/main/zigbee/zbApsData.cpp](https://github.com/akira215/esp-ash-components/blob/main/zigbee/zbApsData.cpp)

xieqinan, 2024-10-30:
Hi @jkano ,

>  This way, the information about bindings in our cloud are updated and we dont have to create like a periodic task to ask the binding table and update it.

I understand your point. Currently, correct bind requests are automatically added to the bind table in the APS layer without any callback. However, the solution provided by @akira215 can be used as a workaround. Please note to return `false` in `esp_zb_aps_data_indication_handler()` so that the stack does not skip the bind request command.

closed 2024-12-20 by chshu