espressif/esp-zigbee-sdk #427: Wifi + SSL + Zigbee Crash

kioelka, 2024-09-12:
### Question

I'm trying to add zigbee function to project.
It works only if wifi isn't connected to AP, else app crashes in ssl task when it starts to connect to server or when it gets white IP address  from api.ipify.org

```ansi
I (5601) zigbee.c:0219 : Network(0x239d) closed, devices joining not allowed.
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 6 comm status 0
I (5611) zigbee.c:0176 : Device started up in non factory-reset mode
I (5621) zigbee.c:0181 : Device rebooted
I (6631) contikiIntegration.c:0393 : INFO: msID: 1, cmd: 05
I (7131) flash.c:0234 : WARNING: getKoefs start!
I (7131) flash.c:0504 : Blob "koefs" length unknown, detect it
I (7131) flash.c:0251 : pKoefs.crc: 0x7c9ec761, getKoefsCRC_CH32:  0x7c9ec761
I (7141) flash.c:0253 : Koefs CRC=0x7c9ec761 OK
I (7141) esp-efuse.c:0059 : xtest efuse: bitsWritten: 1
I (7151) settings.c:0058 : Got UTC time 0,convert to 0 and save RAM
I (7161) settings.c:0058 : Got UTC time 0,convert to 0 and save RAM
I (7171) settings.c:0047 : Got Local time 0,convert to 0 and save KOEFS
I (7181) settings.c:0047 : Got Local time 0,convert to 0 and save KOEFS
I (7181) settings.c:0159 : setControlTypeCheck 2
I (7191) settings.c:0174 : reset PRH
I (7191) settings.c:0192 : coolMode 128 controlType 2
I (7201) settings.c:0195 : return fail
I (7211) esp-efuse.c:0059 : xtest efuse: bitsWritten: 1
I (7211) flash.c:0286 : xtest efuse 0
I (7221) flash.c:0519 : ERROR: Get from nvs error: ESP_ERR_NVS_NOT_FOUND
I (7231) flash.c:0523 : ERROR: Get from nvs backup error: ESP_ERR_NVS_NOT_FOUND
I (8151) wifi:ap channel adjust o:1,1 n:2,1
I (8151) wifi:new:<2,0>, old:<1,1>, ap:<2,1>, sta:<2,0>, prof:1
I (8161) wifi:(connect)dot11_authmode:0x3, pairwise_cipher:0x3, group_cipher:0x3
I (8481) wifi:state: init -> auth (b0)
I (8491) wifi:state: auth -> assoc (0)
I (8491) wifi:state: assoc -> run (10)
I (8491) wifi:(trc)phytype:CBW20-SGI, snr:49, maxRate:144, highestRateIdx:0
I (8491) wifi:(trc)rate(S-MCS7, schedIdx:0), ampdu(rate:S-MCS7, schedIdx(0, stop:8)), snr:49, ampduState:wait operational
I (8501) wifi:ifidx:0, rssi:-47, nf:-96, phytype(0x3, CBW20-SGI), phymode(0x3, 11bgn), max_rate:1440, he:0
I (8511) wifi:max ampdu length exponent:3(65535 bytes), mmss:5(4 us)
W (8591) wifi:[ADDBA]rx delba, code:39, delete tid:0
W (8591) wifi:[ADDBA]rx delba, code:39, delete tid:5
W (8591) wifi:[ADDBA]rx delba, code:39, delete tid:6
W (8591) wifi:[ADDBA]rx delba, code:39, delete tid:7
W (8601) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100864d, TALO:0x720e6ed6, (ssn:3, win:64, cur_ssn:3), CONF:0xc0000005
I (8621) wifi:(extcap)mbssid:0, enhanced_mbssid_advertise:0, complete_nontxbssid_profiles:0, twt_responder: 0
I (8621) wifi:connected with Direct, aid = 2, channel 2, BW20, bssid = d6:6e:0e:72:4d:86
I (8621) wifi:cipher(pairwise:0x3, group:0x3), pmf:0, security:WPA2-PSK, phy:11bgn, rssi:-47
I (8631) wifi:pm start, type: 2, itwt_start:0

I (8641) wifi:pm start, type:2, aid:0x2, trans-BSSID:d6:6e:0e:72:4d:86, BSSID[5]:0x86, mbssid(max-indicator:0, index:0), he:0
I (8651) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (8661) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (8661) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (8681) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (8691) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (8701) wifi.c:0355 : Connected to Direct
I (8711) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (8711) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (8721) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (8731) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (8741) wifi:<ba-add>idx:1, ifx:0, tid:6, TAHI:0x100864d, TALO:0x720e6ed6, (ssn:0, win:64, cur_ssn:0), CONF:0xc0006005
I (9131) functions.c:0674 : current settings time 1/1/0 0:0:0
I (9131) functions.c:0678 : current timetable time 1/1/0 0:0:0
I (9701) esp_netif_handlers: sta ip: 192.168.3.98, mask: 255.255.255.0, gw: 192.168.3.1
I (9701) wifi.c:1079 : STOP STA CONNECT TIMER
I (9701) wifi.c:1085 : STOP STA CONNECT TIMER: DONE
I (9711) wifi.c:0397 : IP_EVENT: IP_EVENT_STA_GOT_IP
I (9711) wifi.c:0409 : IP_EVENT: got ip: 192.168.3.98
I (9721) flash.c:0811 : Set nvs wifiResetCount done: 0
I (9731) flash.c:0831 : Set nvs backup wifiResetCount done: 0
I (9731) wifi.c:1004 : START MULTICAST TIMER
[... 74 lines cut ...]
4082b4f0: 0x00000d03 0x00000000 0x4082b4a0 0x0000000b 0x00000000 0x00000012 0x00000000 0x00000000
4082b510: 0x00000000 0x00003771 0x00000000 0x40828ec4 0x40828f2c 0x40828f94 0x00000000 0x00000000
4082b530: 0x00000001 0x00000000 0x00000000 0x00000000 0x421efb14 0x00000000 0x00000000 0x00000000
0x421efb14: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4082b550: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082b570: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082b590: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082b5b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082b5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082b5f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40000000
0x40000000: _start in ROM

4082b610: 0x0000005c 0x4082b614 0x4082b614 0x4082b614 0x4082b614 0x00000000 0x4082b62c 0xffffffff
4082b630: 0x4082b62c 0x4082b62c 0x00000000 0x4082b640 0xffffffff 0x4082b640 0x4082b640 0x00000001
4082b650: 0x00000001 0x00000000 0x0000ffff 0x00000000 0x4082f038 0x40880003 0xb33fffff 0x00000000
4082b670: 0x00000018 0x42269dd4 0x4082b690 0x408348e4 0x00000000 0x4082b8f0 0x40834b48 0x0000025c
4082b690: 0x4082b6ec 0x4082b73c 0x4082b8ec 0x4082b72c 0x00000000 0x4082b6a8 0xffffffff 0x4082b6a8
4082b6b0: 0x4082b6a8 0x00000001 0x4082b6bc 0xffffffff 0x408348fc 0x408348fc 0x00000000 0x00000020

ELF file SHA256: 9d21eeba42991caf

Rebooting in  3 seconds...
Rebooting...
```

OOHehir, 2024-09-14:
My understanding is that Zigbee & WiFi on current devices, simultaneously, is not supported. I investigated briefly on a ESP32C6& didn't have good results. [This](https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_gateway/README.md) is probably your best solution.

lpy4105, 2024-09-20:
Hi @kioelka 

It seems that the crash log is not related to co-existence. Could you please share more details? E.g. the steps to reproduce the issue, app code snippet?

BTW, did the WiFi part itself could work correctly (without Zigbee functional enabled)?

kioelka, 2024-09-20:
> Hi @kioelka
> 
> It seems that the crash log is not related to co-existence. Could you please share more details? E.g. the steps to reproduce the issue, app code snippet?
> 
> BTW, did the WiFi part itself could work correctly (without Zigbee functional enabled)?
on second question answer - "yes" wifi works well without zigbee. also, wifi + zigbee works together fine if zigbee start after ssl handshake done, but, if internet fall down for a time, when wifi reconnects to AP and SSL session restarts, app crashes cause zigbee is active this time.
```c
/**
 * Init ssl connection
 */
void sslClientTask(void *pvParameters)
{

    if ((devc.isUpdating) && (devc.currentCumulativeOTAStatus.source != ZNOCloudUpdate))
    {
        vTaskDelete(NULL);
    }

    DEBUG_MSG(BDL, (COL_CLOUD "certsOk: %d", devc.certsOk));

    static int tls_error_count;
    static int reconnect_tries;
    static int success_connects;

    char buf[CLOUD_READBUF];
    int ret, flags, err;
    vTaskDelay(pdMS_TO_TICKS(50));
    DEBUG_MSG(BDL, (COL_CLOUD "mbedTLS thread start OK"));

    devc.cloudPin = initState;

    wifiInit((wifi_mode_t)WIFI_MODE_STA); //only debug zigbee, !!TODO RETurn here NULL!!!!

#if defined(CONFIG_IDF_TARGET_ESP32C6)
    DEBUG_MSG(BDL, (COL_EXCHANGE "initZigbee()" ));
    xTaskCreate(esp_zb_task, "Zigbee_main", ZIBGEE_TASK_STACK, NULL, ZIGBEE_TASK_PRIORITY, &devc.zbTaskHandle);
#endif

    time(&now);
    localtime_r(&now, &timeinfo);

    if (!devc.certsOk)
    {
        DEBUG_MSG(BDL, (COL_ERROR COL_CLOUD "Certs corrupted, delete task"));
        vTaskDelete(NULL);
    }

    mbedtls_entropy_context entropy;
    mbedtls_ctr_drbg_context ctr_drbg;
    // mbedtls_ssl_context ssl;
    mbedtls_x509_crt cacert;
    mbedtls_x509_crt clicert;
    mbedtls_pk_context pkey;
    mbedtls_ssl_config conf;
    mbedtls_net_context server_fd;

    devc.cloudCon.sslc = &ssl;

    xEventGroupSetBits(devc.cloudEventGroup, CLOUD_FAIL_BIT);
    xEventGroupWaitBits(devc.wifiEventGroup, WIFI_CONNECTED_BIT,
                        false, true, portMAX_DELAY);
    mbedtls_ssl_init(&ssl);
    mbedtls_ctr_drbg_init(&ctr_drbg);

    mbedtls_ssl_config_init(&conf);
    ...
   
   ...
        if (!waitTimes)
        { // try to connect ot cloud by ip
            waitTimes = CONNECT_WAIT_TIMES;
            while (waitTimes)
            {
                DEBUG_MSG(BDL, (COL_CLOUD "Connecting to %s:%s...", devc.cloudIp, devc.cloudPort)); // TLS_PORT));
                if ((ret = mbedtls_net_connect(&server_fd, devc.cloudIp,
                                               devc.cloudPort, MBEDTLS_NET_PROTO_TCP)) != 0)
                {
                    DEBUG_MSG(IDL, (COL_ERROR COL_CLOUD "mbedtls_net_connect returned -%x", -ret));
                    waitTimes--;
                }
                else
                {
                    break;
                }
                vTaskDelay(pdMS_TO_TICKS(5 * 1000)); // 15 sec here
            }
            if (!waitTimes)
                goto exit;
        }

        devc.connectionState = serverConnection;
        DEBUG_MSG(BDL, (COL_CLOUD "Connected"));

        getExtIP();

```

```c
void getExtIP()
{

#define SERVER_NAME "api.ipify.org"
#define SERVER_PORT "80"
#define PATH "/?format=json"
    const char *getRequest = "GET " PATH " HTTP/1.1\r\nHost: " SERVER_NAME "\r\nAccept: */*\r\n\r\n";
    devc.extIPaddress = 0;
    if (getOtaStage())
        return;
    int sock = openSocket(SERVER_NAME ":" SERVER_PORT);
    char response[256];
    if (sock != ESP_FAIL)
    {
        send(sock, getRequest, strlen(getRequest), 0);
        // Get answer
        recv(sock, response, 256, 0);
        char *ipStr = strstr(response, "\"ip\":\"") + 6;
        if (ipStr != NULL)
        {
            *(strrchr(ipStr, '\"')) = 0;
            devc.extIPaddress = inet_addr(ipStr);
            DEBUG_MSG(CDL, (COL_WIFI "Ip address %s [%d]", ipStr, devc.extIPaddress));
        }
    }
    close(sock);
}
```
usually it falls in getExtIP() function
```ansi
I (72564) wifi.c:2028 : ip addr: 104.26.13.205
I (72739) wifi.c:2044 : 104.26.13.205: connected; socket: 58
I (72742) wifi.c:2062 : Port 80 opened
I (72767) wifi.c:1957 : socket #58 opened, delete task
```
 or sslClientTask in connectibg to cloud
```ansi
I (9791) esp-cloud.c:0346 : Connecting to p1:9030... freeHeap 149636
```