espressif/esp-zigbee-sdk #429: Obtaining at compile/runtime the ESP Zigbee Library version?

monty68, 2024-09-14:
### Question

Can we obtain the version number/details of the ESP Zigbee library version at compile time and/or at runtime?

For ZBoss, there is the zb_get_version() function.

xieqinan, 2024-10-17:
@monty68 ,

> Can we obtain the version number/details of the ESP Zigbee library version at compile time and/or at runtime?

The esp-zigbee-sdk v1.5.1 has already supported this feature, please refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L961.

xieqinan, 2024-10-28:
Hi @monty68 ,

Does this feature https://github.com/espressif/esp-zigbee-sdk/issues/429#issuecomment-2419191799 meet your requirement? If yes, please consider closing this issue.

monty68, 2024-10-28:
Could we not have a simple #define in the header as well?

xieqinan, 2024-11-15:
> Could we not have a simple #define in the header as well?

I cannot clearly understand your request; could you please provide more details about this requirement?

closed 2024-12-20 by chshu