espressif/esp-zigbee-sdk #431: Using analog input clusters in esp-zigbee-sdk

Daniel-Craciun, 2024-09-15:
### Question

Hello,

I'm currently trying to create a presence sensor in zigbee and I'm attempting to use the analog input clusters provided by the ZCl but I am struggling to get it working. I've looked everywhere for a possible example but no luck.

I'm using zigbee sdk version ~1.5.0 and esp-idf version 5.1.3

Any help would be much appreciated :)

prairiesnpr, 2024-09-18:
I have an example here, it's a bit rough, but it does show how to use an Analog Input. https://github.com/prairiesnpr/esp_zha_test_bench

Daniel-Craciun, 2024-09-18:
Thank you for the example it was very helpful. 

Unfortunately I keep having this error which I don't know how to fix. I've followed exactly the example and I get this error `Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:325`  when I run `esp_zb_zcl_report_attr_cmd_req`.

For context I'm running this code on the ESP32C6 as an end device type.

Any help would be very much appreciated :)

prairiesnpr, 2024-09-19:
I just flashed this to an ESP32C6 mini and it runs without issue. I'm have mine configured as a router, but that shouldn't matter, I've tested as an end device also. You are flashing the project as cloned from github or did you just follow as a general outline? Would need sample code to comment further, if it's not the same project.

Daniel-Craciun, 2024-09-19:
I have cloned your repo and flashed your exact code on my ESP32C6 mini configured as an end device and I still have the same error as before. Is there anything extra I need to do? such as configuring anything parameters in the menuconfig?

[image]

prairiesnpr, 2024-09-19:
This is the config I used on mine, maybe compare to what you have?
https://github.com/prairiesnpr/esp_zha_test_bench/blob/master/sdkconfig-c6

prairiesnpr, 2024-09-19:
Ok, I think your issue is you need to update main.h to reflect you are building an endpoint. https://github.com/prairiesnpr/esp_zha_test_bench/blob/fcad8d2fe90ae8f3a62bffb625988fd6e84d5f01/main/main.h#L36

Change ```ESP_ZB_DEVICE_TYPE_ROUTER``` to ```ESP_ZB_DEVICE_TYPE_ED```

Daniel-Craciun, 2024-09-20:
I've finally managed to get your example working :)

The issue was not about using the incorrect zigbee endpoint type, however it was the fact that I did not initialise the nvs hence why you see an error in the previous image I've shown. I have inserted the nvs init code before the `esp_zb_platform_config` function and everything works now fine. For reference this is the code I added incase others have an issue.

`
// Initialize NVS
esp_err_t ret = nvs_flash_init();
if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    ESP_ERROR_CHECK( nvs_flash_erase() );
    ret = nvs_flash_init();
}
ESP_ERROR_CHECK( ret );
`

Many thanks for the example which I will now use as a boilerplate code for my presence sensor! I will leave the issue open incase of any other errors (hopefully no more!).

Daniel-Craciun, 2024-09-21:
Hello this is a follow up incase anyone has any issues with this. 

If you are writing Zigbee code and you are changing the endpoints, clusters and attribute ids. **You MUST always erase the chip flash before flashing your new code** with updated ids. As I think the old ids are not being cleared out if you don't erase the flash. 

For example in esp-idf you should run the command `idf.py -p (PORT) erase-flash` before uploading any new code to ensure nothing is cached from previous projects uploaded to the board. This command helped me to get the analog input cluster example code finally working!

Many thanks again for this code sample it is much appreciated!

chshu, 2024-09-29:
@prairiesnpr Thanks a lot for your supporting on the question!

closed 2024-09-29 by chshu

amne, 2025-06-24:
@Daniel-Craciun what you said should be pinned on top of every SDK user guide page!