espressif/esp-zigbee-sdk #432: zdo -n equivalent in API: How to get nwk address by IEEE address with the SDK?

remenyo, 2024-09-16:
### Question

How to request network address by ieee address? I want to send requests with esp_zb_aps_data_request, that seemingly only accepts short address, and I did not find the function that allows me to query that by known ieee address.
I would like basically a zigbee equivalent of ARP requests.

Thank you for your help.

### Additional context.

_Already asked by me as a followup here, but did not get anwser: https://github.com/espressif/esp-zigbee-sdk/issues/395#issuecomment-2257769013_

ahmetcobanoglu, 2024-09-17:
@remenyo 

I believe the function esp_zb_address_short_by_ieee(esp_zb_ieee_addr_t address) should work for this purpose. The device's network table likely stores both IEEE and short addresses. You can handle this using the function above without sending a ZDO command.

remenyo, 2024-09-18:
The problem is that I know it does not hold in the table, in cases for example when the device is freshly added to the network, knowing only the IEEE addresses of the devices it should control, but not their network addresses.

esp_zb_zdo_ieee_addr_req() gets the IEEE address of a device with a given nwk address, but there is no
esp_zb_zdo_nwk_addr_req() variant. Perhaps it does not exist?

ahmetcobanoglu, 2024-09-19:
@remenyo 

You can send a request using: `zb_uint8_t zb_zdo_nwk_addr_req(zb_uint8_t param, zb_callback_t cb)` You can find the function in the ZBOSS stack. This function can be requested as a Feature Request.

lpy4105, 2024-09-19:
@ahmetcobanoglu 
Thanks for your answer.

@remenyo 
`esp_zb_zdo_nwk_addr_req` currently not supported, we will add it in the next release.

remenyo, 2024-09-19:
Thank you for both of you. Should I leave this open until it gets released?

lpy4105, 2024-09-19:
Yes, of course

xieqinan, 2024-10-18:
Hi @remenyo ,

The `esp_zb_zdo_nwk_addr_req()` feature is already supported in esp-zigbee-sdk v1.5.1; you can refer to [this link](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_command.h#L501). 

If this feature meets your needs, please consider closing the issue.

remenyo, 2024-10-21:
Thank you, I tried to use `esp_zb_zcl_on_off_cmd_req` and `esp_zb_aps_data_request` and they both work now directly with 
ESP_ZB_APS_ADDR_MODE_64_ENDP_PRESENT, and the destination IEEE address memcpy-ed into the `.zcl_basic_cmd.dst_addr_u.addr_long` and `.dst_addr.addr_long` respectively.

closed 2024-10-21 by remenyo