espressif/esp-zigbee-sdk #433: Get external temperature to thermostat router

kioelka, 2024-09-18:
### Question

chip ESP32C6
IDF 5.3.0

I have a thermostat connected to the gateway in router mode, but I faced a question. in the examples all thermostats have their own temperature sensor and send in cluster 0x201 all attributes of the thermostat to the coordinator. but if I want to do the opposite how can I do it? i.e. I have a temperature sensor xiaomi LYWSD03MMC and I want to receive temperature reports from it. how can I do it at the transport level or configure smart home software to retransmit the temperature to the thermostat in any way?

lpy4105, 2024-09-19:
If you are able to get the address of the temperature sensor, you can try to use `esp_zb_zcl_read_attr_cmd_req` to read the attributes from the specific endpoint on the temperature sensor. Please refer to our example:
https://github.com/espressif/esp-zigbee-sdk/blob/d4af0c84351b75395a984277f6a55a6d7a46c07b/examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c#L61-L73

kioelka, 2024-09-23:
> If you are able to get the address of the temperature sensor, you can try to use `esp_zb_zcl_read_attr_cmd_req` to read the attributes from the specific endpoint on the temperature sensor. Please [...]

Thanks, it can get temperature. but,  how to apply this code but in router mode instead of coordinator mode? i tried to edit the code and pair sensor, thermostat and coordinator, but by pressing "READ ATTR" button i get 0.0 celsius on "ROUTER" and homeassistant  gets temperature from sensor

```
I (471) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:4 
I (471) ESP_ZB_THERMOSTAT: Deferred driver initialization successful
I (481) ESP_ZB_THERMOSTAT: Device started up in non factory-reset mode
I (491) ESP_ZB_THERMOSTAT: Device rebooted
I (551) ESP_ZB_THERMOSTAT: Network(0x4d95) is open for 180 seconds
I (6981) ESP_ZB_THERMOSTAT: ZDO signal: ZDO Leave Indication (0x13), status: ESP_OK
I (6991) ESP_ZB_THERMOSTAT: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (11031) ESP_ZB_THERMOSTAT: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (11041) ESP_ZB_THERMOSTAT: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (11551) ESP_ZB_THERMOSTAT: New device commissioned or rejoined (short: 0x7e25)
I (12071) ESP_ZB_THERMOSTAT: Found temperature sensor
I (12071) ESP_ZB_THERMOSTAT: Request temperature sensor to bind us
I (12071) ESP_ZB_THERMOSTAT: Bind temperature sensor
I (12081) ESP_ZB_THERMOSTAT: Successfully bind the temperature sensor from address(0x7e25) on endpoint(1)
I (12611) ESP_ZB_THERMOSTAT: The temperature sensor from address(0x7e25) on endpoint(1) successfully binds us
I (12631) ESP_ZB_THERMOSTAT: Read attribute response: from address(0x7e25) src endpoint(1) to dst endpoint(1) cluster(0x0)
I (12631) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x0), attribute(0x4), type(0x42), value(16)
I (12641) ESP_ZB_THERMOSTAT: Peer Manufacturer is "_TZE200_znbl8dj5"
I (12651) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x0), attribute(0x5), type(0x42), value(6)
I (12661) ESP_ZB_THERMOSTAT: Peer Model is "TS0601"
I (14401) ESP_ZB_THERMOSTAT: Network(0x4d95) is open for 180 seconds
I (18581) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (22621) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (28361) ESP_ZB_THERMOSTAT: Send 'configure reporting' command
I (28361) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (30481) ESP_ZB_THERMOSTAT: ZDO signal: NWK No Active Links Left (0x18), status: ESP_OK
I (34871) ESP_ZB_THERMOSTAT: Send 'configure reporting' command
I (34871) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (36051) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (36051) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (36441) ESP_ZB_THERMOSTAT: Configure report response: status(134), cluster(0x402), direction(0x0), attribute(0x0)
I (36451) ESP_ZB_THERMOSTAT: Read attribute response: from address(0x7e25) src endpoint(1) to dst endpoint(1) cluster(0x402)
I (36451) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(0)
I (36461) ESP_ZB_THERMOSTAT: Measured Value is 0.00 degrees Celsius
```

[image]
[image]

lpy4105, 2024-09-23:
It's not related to the device type (ZC/ZR/ZED). You need to update the attribute of the thermostat cluster. I believe it should be `LocalTemperature` attribute.

I'm a little bit worried about your provided logs:
>I (36451) ESP_ZB_THERMOSTAT: Read attribute response: from address(0x7e25) src endpoint(1) to dst endpoint(1) cluster(0x402)
I (36451) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(0)
I (36461) ESP_ZB_THERMOSTAT: Measured Value is 0.00 degrees Celsius

It seems that your thermostat failed to get correct value from the temperature sensor. If that was the case, could you please setup a sniffer and provide wireshark pcap files for both setups?

kioelka, 2024-09-23:
[zr.zip](https://github.com/user-attachments/files/17093738/zr.zip)

kioelka, 2024-09-23:
> It's not related to the device type (ZC/ZR/ZED). You need to update the attribute of the thermostat cluster. I believe it should be `LocalTemperature` attribute.
>
> I'm a little bit worried about your [...]

[zr.zip](https://github.com/user-attachments/files/17093747/zr.zip)

kioelka, 2024-09-23:
i noticed ZR thermostat tis triyng to answer coordinator 0x402 cluter but does it incorrectly

lpy4105, 2024-09-23:
Going over the traffics, I think the behaviour of the temperature sensor is weird.
- Reading the `MeasuredValue` can't get correct value, but it can be reported with correct value.
- The reporting was not going through a binding table but to a fixed address "0x0000" which indicating the coordinator.

It's not obviously a issue in our SDK. Could you please also consult with the firmware provider of the temperature sensor?

On the thermostat side, the coordinator is continuesly requesting a customized attribute with manufacturer code `0x0234`, can you please check if the device is correctly configured on homeassistant?

kioelka, 2024-09-23:
Hmmm, read attribute packetsalways include zeros, but report packets always include correct temperature.
Can I somehow “subscribe” to the reports so that they go to the router address instead of 0x0000 ZC?

lpy4105, 2024-09-23:
> Can I somehow “subscribe” to the reports so that they go to the router address instead of 0x0000 ZC?

In theory, report will send via bindings. The traffics has shown that you have bound the sensor correctly. However the reports were not sending as expected...

kioelka, 2024-09-23:
> > Can I somehow “subscribe” to the reports so that they go to the router address instead of 0x0000 ZC?
> 
> In theory, report will send via bindings. The traffics has shown that you have bound the sensor correctly. However the reports were not sending as expected...

thank you very much.
i tried again all sensors, only Xiaomi works well.
araqa TVOC doesnt answer on attr read requset, status 0x86 (invalid attribute)
TUYA answers on request, but send zeros.
XIAOMI sends reports well and sends correct temperature by pressing button (read_attr)
```
I (15261) ESP_ZB_THERMOSTAT: New device commissioned or rejoined (short: 0xb32e)
I (15561) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (16041) ESP_ZB_THERMOSTAT: Found temperature sensor
I (16041) ESP_ZB_THERMOSTAT: Request temperature sensor to bind us
I (16041) ESP_ZB_THERMOSTAT: Bind temperature sensor
I (16051) ESP_ZB_THERMOSTAT: Successfully bind the temperature sensor from address(0xb32e) on endpoint(1)
I (16581) ESP_ZB_THERMOSTAT: The temperature sensor from address(0xb32e) on endpoint(1) successfully binds us
I (16611) ESP_ZB_THERMOSTAT: Read attribute response: from address(0xb32e) src endpoint(1) to dst endpoint(1) cluster(0x0)
I (16611) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x0), attribute(0x4), type(0x42), value(6)
I (16631) ESP_ZB_THERMOSTAT: Peer Manufacturer is "Xiaomi"
I (16631) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x0), attribute(0x5), type(0x42), value(10)
I (16641) ESP_ZB_THERMOSTAT: Peer Model is "LYWSD03MMC"
I (20931) ESP_ZB_THERMOSTAT: Received report from address(0xb32e) src endpoint(1) to dst endpoint(1) cluster(0x402)
I (20931) ESP_ZB_THERMOSTAT: Measured Value is 26.56 degrees Celsius
I (23631) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (28681) ESP_ZB_THERMOSTAT: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (31531) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (31571) ESP_ZB_THERMOSTAT: Send 'read attributes' command
I (35371) ESP_ZB_THERMOSTAT: Read attribute response: from address(0xb32e) src endpoint(1) to dst endpoint(1) cluster(0x402)
I (35371) ESP_ZB_THERMOSTAT: Read attribute response: status(0), cluster(0x402), attribute(0x0), type(0x29), value(114)
I (35381) ESP_ZB_THERMOSTAT: Measured Value is 26.74 degrees Celsius
```

closed 2024-09-23 by kioelka