espressif/esp-zigbee-sdk #435: coordinator ignores *_network_channel_set

mw75, 2024-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.1

### esp-zigbee-lib version.

1.0.9

### esp-zboss-lib version.

1.0.9

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Coordinator running on Channel 11 when i call 
```
#define ESP_ZB_PRIMARY_CHANNEL_MASK     (1l << 11)
...
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    esp_zb_set_secondary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
```

### What is the actual behavior?

Coordinator is running on channel 13
```
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct){
...
case ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
        if (err_status == ESP_OK) {
...
ESP_LOGI(TAG,"Channel %d",esp_zb_get_current_channel());
...
```

> I (1295) Zigbee2CAN: Channel 13

### Steps to reproduce.

1. Take an Espressif example, i took HA_ON_OFF_SWITCH
2. set channel by esp_zb_set_primary_network_channel_set and esp_zb_set_secondary_network_channel_set
3. check by esp_zb_get_current_channel

### More Information.

The binary only distribution of esp-zigbee-lib and esp-zboss-lib makes it quite hard to help yourself. I understand, that zboss is licensed, but why is esp-zigbee-lib binary only? Or do i miss something and the code is published somewhere?

lpy4105, 2024-09-20:
You can try our [HA_on_off_switch example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/).

mw75, 2024-09-20:
Please review "Steps to reproduce." step one!

The binary only distribution of esp-zigbee-lib and esp-zboss-lib makes it quite hard to help yourself. I understand, that zboss is licensed, but why is esp-zigbee-lib binary only? Or do i miss something and the code is published somewhere?

Problem is fixed by erase-flash. Looks like stored settings in flash have priority.

chshu, 2024-09-29:
> Problem is fixed by erase-flash. Looks like stored settings in flash have priority.

Exactly, some leftover info in the flash may cause some unexpected issues. So the `erase-flash` operation is recommended.

The current esp-zigbee-sdk is released in library, we don't have the plan to opensource in short term. Feel free to raise github issues for any particular questions, you can also contact our business team via [technical-inquiries](https://www.espressif.com/en/contact-us/technical-inquiries) to request the source code, but there is some order quantity requirement.

closed 2024-09-29 by chshu

xieqinan, 2026-02-25:
@lsroka76 

> Please reopen that issue. Currently it's not possible to change coordinator primary channel without Zigbee stack factory reset, which require repairing every end device again. In case of my gateway sometimes it means repairing more than 20 devices, often hardly accessible.

Since this issue was origined the old versioni of SDK, it is not easy to trace. I suggest testing your case using the latest version. If the issue still exists, you can submit a new issue with a detailed description.