espressif/esp-zigbee-sdk #437: No attribute set callback(s) when remaining time attribute reaches zero

monty68, 2024-09-25:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

When receiving move-to type commands for level or color type changes, the stack sends 
ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID messages updating the relevant remaining time attribute, but 
when the move has been completed, the stack should issue a final attribute update with the remaining 
time set to zero.

### What is the actual behavior?

When receiving move-to type commands, the stack sends ESP_ZB_CORE_SET_ATTR_VALUE_CB_ID messages
updating the relevant remaining time attribute, but when the move has been completed, it never calls back 
with the remaining time set to zero so the application has no idea that the command has been completed 
and what the final (level/color) value is.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

xieqinan, 2024-10-18:
Hi @monty68 ,

It doesn't appear to be a bug. The ZCL implementation in the esp-zigbee-sdk is designed to simplify operations for developers by having the stack handle ZCL commands entirely. Developers mainly need to integrate with the device driver. For instance, when the stack receives a `move_to_level` command with `transition_time` set to 10 and `level` set to 100, the ZCL implementation of the level cluster manages the gradual transition to the level of 100 over 1 second and triggers callbacks at minimum intervals defined by the ZCL specification. Developers only need to update these values in the device driver through the callback without worrying about the final value or time. However, if the final value or timing deviates from the ZCL specification or expected result, then that would be a bug, and we'd correct it.

Could you please provide more details and applicaiton case on why you're expecting a final callback? If it's a reasonable request, we'll consider adding it.

monty68, 2024-10-18:
Thanks for your reply, the use case, for example, is where I need to save the level state in NVRAM to be able to use the "previous" startup level attribute setting, at the moment if I save the level each time I get a stack callback I'm hammering NVRAM where if I know a transition has ended I only need to save the final value once.

I hope that makes sense?

xieqinan, 2024-10-28:
@monty68 ,

> Thanks for your reply, the use case, for example, is where I need to save the level state in NVRAM to be able to use the "previous" startup level attribute setting, at the moment if I save the level [...]

I think it's important to record every attribute value to NVRAM if the attribute has a transition. Recording only the final value may lead to discrepancies due to unexpected interruptions or reboots, resulting in actual behavior that differs significantly from what's recorded in NVRAM, which is undesirable. Furthermore, the time taken to read or write to NVRAM is short and will not significantly impact the attribute transition process.

monty68, 2024-10-29:
I was less concerned about performance and more concerned about unnecessary wear on the NVRAM, I find it unusual to restore a device into an interim transitionary state, rather than a known existing (before transition) or the intended state, knowing when the transition has finished or even what the intended state is give developers the option to go with either of our views :-)

xieqinan, 2024-11-04:
Hi @monty68 ,

> I was less concerned about performance and more concerned about unnecessary wear on the NVRAM

Unnecessary wear on the NVRAM is indeed an issue that needs optimization. A delayed caching solution could help reduce NVRAM stress. Here’s how it could work: you allocate a variable to record the reported value from the callback in time. This variable doesn’t need to write to NVRAM immediately; instead, it can be written after a short delay, such as 1 second. If the callback triggers every 100 ms, this approach would reduce NVRAM operation frequency by 90%. What do you think of this solution?

closed 2024-12-20 by chshu