espressif/esp-zigbee-sdk #439: Backup and Restore Coordinator

mw75, 2024-09-25:
### Question

Please describe the process and required tools to backup an ESP32-C6 Coordinator in an way, that allows restore to an equivalent hardware without the need of repairing all devices and with persistent short addresses. Is a "service PC" required to backup, or is it possible to collect all required information right on the ESP and write it to a connected SD-Card?

Thanks and regards,
Mario Wolff

lpy4105, 2024-09-29:
Please check if the solution discussed in https://github.com/espressif/esp-zigbee-sdk/issues/289 satisfies your requirements.

mw75, 2024-09-30:
Hi @lpy4105
thanks for your hint regarding an already existing question with already two different solutions! The only missing part is the corresponding process step to dump zb_storage. Is it save to use "ZDO Device Authorized (0x2f)" for this?

Is NVS also used by the ZigBee-SDK as it is initialized in all examples i could find? Is the content of zb_fct volatile as it is not recommended to backup it the referenced issue?

Thanks again!

Regards,
Mario Wolff

xieqinan, 2024-10-18:
Hi @mw75 ,

All non-volatile Zigbee data is stored in `zb_storage`. The `zb_fct` is reserved for factory configuration and is typically not used during regular development.

closed 2024-12-20 by chshu