espressif/esp-zigbee-sdk #440: What is the max amount of routers in a network?

BaWe155, 2024-09-25:
### Question

I get an ERROR: Extended status 87 (ZPS_XS_E_NO_FREE_EXTENDED_ADDR) by 21 routers in a network.

What is the max number of routers in a network?

What is the size of the following tables in a router or coordinator: 
- mac address table
- routing table
- address_map tables
How to increase these tables?

chshu, 2024-09-29:
@BaWe155 You can adjust the supported network scale with this API: [esp_zb_overall_network_size_set()](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/esp_zigbee_core.html#_CPPv431esp_zb_overall_network_size_set8uint16_t).

BaWe155, 2024-09-29:
@chshu Thank you, is this for a coordinator or also for routers?

chshu, 2024-09-30:
@BaWe155 For both coordinator and routers. Please share the device logs if you still see the issue after the change.

closed 2024-12-20 by chshu