espressif/esp-zigbee-sdk #442: Zigbee coordinator working with any zigbee device

Daniel-Craciun, 2024-09-28:
### Question

Hello, 

I'm trying to create a Zigbee coordinator which will connect to my custom made Zigbee devices and hopefully any other valid Zigbee device on the market. I've noticed in the examples given on the esp-zigbee-sdk, that all the coordinators define the endpoints/clusters they will 'listen' for (act as a client role for). However I don't want this behaviour but instead I want my coordinator to connect to any Zigbee device, without needing me to define which clusters I will 'listen' for.

Any thoughts on this if it's possible?

Many thanks!

lpy4105, 2024-09-29:
If you are trying to create a gateway, you could use `esp_zb_ep_list_add_gateway_ep`, see our [Zigbee Gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway).

Daniel-Craciun, 2024-09-29:
Thank you the example was helpful. I had to erase the flash to get it working.

But I don't know how to get the attribute reporting working on it? Don't I have to use the `esp_zb_core_action_handler_register`?

Daniel-Craciun, 2024-09-30:
Thank you. Would I call this function in the coordinator or the end device? Also could I have an example snippet code to show how could I use this function as I feel that the Zigbee documentation is not very clear for me :)

Daniel-Craciun, 2024-09-30:
In addition I'm also trying to connect to end devices with multiple endpoints if that has any effect?

mundevx, 2024-10-01:
same issue, created new thread  on [https://github.com/espressif/esp-zigbee-sdk/issues/444]

Daniel-Craciun, 2024-10-01:
I've managed in the past to get an end device to report to a custom coordinator, where the coordinator does not need to constantly send read commands to the end device. I made it using the temperature/thermostat example (which instead of using temperature clusters I used analog input clusters which worked fine) but when I added multiple endpoints to the end device, only the first endpoint was able to work with the coordinator i.e the coordinator successfully picked up the report commands but only from the first end point. However the rest of the endpoints didn't work. It's almost like the coordinator can't bind to the other endpoints?? Am I missing something here? the documentation for zigbee-sdk is not very clear in my opinion for me.

Daniel-Craciun, 2024-10-02:
Hello everyone, I've finally managed to get multiple endpoints working in zigbee, my solution has been posted on this issue here https://github.com/espressif/esp-zigbee-sdk/issues/407. Hope this is helpful for anyone :)

Daniel-Craciun, 2024-10-13:
Hello, sorry for the slow reply!

Thank you for the links. I have solved my problem for me. What I did is I adapted an exisiting Zigbee coordinator example (I used the thermostat example) and I just listed after the line of code, which defined that the coordinator will listen for temperature measurements, all the standard ZCL clusters which worked for me :).

Thank you everyone for your help again. Helpful as always!

closed 2024-12-20 by chshu