espressif/esp-zigbee-sdk #443: ESP_ZB_ZCL_STATUS_UNSUP_ATTRIB Error when Writing/Reading Attribute from Manufacturer Specific Cluster

ahmetcobanoglu, 2024-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

Attributes should be successfully read or written to the manufacturer-specific cluster.

### What is the actual behavior?

ESP_ZB_ZCL_STATUS_UNSUP_ATTRIB error is returned in the status field of the response message when trying to read or write attributes.

### Steps to reproduce.

1. Set up ZigBee Coordinator with Gateway endpoint (no clusters registered).

2. Receive attribute reports from a manufacturer-specific cluster (reports received correctly).

3. Attempt to read or write an attribute from the manufacturer-specific cluster.

### More Information.

Cluster ID (attempted for reading/writing): 0xFCC0

Attribute IDs: 0x0102, 0x010C, 0x0152

Coordinator configuration: ZigBee Coordinator with Gateway endpoint (no clusters registered).

I can successfully read or write these attributes using zigpy and EFR32, but I encounter the issue only in the current setup.

ahmetcobanoglu, 2024-10-15:
@chshu Is there any development on this issue?

xieqinan, 2024-10-18:
@ahmetcobanoglu ,

Could you please provide a `.pcap` file for reading/writing the attribute of cluster 0xFCC0?

xieqinan, 2024-11-15:
@ahmetcobanoglu ,

The `esp_zb_zcl_read_attr_cmd_req()` had already support to send request with `specific manufacturer`, please refer to  https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L129

closed 2024-12-20 by chshu