espressif/esp-zigbee-sdk #444: zigbee gateway example code not accepting commands

mundevx, 2024-10-01:
### Question

All of these functions in End Device or Router is not working when send to gateway. But Gateway to End Devices are working Ok. 

All functions are listed down

	        esp_zb_zcl_on_off_cmd_t cmd_req;
	        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = 0;
	        cmd_req.zcl_basic_cmd.dst_endpoint = 1;
	        cmd_req.zcl_basic_cmd.src_endpoint = 1;
	        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
	        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
	        esp_zb_lock_acquire(portMAX_DELAY);
	        esp_zb_zcl_on_off_cmd_req(&cmd_req);
	        esp_zb_lock_release();

            // esp_zb_zcl_attribute_t attr_field;
            // attr_field.id = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID;
            // attr_field.data.type = ESP_ZB_ZCL_ATTR_TYPE_BOOL;
            // attr_field.data.size = 1;
            // attr_field.data.value = (uint8_t*)&toggle;

            // esp_zb_zcl_write_attr_cmd_t write_req;
            // memset(&write_req, 0, sizeof(write_req));
          
            // write_req.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;
            // write_req.zcl_basic_cmd.dst_endpoint = 1; // Target endpoint on the device
            // write_req.zcl_basic_cmd.src_endpoint = 1; // Source endpoint (set to 1 or 0 depending on your setup)
            // write_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
            // write_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
            // write_req.attr_field = &attr_field;
            // write_req.attr_number = 1;
            // esp_zb_lock_acquire(portMAX_DELAY);
            // esp_err_t result = esp_zb_zcl_write_attr_cmd_req(&write_req);
            // esp_zb_lock_release();

            // esp_zb_zcl_status_t err = esp_zb_zcl_set_attribute_val(
            //     1,
            //     ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
            //     ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            //     ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
            //     (bool*)&toggle,
            //     false
            // ); 

            //printf("err:%d state:%d\n", err, toggle);
            // esp_zb_zcl_report_attr_cmd_t report_cmd = {
            // .zcl_basic_cmd = {
            //     .dst_addr_u.addr_short = 0x0000,
            //     .dst_endpoint = 1,
            //     .src_endpoint = 1,
            // },
            // .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
            // .clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
            // .attributeID = ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID,
            // .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            // };
            // esp_zb_zcl_report_attr_cmd_req(&report_cmd);     

but when i use this function,  esp_zb_zcl_ias_zone_status_change_notif_cmd_req ,
it is working fine and callback occur on gateway example code, but all other functions listed above are not working.

// esp_zb_zcl_ias_zone_status_change_notif_cmd_t resp;
    // resp.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    // resp.extend_status = 1;
    // resp.zone_id = 0;
    // resp.zone_status = toggle;
    // resp.zcl_basic_cmd.dst_addr_u.addr_short = 0;
    // resp.zcl_basic_cmd.dst_endpoint = 1;
    // resp.zcl_basic_cmd.src_endpoint = 1;
    // esp_zb_zcl_ias_zone_status_change_notif_cmd_req(&resp);

Can you help me regarding this issue?

mundevx, 2024-10-01:
In simple words, Zigbee coordinator gateway only working one sided (i.e. from gateway to zigbee end device). But not working when i send command from  Zigbee end device to Gateway (i.e. No callback occur on gateway side). But gateway callbacks are working for IAS Zone clusters only(From Sensor to gateway and gateway to Sensor).

ahmetcobanoglu, 2024-10-09:
You need to register gateway endpoint. @mundevx

mundevx, 2024-10-10:
Hi @ahmetcobanoglu , 
I had already registered the gateway endpoint. But its working now but now issue arise that the gateway send data immediately to end device and received but end device immediately,  but from end device, when i send data, then it reaches gateway at 5 second timeout. What i am sending... 
        esp_err_t status = esp_zb_zcl_set_attribute_val(
            1,
            ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL,
            ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            ESP_ZB_ZCL_ATTR_FAN_CONTROL_FAN_MODE_ID,
            (uint8_t*)&fan_speed,
            false
        );
and then then 
    esp_zb_zcl_report_attr_cmd_t report_cmd_2 = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000,
            .dst_endpoint = 1,
            .src_endpoint = 1,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL,
        .attributeID = ESP_ZB_ZCL_ATTR_FAN_CONTROL_FAN_MODE_ID,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    };
    esp_zb_zcl_report_attr_cmd_req(&report_cmd_2);

mundevx, 2024-10-10:
Hi @ahmetcobanoglu , 
I had already registered the gateway endpoint. But its not fully working. Now issue arise that the gateway send data immediately to end device and received by end device immediately,  but from end device, when i send data, then it reaches gateway at 5 second timeout. What i am sending... 
        esp_err_t status = esp_zb_zcl_set_attribute_val(
            1,
            ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL,
            ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            ESP_ZB_ZCL_ATTR_FAN_CONTROL_FAN_MODE_ID,
            (uint8_t*)&fan_speed,
            false
        );
and then then 
    esp_zb_zcl_report_attr_cmd_t report_cmd_2 = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000,
            .dst_endpoint = 1,
            .src_endpoint = 1,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL,
        .attributeID = ESP_ZB_ZCL_ATTR_FAN_CONTROL_FAN_MODE_ID,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    };
    esp_zb_zcl_report_attr_cmd_req(&report_cmd_2);

xieqinan, 2024-10-28:
@mundevx ,

Please note that the gateway typically functions as the client side of a cluster to connect with other interfaces and send commands to control other devices. Using `esp_zb_ep_list_add_gateway_ep()` registers the cluster clients, enabling the gateway device to send commands. However, if the on/off server cluster is not registered on the gateway, it will not process on/off commands.

Additionally, it’s essential to consider providing the application context. Using a gateway as a server to receive and handle cluster commands is generally not recommended.

closed 2024-12-20 by chshu