espressif/esp-zigbee-sdk #445: Set Extended PAN ID not working as coordinator

MzeDeVi, 2024-10-01:
### Question

Im trying to use the extended PAN-ID as SSID for my network. this code does not change the extended PAN-ID. Can someone tell me why? (ESP32H2, espressif/esp-zboss-lib: "1.5.0" espressif/esp-zigbee-lib: "1.5.0" idf: version: ">=5.0.0")

### Additional context.

void Zigbee_Initialize(void)
{
    esp_zb_cfg_t Zigbee_Config = ESP_ZIGBEE_CONFIG();
    (void) esp_zb_init(&Zigbee_Config);
    esp_zb_ieee_addr_t Address = {'U', 'P', '0', '0', '0', '0', '0', '0'};
    printf("%c:%c:%c:%c:%c:%c:%c:%c\r\n",Address[0],Address[1],Address[2],Address[3],Address[4],Address[5],Address[6],Address[7]);
    esp_zb_set_extended_pan_id(Address);
    esp_zb_start(true);
    esp_zb_stack_main_loop_iteration();
    esp_zb_get_extended_pan_id(Address);
  printf("%c:%c:%c:%c:%c:%c:%c:%c\r\n",Address[0],Address[1],Address[2],Address[3],Address[4],Address[5],Address[6],Address[7]);
    printf("ZigBee Started\r\n");
}

I (23) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (24) boot: compile time Oct  1 2024 07:21:27
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (32) boot.esp32h2: SPI Mode       : DIO
I (37) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 000e1000
I (80) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42070020 size=0f0d4h ( 61652) map
I (127) esp_image: segment 1: paddr=0001f0fc vaddr=40800000 size=00f1ch (  3868) load
I (129) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=6ee8ch (454284) map
I (273) esp_image: segment 3: paddr=0008eeb4 vaddr=40800f1c size=0c4e4h ( 50404) load
I (291) esp_image: segment 4: paddr=0009b3a0 vaddr=4080d400 size=01d7ch (  7548) load
I (299) boot: Loaded app from partition at offset 0x10000
I (299) boot: Disabling RNG early entropy source...
I (312) cpu_start: Unicore app
I (321) cpu_start: Pro cpu start user code
I (321) cpu_start: cpu freq: 96000000 Hz
I (321) app_init: Application information:
I (324) app_init: Project name:     Receiver
I (328) app_init: App version:      1
I (333) app_init: Compile time:     Oct  1 2024 07:21:12
I (339) app_init: ELF file SHA256:  db3e66ee1...
I (344) app_init: ESP-IDF:          -128-NOTFOUND
I (349) efuse_init: Min chip rev:     v0.0
I (354) efuse_init: Max chip rev:     v0.99 
I (359) efuse_init: Chip rev:         v0.1
I (364) heap_init: Initializing. RAM available for dynamic allocation:
I (371) heap_init: At 40814410 len 00038F70 (227 KiB): RAM
I (377) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (385) spi_flash: detected chip: generic
I (388) spi_flash: flash io: dio
W (392) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (406) sleep: Configure to isolate all GPIO pins in sleep state
I (412) sleep: Enable automatic switching of GPIO sleep configuration
I (420) main_task: Started on CPU0
I (430) main_task: Calling app_main()
I (440) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (450) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
85:80:48:48:48:48:48:48
12:49:96:254:255:189:77:116
ZigBee Started

mw75, 2024-10-02:
Maybe you should try a `idf.py erase-flash`. Setting the channel seams to also not work when there is a corresponding setting somewhere in flash. See https://github.com/espressif/esp-zigbee-sdk/issues/435

xieqinan, 2024-10-21:
Hi @mw75,
Thanks for your support.

The `esp_zb_set_extended_pan_id()` only will take effect, when the coordinator starts in factoryreset state. It seems that the issue has been resolved by yourself @MzeDeVi, could you please consider closing it?

closed 2024-10-21 by MzeDeVi