espressif/esp-zigbee-sdk #446: Time cluster

akira215, 2024-10-01:
### Question

In `esp_zigbee_type.h` the mandatories attributes are : `uint16_t time;` and `uint16_t time_status;`. However, the ZCL spec rev 8 instructs that _Time_ attribute is UTC type which should be 32 bits in length, and _TimeStatus_ attribute is map8 (which should obviously be 8bits length).

Am I missing something or is there any mistake in the struct declaration ?

xieqinan, 2024-10-21:
Hi @akira215 ,

Thank you for testing. It is indeed an error in the declaration of the `esp_zb_time_cluster_cfg_s` structure in the esp-zigbee-sdk. We will address this issue in the next release. In the meantime, you can use the following code as a workaround.

```
esp_zb_attribute_list_t *esp_zb_app_time_cluster_create(void)
{
    uint32_t lu32_time_default;
    uint8_t  lu8_time_status_default;

    esp_zb_attribute_list_t *esp_zb_time_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TIME);
    esp_zb_time_cluster_add_attr(esp_zb_time_cluster, ESP_ZB_ZCL_ATTR_TIME_TIME_ID, &lu32_time_default);
    esp_zb_time_cluster_add_attr(esp_zb_time_cluster, ESP_ZB_ZCL_ATTR_TIME_TIME_STATUS_ID, &lu8_time_status_default);
    return esp_zb_time_cluster;
}

```

akira215, 2024-10-21:
Thanks for the update, and for the incoming fix !

closed 2024-10-21 by akira215