espressif/esp-zigbee-sdk #447: esp_zb_lock_acquire in IOT-Button-Callback freezes

mw75, 2024-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

On a button press, a connected zigbee powerplug should be toggled and the implemented coordinator should open the network for joining.

```
static void button_single_click_cb(void *arg,void *usr_data)
{
    esp_zb_apsde_data_req_t req = {
        .dst_addr_mode  = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .dst_short_addr = 0x8d3e,
        .dst_endpoint   = 1,
        .src_endpoint   = 1,
        .profile_id     = 0x0104,
        .cluster_id     = 0x0006,
        .tx_options     = 0,
        .use_alias      = false,
        .alias_src_addr = 0,
        .alias_seq_num  = 0,
        .radius         = 5,
        .asdu_length    = 3,
        .asdu           = (uint8_t[3]){0x01,0x42,0x02}
    };
    if(esp_zb_lock_acquire(1000/portTICK_PERIOD_MS)){
        ESP_ERROR_CHECK(esp_zb_aps_data_request(&req));  
    }else{
        ESP_LOGE(TAG, "esp_zb_lock_acquire failed");
    }
    ESP_ERROR_CHECK(esp_zb_bdb_open_network(240));
}
```

By commenting the esp_zb_lock_acquire everything works as expected!

### What is the actual behavior?

The SoC - or at least the zigbee stack - looks like frozen. I have led_indicator_set_on_off calls on the ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS signal of the esp_zb_app_signal_handler and the led is not turned on or off. I send received aps-data via TWAI and the bus shows nothing. 

### Steps to reproduce.

1. use esp_zb_switch example from the SDK
2. add espressif/button component
3. create an iot_button with gpio_btn_cfg and register the function above
4. see the example working until you press the configured key

### More Information.

When debugging the hole handler is passed, but a breakpoint right on the entry of esp_zb_app_signal_handler is never triggered.

mw75, 2024-10-18:
In https://github.com/espressif/esp-zigbee-sdk/issues/435 @chshu  wrote

> Feel free to raise github issues for any particular questions,...

That does not work out that well here....

chshu, 2024-10-18:
@mw75 The `esp_zb_lock_acquire()` and `esp_zb_lock_release()` should be used in pairs:

```
if(esp_zb_lock_acquire(1000/portTICK_PERIOD_MS)){
        ESP_ERROR_CHECK(esp_zb_aps_data_request(&req));  
        esp_zb_lock_release();
    }else{
```

closed 2024-12-20 by chshu