espressif/esp-zigbee-sdk #448: Is there any esp_zb_zcl_fan_mode_cmd_req

mundevx, 2024-10-05:
### Question

Like esp_zb_zcl_on_off_cmd_req, is there any esp_zb_zcl_fan_mode_cmd_req like function.

xieqinan, 2024-10-21:
@mundevx ,

Regarding the `esp_zb_zcl_fan_mode_cmd_req`, are you referring to a fan mode command for the ZCL fan_control (0x202U) cluster? If so, the ZCL specification does not define any commands for this cluster, which is why it is not included in the SDK. If you still require this functionality, I recommend using `esp_zb_zcl_custom_cluster_cmd_req()` to create and send a custom command.

mundevx, 2024-10-23:
@xieqinan ,

I want to change the value of attribute ESP_ZB_ZCL_ATTR_FAN_CONTROL_FAN_MODE_ID in cluster ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL. what will be the value of "custom_cmd_id" in                             esp_zb_zcl_custom_cluster_cmd_req_t fan_mode_cmd_req = {
                                .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
                                .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_FAN_CONTROL,
                                .custom_cmd_id = ??, 
                                .data.value = &fan_speed,
                                .data.size = 1,
                                .data.type = ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
                                .direction = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                                .profile_id = 0x0104,
                                .zcl_basic_cmd.dst_addr_u.addr_short = 0,
                                .zcl_basic_cmd.dst_endpoint = 1,
                                .zcl_basic_cmd.src_endpoint = 1,
                            };  esp_zb_zcl_custom_cluster_cmd_req(&fan_mode_cmd_req );    
I have fan_speed from 0 to 4.

xieqinan, 2024-10-23:
The `esp_zb_zcl_custom_cluster_cmd_req()` provides a common API for sending custom commands, meaning you can set `custom_cmd_id` to any value. However, you will need to handle this custom command yourself in the `ESP_ZB_CORE_CMD_CUSTOM_CLUSTER_REQ_CB_ID` callback on the receiver side.

closed 2024-12-20 by chshu