espressif/esp-zigbee-sdk #452: How to join a network with a preconfigured network key?

dark-squirrel, 2024-10-13:
### Question

Hello,

I am working on a project where I want to join a ZigBee network with a pre-configured network key. I used the temperature sensor example as a basis and tried to set the network key with the esp_zb_secur_network_key_set() function. I have placed this between the calls to esp_zb_init() and esp_zb_start().

The function to set the key seems to be successful. However, shortly after receiving the association response from the coordinator, I receive a leave message from my device.

I hope someone can help me with this.

xieqinan, 2024-10-21:
@dark-squirrel ,

It is impossible for a factory-reset device to join a network using only the network key. Could you please provide more details about the application scenario in your project? What is the purpose of performing this operation?

dark-squirrel, 2024-10-21:
Hi @xieqinan , 
The procedure is as follows. The joining process in the project is the normal one with association request. However, the network key is not supplied by the Trust Center, but is to be implemented in the firmware from the outset. Therefore the transport key frame is omitted. I assume that the device waits for this and since it does not arrive, it leaves the network. Is it possible to implement this?

xieqinan, 2024-10-21:
@dark-squirrel ,

> The procedure is as follows. The joining process in the project is the normal one with association request. However, the network key is not supplied by the Trust Center, but is to be implemented in [...]

I believe the explanation above clarifies why the coordinator forces the device to leave. It's not an ideal method for getting a device to join a network. Could you explain why you prefer this approach?

dark-squirrel, 2024-10-21:
The aim should be to ensure that the network key is not sent with the standard link key encryption. My coordinator has the setting that the network key can be preconfigured, I wanted to use this method. But the end device sends the leave frame afte the association. However, I could also preconfigure a link key if that is easier to implement.

xieqinan, 2024-10-21:
> The aim should be to ensure that the network key is not sent with the standard link key encryption.

If you want to prevent the network key from being encrypted by the standard link key (Alliance09) from leaking, I recommend using `esp_zb_secur_TC_standard_preconfigure_key_set()` to set a custom standard link key in your network or use the install code method to create the network. Pre-configuring network key method is not compliant with the Zigbee BDB specification flow.

closed 2024-10-21 by dark-squirrel