espressif/esp-zigbee-sdk #453: In light_example, is it possible to do one switch and multiple lights? I have tried but problems...

josthlee, 2024-10-14:
### Question

I just flash one switch (HA_on_off_switch), and two lights (HA_on_off_light) say light A and light B on ESP32H2 and the switch seems bind to the two lights successfully. When I try to remove (unplug) light A, you can see ESP_ZB_NLME_STATUS_INDICATION (I think it's parent link failure or no route and it makes sense to me) on the switch side and it is so far working properly. Then I try to plugin light A back and unplug light B, and the problem emerged. 

You still can see the ESP_ZB_NLME_STATUS_INDICATION shows up, however, I try to press the toggle button and I got a long delay to see the light ON/OFF. 

Then I tried to set the ED_AGING_TIMEOUT on end device side to 10 seconds you can see that the end device successfully disappeared from the neighbor info table via esp_zb_nwk_get_next_neighbor call, both light A and light B. 

My question is, it seems like ED_AGING_TIMEOUT is working properly and I unplug light A and light B. But why I got long delay when I tried to unplug light B only? Is it possible to do "one switch and multiple lights" by the light_example? How can I correctly deal with the offline end device?

Or, how can I do with "one switch and multiple lights" scenario in a correct way? 

I just tried to read document many times and search for some clues but in vein. Please help and thank you!

### Additional context.

My environment:

IDF v5.3
espressif/esp-zboss-lib: "1.0.9"
espressif/esp-zigbee-lib: "1.0.9"

also I tried to upgrade to 
espressif/esp-zboss-lib: "1.5.0"
espressif/esp-zigbee-lib: "1.5.0"

xieqinan, 2024-10-21:
@josthlee ,

Thank you for your testing. The issue you encountered is a known one, similar to [this issue](https://github.com/espressif/esp-zigbee-sdk/issues/354). We are currently discussing an appropriate solution to resolve it. We will notify you once the issue is fixed.

josthlee, 2024-10-21:
> @josthlee ,
> 
> 
> 
> Thank you for your testing. The issue you encountered is a known one, similar to [this issue](https://github.com/espressif/esp-zigbee-sdk/issues/354). We are currently discussing an appropriate solution to resolve it. We will notify you once the issue is fixed.

Noted and thank you so much! I'll be waiting for the solution!

closed 2024-12-06 by espressif-bot

xieqinan, 2024-12-06:
Hi,

We have provided a solution to address this issue in esp-zigbee-sdk v1.6.1. Please test with the new version.