espressif/esp-zigbee-sdk #454: Power ON issue

mundevx, 2024-10-16:
### Question

2 out of 10 power reset, i am getting this log message :-->
_rst:0x1 (POWERON),boot:0x0 (USB_BOOT)
wait usb download_

And on successful boot, i am getting :-->
_rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee_

GPIO9 is 10K pullup
GPIO8 is floating (10k +1uF)
GPIO25 is floating.

what will be the issue?

mundevx, 2024-10-16:
I am using esp32h2-mini1 and my own external power supply.

mundevx, 2024-10-22:
Hi @xieqinan ,

My zigbee router light is creating issue on POWER-ON. Some times it boots successfully and app_main() functionality runs and sometimes it getting hang on  "rst:0x1 (POWERON),boot:0x0 (USB_BOOT)  wait usb download".

xieqinan, 2024-10-28:
@mundevx ,

This is not Zigbee issue, I think you can get a better solution if you push the issue [here](https://github.com/espressif/esp-idf/issues).

chshu, 2024-10-28:
@mundevx Please contact our hardware support team: https://www.espressif.com/en/contact-us/technical-inquiries/hardware-issues, it seems some issue in your hardware design.

closed 2024-12-20 by chshu