espressif/esp-zigbee-sdk #455: esp_zb_core_action_handler_register callback sometimes not handling two diiferent endpoints responses at same time

mundevx, 2024-10-18:
### Question

I am sending on-off toggle command for both endpoints 1 and 2 from coordinator (like home assistant or tuya) at SAME TIME to my zigbee end device with two endpoints 1,2. The end device is running with esp_zb_core_action_handler_register  callback. Not all callbacks are received, it is skipping some messages in esp_zb_core_action_handler_register.

I can show you the debug messages:
I (743688) ESP_ZB_CLUSTERS: ======> Callback 2 EP2 ON-OFF=On <======
I (743705) ESP_ZB_CLUSTERS: ======> Callback 3 EP1 ON-OFF=On <======

I (744503) ESP_ZB_CLUSTERS: ======> Callback 4 EP2 ON-OFF=Off <======
I (744509) ESP_ZB_CLUSTERS: ======> Callback 5 EP1 ON-OFF=Off <======

I (744946) ESP_ZB_CLUSTERS: ======> Callback 6 EP2 ON-OFF=On <======
I (744963) ESP_ZB_CLUSTERS: ======> Callback 7 EP1 ON-OFF=On <======

I (745610) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (745689) ESP_ZB_CLUSTERS: ======> Callback 8 EP2 ON-OFF=Off <======
I (745756) ESP_ZB_CLUSTERS: ======> Callback 9 EP1 ON-OFF=Off <======

**I (746211) ESP_ZB_CLUSTERS: ======> Callback 10 EP1 ON-OFF=On <======
I (748519) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (748684) ESP_ZB_CLUSTERS: ======> Callback 11 EP1 ON-OFF=Off <======**

I (745689) ESP_ZB_CLUSTERS: ======> Callback 12 EP2 ON-OFF=On <======
I (745756) ESP_ZB_CLUSTERS: ======> Callback 13 EP1 ON-OFF=On <======

Why is this so, Is there any timeout for callback when 2 messages send at same time?

mundevx, 2024-10-18:
I also observe that, in end device, the issue arise only in endpoint 2, not in endpoint 1 when coordinator send message simultaneously(at same time). But if i send message from coordinator one by one (separately) to both end points, then end device not skipping endpoint 2 callback.

xieqinan, 2025-09-30:
Hi,

Could you please set up a sniffer to check whether every on/off frame is acknowledged? If a frame is not sent by the sender or is dropped over the air, the corresponding message will be triggered.