espressif/esp-zigbee-sdk #457: OnOff: Off With Effect message from coordinator gets a Default Response Invalid Field (0x85) from my device.

honusz, 2024-10-18:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

The message should be passed to the attribute_handler.

### What is the actual behavior?

Device replies to coordinator with Default Response (0x0b) Invalid Field (0x85)

### Steps to reproduce.

NETWORK KEY: 6c:60:0a:40:c5:04:9a:b47f:0b:e9:62:17:4b:08:4b
[off_with_effect-default_response.pcapng.zip](https://github.com/user-attachments/files/17440686/off_with_effect-default_response.pcapng.zip)

### More Information.

I'm running a slightly modified version of the HA_Color_Dimmable_Light example @xieqinan has posted in other issues. It connects to my Philips Hue hub no problem, the light is added properly.
On commands (0x01) to the On/Off Cluster work fine. 
Philips Hue uses `Off with effect (0x40)` with an Effect ID of `Delayed All Off (0x00)`, and an Effect Variant of `Fade to off in 0.8 seconds (0x00)` which looks in-spec, but the command doesn't get to Attribute Handler, and the device responds with `Default Response (0x0b) - Invalid Field (0x85)`

Attaching the debug output, apologies I don't have the familiarity with this to isolate the relevant messages so just including it all. [debug_output.txt](https://github.com/user-attachments/files/17440761/debug_output.txt)

Does anything jump out? From as best as I can tell, the Off_with_effect message doesn't have any invalid fields. 
From reading the docs, it looks like this command is supported.

QDUNI, 2024-10-19:
I’ve been working on the exact same approach and encountered the same outcome as described.

xieqinan, 2024-10-21:
@honusz @QDUNI ,

I think this issue is triggered by the missing optional `on_off` attributes `ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME` and `ESP_ZB_ZCL_ATTR_ON_OFF_GLOBAL_SCENE_CONTROL`. Could you please add them to your project? You can refer to the code below:
```
    uint16_t on_off_on_time = 0;
    bool on_off_global_scene_control = 0;
    esp_zb_cluster_list_t *cluster_list = esp_zb_ep_list_get_ep(esp_zb_on_off_light_ep, HA_ESP_LIGHT_ENDPOINT);
    esp_zb_attribute_list_t *attr_list =
        esp_zb_cluster_list_get_cluster(cluster_list, ESP_ZB_ZCL_CLUSTER_ID_ON_OFF, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_on_off_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_ON_TIME, &on_off_on_time);
    esp_zb_on_off_cluster_add_attr(attr_list, ESP_ZB_ZCL_ATTR_ON_OFF_GLOBAL_SCENE_CONTROL,
                                   &on_off_global_scene_control);
```
More details for them can refer to the On/Off section of ZCL spec.

honusz, 2024-10-21:
@xieqinan That did the trick, thank you!

Reviewing the spec, I do see that I overlooked these indirect references in 3.8.2.3.4.3:

> On receipt of the Off With Effect command and if the GlobalSceneControl attribute is equal to TRUE, the application on the associated endpoint SHALL store its settings in its global scene then set the GlobalSceneControl attribute to FALSE. The application SHALL then enter its “off” state, update the OnOff attribute accordingly and set the OnTime attribute to 0x0000.

I'm curious though: Is `0x85 INVALID_FIELD` the appropriate response for this? When I see this, or `0x80 MALFORMED_COMMAND`, this indicates to me that the issue it that the issue is with _what_ the device is receiving, not that something isn't properly configured on the device. 
Would `0x81 UNSUP_COMMAND` or `0x86 UNSUPPORTED_ATTRIBUTE` not be a more appropriate response from the device?

Regardless, I'll close this issue as complete, with my thanks. If you think the response should be changed, I'm happy to open a new issue.

Thanks again!

closed 2024-10-21 by honusz