espressif/esp-zigbee-sdk #458: How can I correctly read/use the parameters in partition zb_fct written by esp_zb_mfg_tool.py tool?

josthlee, 2024-10-21:
### Question

According to the document [mfg_tool](https://github.com/espressif/esp-zigbee-sdk/blob/main/tools/mfg_tool/README.md):

1. Generate bin
```
python3 esp_zb_mfg_tool.py -i 966b9f3ef98ae6059708 -m 0BE961FEFFBD4D74 -c 0x8000 -mn XXXX -mc 0x1111

Created NVS binary: ===> /Workspace/esp-zigbee-sdk/tools/mfg_tool/bin/0BE961FEFFBD4D74.bin

Files generated in /Workspace/esp-zigbee-sdk/tools/mfg_tool/ ...
```
2. Flash memory
```
esptool.py -p  /dev/cu.usbmodem14214401 write_flash 0xf5000 bin/0BE961FEFFBD4D74.bin
esptool.py v4.8.dev5
Serial port /dev/cu.usbmodem14214401
Connecting...
Detecting chip type... ESP32-H2
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:61:e9:0b
BASE MAC: 74:4d:bd:61:e9:0b
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x000f5000 to 0x000f5fff...
Compressed 204 bytes to 58...
Wrote 204 bytes (58 compressed) at 0x000f5000 in 0.0 seconds (effective 47.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
3. Try to check if the zb_cft is correct
```
esptool.py -p /dev/cu.usbmodem14214401  read_flash 0xf5000 204 zb_fct.txt
esptool.py v4.8.dev5
Serial port /dev/cu.usbmodem14214401
Connecting...
Detecting chip type... ESP32-H2
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:61:e9:0b
BASE MAC: 74:4d:bd:61:e9:0b
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
204 (100 %)
204 (100 %)
Read 204 bytes at 0x000f5000 in 0.0 seconds (944.4 kbit/s)...
Hard resetting via RTS pin...

diff bin/0BE961FEFFBD4D74.bin zb_fct.txt 
Binary files bin/0BE961FEFFBD4D74.bin and zb_fct.txt differ
```

It is pretty straightforward and make the all things done. However, I try to esp_zb_secur_ic_get after esp_zb_init:

```
...
 esp_zb_init(&zb_nwk_cfg);
 esp_zb_secur_ic_type_t ic_type = ESP_ZB_IC_TYPE_64;
 uint8_t *ic = esp_zb_secur_ic_get((uint8_t *)&ic_type);
...
```

The returned ic (install code) pointer is still NULL.

My questions are:
1. Is the zb_fct I flashed will be loaded by esp_zb_init?
2. What is the correct way to load the parameters I flashed? Install code, manufacturer name and code etc. ...?
3. I know I can manually set install code (manually) on the fly as well as manufacturer name and code by set attr, but how can I do with the pre flashed data?

xieqinan, 2024-10-28:
@josthlee 

> Is the zb_fct I flashed will be loaded by esp_zb_init?

No really, the zb_fct will be loaded after the `esp_zb_start()`.

> What is the correct way to load the parameters I flashed? Install code, manufacturer name and code etc. ...?

There are two methods to flash device parameters. First, `zb_fct` provides a factory method to store basic device information in NVRAM. Second, the SDK offers APIs, such as `esp_zb_secur_ic_set()`, `esp_zb_set_node_descriptor_manufacturer_code()`, and `esp_zb_set_long_address()`, for writing device information. I recommend using the API method during the development stage.

> I know I can manually set install code (manually) on the fly as well as manufacturer name and code by set attr, but how can I do with the pre flashed data?

You can refer to [this link](https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073) for the install code setup. To set the manufacturer code, use `esp_zb_set_node_descriptor_manufacturer_code()`. 
As for the manufacturer name, if it's used at the device endpoint level, you might refer to the [zcl_utility example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/common/zcl_utility/src/zcl_utility.c#L22).

josthlee, 2024-10-30:
> @josthlee
> 
> > Is the zb_fct I flashed will be loaded by esp_zb_init?
> 
> No really, the zb_fct will be loaded after the `esp_zb_start()`. It is better to call the API `uint8_t *ic = esp_zb_secur_ic_get((uint8_t *)&ic_type);` in ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY signal.
> 
Noted and I will make a test.
> There are two methods to flash device parameters. First, `zb_fct` provides a factory method to store basic device information in NVRAM. Second, the SDK offers APIs, such as `esp_zb_secur_ic_set()`, [...]
I'v tried this before esp_zb_mfg_tool.py, it's working properly. Thank you!

josthlee, 2024-10-30:
As per your last reply, I just try to **esp_zb_secur_ic_get** the install code but failed still.

My partition table
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 900K,
zb_storage, data, fat,      0xf1000, 16K,
zb_fct,     data, fat,      0xf5000, 1K,
```
I try to generate the binary of zb_cft by:
```
python3 esp_zb_mfg_tool.py -i 966b9f3ef98ae6059708 -m 0BE961FEFFBD4D74 -c 0xf5000 -mn XXXX -mc 0x1234

Created NVS binary: ===> /esp-zigbee-sdk/tools/mfg_tool/bin/0BE961FEFFBD4D74.bin

Files generated in /esp-idf-v5.3/projects/esp-zigbee-sdk/tools/mfg_tool/ ...
```

And I dump the content of 0BE961FEFFBD4D74.bin,
```
od -x bin/0BE961FEFFBD4D74.bin 
0000000 37e7 f6dd 0370 1aaa 00c6 0002 5000 000f
0000020 0000 0000 0000 0000 0000 0000 0000 0000
0000040 4d74 ffbd 61fe 0be9 0000 0000 0000 0000
0000060 0000 0000 0000 0000 0000 0000 0000 0000
*
0000240 0000 0000 0000 0000 0000 0000 0000 0100
0000260 6b96 3e9f 8af9 05e6 0897 0000 0000 0000
0000300 0000 0001 1234 5858 5858
0000312
```

Then I flash the bin to the device,
```
esptool.py write_flash 0xf5000 bin/0BE961FEFFBD4D74.bin 
esptool.py v4.8.1
Found 34 serial ports
Serial port /dev/ttyS9
.... (SKIPPED SOME USELESS INFO)
Serial port /dev/ttyACM1
Connecting...
Detecting chip type... ESP32-H2
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:61:e9:0b
BASE MAC: 74:4d:bd:61:e9:0b
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x000f5000 to 0x000f5fff...
Compressed 204 bytes to 59...
Wrote 204 bytes (59 compressed) at 0x000f5000 in 0.0 seconds (effective 48.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
Then I try to read the content flash and write to file zb_cft.txt,
```
esptool.py read_flash 0xf5000 1024 zb_cft.txt
esptool.py v4.8.1
Found 34 serial ports
Serial port /dev/ttyS9
/dev/ttyS9 failed to connect: Could not open /dev/ttyS9, the port is busy or doesn't exist.
(Could not configure port: (5, 'Input/output error'))
.... (SKIPPED SOME USELESS INFO)
Serial port /dev/ttyACM1
Connecting...
Detecting chip type... ESP32-H2
Chip is ESP32-H2 (revision v0.1)
Features: BLE, IEEE802.15.4
Crystal is 32MHz
MAC: 74:4d:bd:ff:fe:61:e9:0b
BASE MAC: 74:4d:bd:61:e9:0b
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
1024 (100 %)
1024 (100 %)
Read 1024 bytes at 0x000f5000 in 0.0 seconds (1275.3 kbit/s)...
Hard resetting via RTS pin...
```
And I dump the content of zb_fct.txt,
```
od -x zb_cft.txt 
0000000 37e7 f6dd 0370 1aaa 00c6 0002 5000 000f
0000020 0000 0000 0000 0000 0000 0000 0000 0000
0000040 4d74 ffbd 61fe 0be9 0000 0000 0000 0000
0000060 0000 0000 0000 0000 0000 0000 0000 0000
*
0000240 0000 0000 0000 0000 0000 0000 0000 0100
0000260 6b96 3e9f 8af9 05e6 0897 0000 0000 0000
0000300 0000 0001 1234 5858 5858 ffff ffff ffff
0000320 ffff ffff ffff ffff ffff ffff ffff ffff
*
0002000
```
I think It's pretty much the same with the 0BE961FEFFBD4D74.bin I write_flash. However, I still cannot  **esp_zb_secur_ic_get** in ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY signal. The below is my code snippet,

```
...
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_PRODUCTION_CONFIG_READY:
        ESP_LOGW(TAG, "Zigbee production config ready");

        esp_zb_secur_ic_type_t ic_type = ESP_ZB_IC_TYPE_64;
        uint8_t *ic = esp_zb_secur_ic_get((uint8_t *)&ic_type);
        if ( ic == NULL ) {
            ESP_LOGW(TAG, "Install code size none");
        }
        else {
            ESP_LOGW(TAG, "Install code size %d",sizeof(ic));        
        }
        
        break;
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
...
```
And the result is cropped by the image below,

[image]

I think it's quite a simple thing but failed. Did I miss something and what's wrong with my work?

xieqinan, 2024-10-31:
@josthlee ,

Does the `ESP_ZB_ON_OFF_SWITCH` device function as the coordinator? Setting the install code for a coordinator has no effect, as the stack will ignore install code settings if the device is acting as the coordinator.

josthlee, 2024-10-31:
> @josthlee ,
> 
> Does the `ESP_ZB_ON_OFF_SWITCH` device function as the coordinator? Setting the install code for a coordinator has no effect, as the stack will ignore install code settings if the device is acting as the coordinator.

Got it, and I did it on ZC side. I just try to do it on ED side and it works perfectly. Thank you so much!

closed 2024-10-31 by josthlee