espressif/esp-zigbee-sdk #460: How to add scene functions in HA_color_dimmable_light ?

Bosemani, 2024-10-22:
### Question

Hai,
    I want to include the scene functionalities in the HA_color_dimmable_light example.
From the coordinator I'm able to create and execute Scene.
But From the HA_color_dimmable_light example there is no functions to handle the scene.
I'm getting the callback on "zb_action_handler" 
for ESP_ZB_CORE_SCENES_STORE_SCENE_CB_ID and ESP_ZB_CORE_SCENES_RECALL_SCENE_CB_ID.

how to implement function to handle this event?
Any example functions available?

regards,
Bose

Haerteleric, 2024-12-18:
faceing the same issue. But i implemented the light via:

`esp_zb_color_control_cluster_create`
`esp_zb_level_cluster_create`
`esp_zb_on_off_cluster_create`
`esp_zb_scenes_cluster_create`

AND
`esp_zb_ep_list_add_ep`

xieqinan, 2025-05-30:
@Bosemani ,

Please refer to [zcl scene docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_scene.html), I believe it can explain your question. Please feel free to reopen, if it's still unclear.

closed 2025-05-30 by xieqinan