espressif/esp-zigbee-sdk #461: ZDO Fail

script0803, 2024-10-22:
### Question

When using the example: customized client,

 I encountered two incorrect ZDO statuses, 'ESP_ZB_ZDP_STATUS_INSUFFICIENT_SPACE' and 'ESP_ZB_ZDP_STATUS_BINDING_TABLE_FULL'.

 Do I need to adjust any settings? 

The corresponding log output is as follows:
I (349) main_task: Started on CPU0
I (349) main_task: Calling app_main()
I (359) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (399) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
I (409) main_task: Returned from app_main()
I (419) ESP_HA_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (699) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:4 
I (699) ESP_HA_ON_OFF_SWITCH: Deferred driver initialization successful
I (709) ESP_HA_ON_OFF_SWITCH: Joined network successfully (Extended PAN ID: dd:dd:dd:dd:dd:dd:dd:dd, PAN ID: 0x1a62, Channel:11, Short Address: 0xd41b)
I (749) ESP_HA_ON_OFF_SWITCH: Match desc response: status(0), address(0x0), endpoint(1)
I (819) ESP_HA_ON_OFF_SWITCH: Active endpoint response: status(0) and endpoint count(2)
I (819) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 1
I (819) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 242
I (849) ESP_HA_ON_OFF_SWITCH: 111138
I (879) ESP_HA_ON_OFF_SWITCH: IEEE address: e0:79:8d:ff:fe:bf:0f:af
I (939) ESP_HA_ON_OFF_SWITCH: 140

script0803, 2024-10-22:
Partial code：
```c
static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    ESP_LOGI(TAG, "%d",zdo_status);
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Bind response from address(0x%x), endpoint(%d) with status(%d)", ((zdo_info_user_ctx_t *)user_ctx)->short_addr,
                 ((zdo_info_user_ctx_t *)user_ctx)->endpoint, zdo_status);

        /* configure report attribute command */
        esp_zb_zcl_config_report_cmd_t report_cmd;
        bool report_change = 0;
        report_cmd.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        report_cmd.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        report_cmd.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;

        esp_zb_zcl_config_report_record_t records[] = {
            {ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV, ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID, ESP_ZB_ZCL_ATTR_TYPE_BOOL, 0, 30, &report_change}};
        report_cmd.record_number = sizeof(records) / sizeof(esp_zb_zcl_config_report_record_t);
        report_cmd.record_field = records;

        /* ec report init */
        esp_zb_zcl_config_report_cmd_t report_ec_cmd;
        int16_t report_change_ec = 0;
        report_ec_cmd.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        report_ec_cmd.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        report_ec_cmd.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        report_ec_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_ec_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;

        esp_zb_zcl_config_report_record_t ec_records[] = {
            {
                .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
                .attributeID = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMSVOLTAGE_ID,
                .attrType = ESP_ZB_ZCL_ATTR_TYPE_U16,
                .min_interval = 0,
                .max_interval = 5,
                .reportable_change = &report_change_ec,
            },
            {
                .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
                .attributeID = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMSCURRENT_ID,
                .attrType = ESP_ZB_ZCL_ATTR_TYPE_U16,
                .min_interval = 0,
                .max_interval = 5,
                .reportable_change = &report_change_ec,
            },
            {
                .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
                .attributeID = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_ACTIVE_POWER_ID,
                .attrType = ESP_ZB_ZCL_ATTR_TYPE_S16,
                .min_interval = 0,
                .max_interval = 5,
                .reportable_change = &report_change_ec,
            }};
        report_ec_cmd.record_number = sizeof(ec_records) / sizeof(esp_zb_zcl_config_report_record_t);
        report_ec_cmd.record_field = ec_records;

        /* meter report init */
        esp_zb_zcl_config_report_cmd_t report_meter_cmd;
        int16_t report_change_meter = 1;
        report_ec_cmd.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        report_ec_cmd.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        report_ec_cmd.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        report_ec_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        report_ec_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_METERING;

        esp_zb_zcl_config_report_record_t meter_records[] = {
[... 81 lines cut ...]
        esp_zb_zdo_active_ep_req(&active_ep_req, ep_cb, NULL);
        /* get the node simple descriptor */
        esp_zb_zdo_simple_desc_req_param_t simple_desc_req;
        simple_desc_req.addr_of_interest = addr;
        simple_desc_req.endpoint = endpoint;
        esp_zb_zdo_simple_desc_req(&simple_desc_req, simple_desc_cb, NULL);
        /* get the light ieee address */
        esp_zb_zdo_ieee_addr_req_param_t ieee_req;
        ieee_req.addr_of_interest = on_off_light.short_addr;
        ieee_req.dst_nwk_addr = on_off_light.short_addr;
        ieee_req.request_type = 0;
        ieee_req.start_index = 0;
        esp_zb_zdo_ieee_addr_req(&ieee_req, ieee_cb, NULL);
        esp_zb_zcl_read_attr_cmd_t read_req;
        uint16_t attributes[] = {ESP_ZB_ZCL_ATTR_ON_OFF_ON_OFF_ID};
        read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        read_req.attr_number = sizeof(attributes) / sizeof(uint16_t);
        read_req.attr_field = attributes;
        read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
        read_req.zcl_basic_cmd.dst_endpoint = on_off_light.endpoint;
        read_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        read_req.zcl_basic_cmd.dst_addr_u.addr_short = on_off_light.short_addr;
        esp_zb_zcl_read_attr_cmd_req(&read_req);
    }
}

xieqinan, 2024-10-22:
@script0803 ,

I think if you use the original code, it can work as expected.

> I encountered two incorrect ZDO statuses, 'ESP_ZB_ZDP_STATUS_INSUFFICIENT_SPACE' and 'ESP_ZB_ZDP_STATUS_BINDING_TABLE_FULL'.

Could you please provide more details on where the two incorrect statuses are being triggered? More specific context will help in identifying and resolving the issue effectively.

script0803, 2024-10-22:
@xieqinan ，

I tried using the original code and got the same error.

'ESP_ZB_ZDP_STATUS_INSUFFICIENT_SPACE' is triggered in simple_desc_cb().

'ESP_ZB_ZDP_STATUS_BINDING_TABLE_FULL' is triggered in bind_cb().

xieqinan, 2024-10-28:
Hi @script0803,
The bind table size is 32 by default. I'm not sure what operations on your side caused it to fill up. Could you please provide a `.pcap` file to help us address this issue?

script0803, 2024-10-28:
Hi @xieqinan ,
Of course, but GitHub does not seem to support uploading pcap type files

xieqinan, 2024-10-28:
@script0803 ,

You can use the `zip` compress .pcap file first, then upload it.

script0803, 2024-10-28:
@xieqinan 
Thanks for your help！
[log.zip](https://github.com/user-attachments/files/17539488/log.zip)

xieqinan, 2024-10-28:
Hi @script0803 ,

The `.pcap` file shows that the `Insufficient Space` status originates from the coordinator’s response. However, the coordinator is not an ESPRESSIF device. Could you please double-check the number of clusters on the coordinator's endpoint 1?

[image]

The bind request as the same.

[image]

script0803, 2024-10-29:
Hi @xieqinan ，

I use Zigbee2MQTT for debugging and use Sonoff Dongle-E as coordinator. You can see that I only connect the ESP32H2 development board.

[image]

I noticed that the error was insufficient bind table space, but I tried to adjust the bind table, and the results still showed these two errors.

ESP_ERROR_CHECK(esp_zb_aps_src_binding_table_size_set(64));
ESP_ERROR_CHECK(esp_zb_aps_dst_binding_table_size_set(64))

The following is the cluster I use。
[image]

The following is my log after startup and network access. 138 and 140 are the two errors we just pointed out. I debugged them and asked them to output the errors.

I (23) boot: ESP-IDF v5.4-dev-3489-g6e5a178b31 2nd stage bootloader
I (23) boot: compile time Oct 25 2024 10:50:08
I (25) boot: chip revision: v0.1
I (27) boot: efuse block revision: v0.3
I (32) boot.esp32h2: SPI Speed      : 64MHz
I (37) boot.esp32h2: SPI Mode       : DIO
I (41) boot.esp32h2: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 000e1000
I (85) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (92) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=104b0h ( 66736) map
I (129) esp_image: segment 1: paddr=000204d8 vaddr=40800000 size=07b40h ( 31552) load
I (140) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=603c8h (394184) map
I (239) esp_image: segment 3: paddr=000883f0 vaddr=40807b40 size=05e88h ( 24200) load
I (248) esp_image: segment 4: paddr=0008e280 vaddr=4080d9d0 size=01c98h (  7320) load
I (255) boot: Loaded app from partition at offset 0x10000
I (256) boot: Disabling RNG early entropy source...
I (269) cpu_start: Unicore app
I (278) cpu_start: Pro cpu start user code
I (278) cpu_start: cpu freq: 96000000 Hz
I (278) app_init: Application information:
I (280) app_init: Project name:     HA_customized_switch
I (286) app_init: App version:      238b4a9-dirty
I (292) app_init: Compile time:     Oct 25 2024 10:50:03
I (298) app_init: ELF file SHA256:  a1e106b29...
I (303) app_init: ESP-IDF:          v5.4-dev-3489-g6e5a178b31
I (310) efuse_init: Min chip rev:     v0.0
I (314) efuse_init: Max chip rev:     v0.99 
I (319) efuse_init: Chip rev:         v0.1
I (324) heap_init: Initializing. RAM available for dynamic allocation:
I (331) heap_init: At 40813E40 len 00039540 (229 KiB): RAM
I (337) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (345) spi_flash: detected chip: generic
I (348) spi_flash: flash io: dio
W (352) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (366) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (373) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (380) main_task: Started on CPU0
I (380) main_task: Calling app_main()
I (410) phy: phy_version: 310,2, 463a420, Aug 15 2024, 18:52:11
I (420) phy: libbtbb version: a024da6, Aug 15 2024, 18:52:23
I (430) main_task: Returned from app_main()
I (440) ESP_HA_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (730) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:4 
I (730) ESP_HA_ON_OFF_SWITCH: Deferred driver initialization successful
I (730) ESP_HA_ON_OFF_SWITCH: Joined network successfully (Extended PAN ID: dd:dd:dd:dd:dd:dd:dd:dd, PAN ID: 0x1a62, Channel:11, Short Address: 0x2696)
I (780) ESP_HA_ON_OFF_SWITCH: Match desc response: status(0), address(0x0), endpoint(1)
I (860) ESP_HA_ON_OFF_SWITCH: Active endpoint response: status(0) and endpoint count(2)
I (860) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 1
I (860) ESP_HA_ON_OFF_SWITCH: Endpoint ID List: 242
I (890) ESP_HA_ON_OFF_SWITCH: 111138
I (930) ESP_HA_ON_OFF_SWITCH: IEEE address: e0:79:8d:ff:fe:bf:0f:af
I (990) ESP_HA_ON_OFF_SWITCH: 140
W (1530) ESP_HA_ON_OFF_SWITCH: Receive Zigbee action(0x1005) callback

script0803, 2024-10-29:
Since you just said that endpoint 1 may be bound to too many clusters, I have now changed it to 11, but both errors still exist.

xieqinan, 2024-10-29:
Hi @script0803 ,

> Since you just said that endpoint 1 may be bound to too many clusters, I have now changed it to 11, but both errors still exist.

It seems there may be a misunderstanding. Based on [the reference](https://github.com/espressif/esp-zigbee-sdk/issues/461#issuecomment-2440892895), the ESP device with the short address `0x2dbc` is sending a bind request to the coordinator (`0x0000`). The coordinator responds to `0x2dbc` with a `TABLE FULL` status, indicating that there is no available space in the coordinator’s binding table to store the binding information for `0x2dbc`.  This issue is not related to `0x2dbc` itself; the solution SHOULD be to increase the binding table size on the coordinator (Sonoff Dongle-E).

script0803, 2024-10-29:
Hi @xieqinan ，

But I think the sonoff dongle-e is a very common dongle for connecting zigbee3.0 devices. I tried other zigbee devices and they were all normal. Why is there insufficient space under the esp module? And I haven't bound many clusters. This is very strange.

xieqinan, 2024-10-29:
Have other Zigbee devices attempted to bind their information to the coordinator?

The `esp_zigbee_customized_device` example demonstrates the attribute reporting process, so the `switch` example needs to bind its information to the `light` for the `light` to report its attributes. If you don’t have this requirement, consider using a more general example to join the coordinator, such as the [on/off light example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light).

script0803, 2024-10-29:
Hi @xieqinan ，
I will be using electrical measurement and metering cluster. Which example do you recommend I use to start?

xieqinan, 2024-10-29:
I think it is bettter to refer to the [temperature sensor example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c) and [thermostat example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_thermostat).

script0803, 2024-11-04:
Hi @xieqinan ,

I am making some modifications based on the temperature sensor example. 

I can now use zigbee2mqtt to see the cluster and attribute I registered normally. 

I used an asynchronous function to simulate the data update. But the value of the attribute I read in zigbee2mqtt is always 0. 

The following is my code. Can you help me see where the problem is? Thank you very much!

```c
/*
 * SPDX-FileCopyrightText: 2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee HA_temperature_sensor Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include "esp_zb_temperature_sensor.h"
#include "temp_sensor_driver.h"
#include "switch_driver.h"

#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

#include "time.h"
#include "stdlib.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile sensor (End Device) source code.
#endif

static const char *TAG = "ESP_ZB_TEMP_SENSOR";

typedef struct {
    uint32_t low;  // low 32
    uint16_t high; // high 16
} uint48_t;

uint48_t convertToUint48(double value) {
    uint64_t intValue = (uint64_t)(value * 1000);
    uint48_t result;
    result.low = (uint32_t)(intValue & 0xFFFFFFFF);
    result.high = (uint16_t)((intValue >> 32) & 0xFFFF);
    return result;
}

uint48_t currentSummDelivered;
uint48_t currentSummReceived;
uint16_t voltage;
uint16_t current;
int16_t activePower;

void update_attributes() {
    // random value
    voltage = ((double)rand() / RAND_MAX) * 12000.0 + 12000.0;
    current = ((double)rand() / RAND_MAX) * 10000.0 + 1.0;
    activePower = ((double)rand() / RAND_MAX) * 10000.0;
    double newDeliveredValue = ((double)rand() / RAND_MAX) * 10.0;
    double newReceivedValue = ((double)rand() / RAND_MAX) * 5.0;
[... 180 lines cut ...]
        .manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC,
    };

    esp_zb_zcl_update_reporting_info(&reporting_info);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));

    esp_zb_stack_main_loop();
}

void app_main(void)
{
    srand(time(NULL));
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    /* Start Zigbee stack task */
    xTaskCreate(esp_zb_task, "Zigbee_main", 8192, NULL, 5, NULL);
    xTaskCreate(update_task, "UpdateTask", 4096, NULL, 5, NULL);
}

xieqinan, 2024-11-04:
Hi @script0803 ,

It seems there’s a misunderstanding regarding the `esp_zb_xxx_cluster_add_attr()` API. This API allows users to add an attribute to a cluster and initialize its value, but the attribute is deep-copied into the stack. Therefore, updating `voltage` in `update_attributes()` will not sync with the attribute in the stack.

You might try using `esp_zb_zcl_set_attribute_val()` within `update_attributes()` to synchronize the `ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMSVOLTAGE_ID` attribute with the stack.

By the way, there is having a error in `esp_app_buttons_handler()`, correct it.
```
static void esp_app_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* Send report attributes command */
        esp_zb_zcl_report_attr_cmd_t report_attr_cmd;
        report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMSVOLTAGE_ID;
        report_attr_cmd.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
        report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
        report_attr_cmd.zcl_basic_cmd.src_endpoint = HA_ESP_SENSOR_ENDPOINT;

        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd);
        esp_zb_lock_release();
        ESP_EARLY_LOGI(TAG, "Send 'report attributes' command");
    }
}
```

script0803, 2024-11-05:
This works, thank you very much！

closed 2024-11-14 by script0803