espressif/esp-zigbee-sdk #463: Unable to setup power config reporting for battery

alexbegoon, 2024-10-22:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### esp-zigbee-lib version.

v1.5.1

### esp-zboss-lib version.

v1.5.1

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

I am trying to setup reporting for battery voltage, remaining percentage and alerting.
```c
// Create reporting info for battery
esp_zb_zcl_reporting_info_t battery_reporting_info;

memset(&battery_reporting_info, 0, sizeof(battery_reporting_info));
battery_reporting_info.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
battery_reporting_info.ep = HA_ESP_SENSOR_ENDPOINT;
battery_reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG;
battery_reporting_info.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
battery_reporting_info.attr_id = ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID;
battery_reporting_info.u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.max_interval = 0;
battery_reporting_info.u.send_info.delta.u8 = 0x00;
battery_reporting_info.u.send_info.reported_value.u8 = 0;
battery_reporting_info.u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&battery_reporting_info));
```

### What is the actual behavior?

I am getting error
```
ZBOSS: common/zb_init_default.c:183   FFD build
ZBOSS: common/zb_init_default.c:187   sizes: g_zb 7720 sched 224 bpool 16 nwk 1032 aps 3048 addr 284 zdo 840
ZBOSS: common/zb_init_default.c:196   sec 2
ZBOSS: common/zb_init_default.c:198   zcl 768
ZBOSS: common/zb_init_default.c:201   zll 960
ZBOSS: common/zb_init_default.c:204   zgp 2896
ZBOSS: common/zb_init_default.c:207   nvram 140
ZBOSS: common/zb_init_default.c:210   buttons 200
ZBOSS: common/zb_init_default.c:212   err_ind 4
ZBOSS: common/zb_init_default.c:218   scheduler q size 80
ZBOSS: common/zb_init_default.c:221   g_mac 816 g_imac 28
ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 64 ZB_IEEE_ADDR_TABLE_SIZE 80 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 64 ZB_APS_DUPS_TABLE_SIZE 32
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
ZBOSS: zcl/zcl_reporting.c:305   Error, attribute not found
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4200ead4
```

### Steps to reproduce.

1. Create power config cluster. That part works just well, after pairing, I am getting these values just right.
```c
esp_zb_attribute_list_t *power_config_cluster = esp_zb_power_config_cluster_create(&power_cfg);
    ESP_ERROR_CHECK(
            esp_zb_power_config_cluster_add_attr(power_config_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID,
                                                 &battery_voltage_measured));
    ESP_ERROR_CHECK(
            esp_zb_power_config_cluster_add_attr(power_config_cluster,
                                                 ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_PERCENTAGE_REMAINING_ID,
                                                 &battery_percentage));
    ESP_ERROR_CHECK(
            esp_zb_power_config_cluster_add_attr(power_config_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_QUANTITY_ID,
                                                 &battery_quantity));
    ESP_ERROR_CHECK(
            esp_zb_power_config_cluster_add_attr(power_config_cluster, ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_SIZE_ID,
                                                 &battery_size));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_power_config_cluster(cluster_list, power_config_cluster,
                                                                 ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
```
2. Then update reporting
```c
// Create reporting info for battery
esp_zb_zcl_reporting_info_t battery_reporting_info;

memset(&battery_reporting_info, 0, sizeof(battery_reporting_info));
battery_reporting_info.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
battery_reporting_info.ep = HA_ESP_SENSOR_ENDPOINT;
battery_reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG;
battery_reporting_info.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
battery_reporting_info.attr_id = ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID;
battery_reporting_info.u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.max_interval = 0;
battery_reporting_info.u.send_info.delta.u8 = 0x00;
battery_reporting_info.u.send_info.reported_value.u8 = 0;
battery_reporting_info.u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&battery_reporting_info));

memset(&battery_reporting_info, 0, sizeof(battery_reporting_info));
battery_reporting_info.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
battery_reporting_info.ep = HA_ESP_SENSOR_ENDPOINT;
battery_reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG;
battery_reporting_info.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
battery_reporting_info.attr_id = ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_PERCENTAGE_REMAINING_ID;
battery_reporting_info.u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
battery_reporting_info.u.send_info.delta.u8 = 0x00;
battery_reporting_info.u.send_info.reported_value.u8 = 0;
battery_reporting_info.u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&battery_reporting_info));

memset(&battery_reporting_info, 0, sizeof(battery_reporting_info));
battery_reporting_info.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
battery_reporting_info.ep = HA_ESP_SENSOR_ENDPOINT;
battery_reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG;
battery_reporting_info.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
battery_reporting_info.attr_id = ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_ALARM_STATE_ID;
battery_reporting_info.u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
battery_reporting_info.u.send_info.delta.u8 = 0x00;
battery_reporting_info.u.send_info.reported_value.u32 = 0;
battery_reporting_info.u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&battery_reporting_info));
```

xieqinan, 2024-10-23:
@alexbegoon ,

Do you have try to call the `esp_zb_zcl_update_reporting_info()` after `esp_zb_device_register()`?

alexbegoon, 2024-10-23:
> @alexbegoon ,
> 
> Do you have try to call the `esp_zb_zcl_update_reporting_info()` after `esp_zb_device_register()`?

@xieqinan hi, this is how I am doing that. In fact, this part works just fine. I am receiving these measurements periodically:

```c
/* Register the device */
esp_zb_device_register(esp_zb_sensor_ep);

// Create reporting info for temperature
esp_zb_zcl_reporting_info_t temperature_reporting_info = {
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
        .ep = HA_ESP_SENSOR_ENDPOINT,
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        .u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.delta.u16 = 10,
        .attr_id = ESP_ZB_ZCL_ATTR_TEMP_MEASUREMENT_VALUE_ID,
        .manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC,
};
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&temperature_reporting_info));

// Create reporting info for humidity
esp_zb_zcl_reporting_info_t humidity_reporting_info = {
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
        .ep = HA_ESP_SENSOR_ENDPOINT,
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_REL_HUMIDITY_MEASUREMENT,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        .u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.delta.u16 = 10,
        .attr_id = ESP_ZB_ZCL_ATTR_REL_HUMIDITY_MEASUREMENT_VALUE_ID,
        .manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC,
};
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&humidity_reporting_info));

// Create reporting info for pressure
esp_zb_zcl_reporting_info_t pressure_reporting_info = {
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
        .ep = HA_ESP_SENSOR_ENDPOINT,
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_PRESSURE_MEASUREMENT,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        .u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC,
        .u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2,
        .u.send_info.delta.u16 = 1,
        .attr_id = ESP_ZB_ZCL_ATTR_PRESSURE_MEASUREMENT_VALUE_ID,
        .manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC,
};
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&pressure_reporting_info));
```

Then, attempting to update the reporting for battery end up with:
```c
// Create reporting info for battery
esp_zb_zcl_reporting_info_t battery_reporting_info;

memset(&battery_reporting_info, 0, sizeof(battery_reporting_info));
battery_reporting_info.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
battery_reporting_info.ep = HA_ESP_SENSOR_ENDPOINT;
battery_reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_POWER_CONFIG;
battery_reporting_info.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE;
battery_reporting_info.attr_id = ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID;
battery_reporting_info.u.send_info.min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.max_interval = 0;
battery_reporting_info.u.send_info.delta.u8 = 0x00;
battery_reporting_info.u.send_info.reported_value.u8 = 0;
battery_reporting_info.u.send_info.def_min_interval = CONFIG_WAKEUP_TIME_SEC;
battery_reporting_info.u.send_info.def_max_interval = CONFIG_WAKEUP_TIME_SEC * 2;
battery_reporting_info.dst.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
battery_reporting_info.manuf_code = ESP_ZB_ZCL_ATTR_NON_MANUFACTURER_SPECIFIC;
ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&battery_reporting_info));
```

```text
ZBOSS: common/zb_init_default.c:183   FFD build
ZBOSS: common/zb_init_default.c:187   sizes: g_zb 7720 sched 224 bpool 16 nwk 1032 aps 3048 addr 284 zdo 840
ZBOSS: common/zb_init_default.c:196   sec 2
ZBOSS: common/zb_init_default.c:198   zcl 768
ZBOSS: common/zb_init_default.c:201   zll 960
ZBOSS: common/zb_init_default.c:204   zgp 2896
ZBOSS: common/zb_init_default.c:207   nvram 140
ZBOSS: common/zb_init_default.c:210   buttons 200
ZBOSS: common/zb_init_default.c:212   err_ind 4
ZBOSS: common/zb_init_default.c:218   scheduler q size 80
ZBOSS: common/zb_init_default.c:221   g_mac 816 g_imac 28
ZBOSS: common/zb_init_default.c:225   Configurable mem build, use ZBOSS lib defaults = 0
ZBOSS: common/zb_init_default.c:231   ZB_IOBUF_POOL_SIZE 80 ZB_NWK_IN_Q_SIZE 40 ZB_MAC_PENDING_QUEUE_SIZE 20 ZB_APS_DST_BINDING_TABLE_SIZE 16 ZB_APS_BIND_TRANS_TABLE_SIZE 20 ZB_N_APS_RETRANS_ENTRIES 27
ZBOSS: common/zb_init_default.c:261   ZB_N_APS_KEY_PAIR_ARR_MAX_SIZE 64 ZB_IEEE_ADDR_TABLE_SIZE 80 ZB_NEIGHBOR_TABLE_SIZE 64 ZB_NWK_ROUTING_TABLE_SIZE 64 ZB_APS_DUPS_TABLE_SIZE 32
ZBOSS: zdo/af_descriptor.c:357   ep_count 1
ZBOSS: zcl/zcl_reporting.c:299   Error, no free slots for reporting info
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4200eaee
```

xieqinan, 2024-10-23:
@alexbegoon ,

The `ESP_ZB_ZCL_ATTR_POWER_CONFIG_BATTERY_VOLTAGE_ID(0x0020)` is not a reportable attribute, you can refer to below figure.
[image]

alexbegoon, 2024-10-23:
Ah, that make sense. Thanks. Closing this one.

closed 2024-10-23 by alexbegoon