espressif/esp-zigbee-sdk #464: HA with an ESP32 Zigbee switch and light end device won't find short address.

djorr5, 2024-10-27:
### Question

Hi, anyone had any experience with Homeassistant and ESP32 end devices trying to bind to each other? I have a HA setup with a Sonoff coordinator running. I used the ESP32 light example and it will join the network and can be controlled no problems. But when I went to bind the light to the ESP32 switch example the `esp_zb_zdo_find_on_off_light` function always just returns a 0x0000 address. If I manually enter in the known short address of the light everything works perfectly. But if I use a global search, like:

`            esp_zb_zdo_match_desc_req_param_t  find_req;
            find_req.addr_of_interest = 0x0000;
            find_req.dst_nwk_addr = 0x0000;
            /* find the match on-off light device */
            esp_zb_zdo_find_on_off_light(&find_req, user_find_cb, NULL);`

I only ever get back the cooridinator address of 0x0000

`I (8739) ESP_ZB_ON_OFF_SWITCH: Match desc response: status(0), address(0x0), endpoint(1)`

I was under the assumption that an addr_of_interest and dst_nwk_addr of 0 would send to the coordinator to find any associated end device on a network. As per the esp_zigbee_customized_devices/customized_client. Or am I incorrect in this assumption.

closed 2024-10-27 by djorr5

djorr5, 2024-10-27:
I jumped the gun. Sorry. 0xffff is what I should have been using.

xieqinan, 2024-10-28:
@djorr5 

The `esp_zb_zdo_find_on_off_light()` allows you to locate the on/off cluster on the `addr_of_interest` device within the network. If `dst_nwk_addr` is set to `0xffff`, all RxWhenIdle devices will receive the command, and only the device with the short address matching `addr_of_interest` will respond.

djorr5, 2024-10-28:
Interesting. I was just mucking around with it and with this struct:

`            find_req.addr_of_interest = 0xffff;
            find_req.dst_nwk_addr = 0xffff;`

The `esp_zb_zdo_find_on_off_light()` function actually finds the light end device and succesfully binds. As I only have 1 currently on the network I didn't worry about iterating or seeing if it discovers more than 1 end device this way. But are you saying that because the addr_of_interest is 0xffff it shouldn't find any end devices?

xieqinan, 2024-10-28:
> But are you saying that because the addr_of_interest is 0xffff it shouldn't find any end devices?

Yes, I think so.

djorr5, 2024-10-28:
Well it does find the light example with addr_of_interest = 0xffff