espressif/esp-zigbee-sdk #468: Zigbee Host example components folder?

Daniel-Craciun, 2024-11-01:
### Question

Hello, I've been looking through the Zigbee examples and I've noticed that the host example (which I hope to use) has a components folder with what it looks like some host version of Zigbee? And I was wondering why is this special version used for the host example. As I would like to use this example with the other types of clusters Zigbee has to offer such as the temperature and humidity clusters but the Zigbee host specific code doesn't provide these clusters and only has the on off switch cluster? What is the reason for this? Why can't this example just use the Zigbee sdk which I include via idf_component.yml and not the components folder?

lhespress, 2024-11-04:
@Daniel-Craciun 

> * host version of Zigbee

- It's just the host component version and is not [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) related.

> * the Zigbee host specific code doesn't provide these clusters and only has the on off switch cluster

- Yes, on/off switch cluster just as an example for reference, the customer need develop other clusters by themselves.

> * Why can't this example just use the Zigbee sdk which I include via idf_component.yml and not the components folder?

- This component will not release on [the ESP Component Registry](https://components.espressif.com/) because of the customer can develop the function on third-party MCU.

BTW, The host example just as a reference for the third-party SOC, it's better to use [esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk) on Espressif's SOC.

Daniel-Craciun, 2024-11-28:
Okay thank you!

closed 2024-11-28 by Daniel-Craciun