espressif/esp-zigbee-sdk #469: Customized server example request Aqara water leak sensor after joining network few seconds

vanbuong, 2024-11-05:
### Question

I have 1 ESP32-H2 module running  customized_server example. I add one more client cluster for Aqara sensor as following:
`esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);`
 **`esp_zb_ias_zone_cluster_cfg_t ias_zone_cfg = {0};`**
 **`esp_zb_cluster_list_add_ias_zone_cluster(esp_zb_cluster_list, esp_zb_ias_zone_cluster_create(&ias_zone_cfg), ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);`**
 `esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();`
 
 Then, I tried to pair the sensor to the network but it automatically left the network after 15 seconds, the log looks like this:
 I (403) sleep: Configure to isolate all GPIO pins in sleep state
I (409) sleep: Enable automatic switching of GPIO sleep configuration
I (417) main_task: Started on CPU0
I (417) main_task: Calling app_main()
I (457) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (457) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (467) main_task: Returned from app_main()
I (477) ESP_HA_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (477) ESP_HA_ON_OFF_LIGHT: Initialize Zigbee stack channel mask 7fff800 channel 15
W (487) ESP_HA_ON_OFF_LIGHT: Network(0x1c16) closed, devices joining not allowed.
W (497) rmt: channel resolution loss, real=10666666
I (497) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (497) ESP_HA_ON_OFF_LIGHT: Deferred driver initialization successful
I (507) ESP_HA_ON_OFF_LIGHT: Device started up in non factory-reset mode
I (517) ESP_HA_ON_OFF_LIGHT: Device rebooted
I (1127) ESP_HA_ON_OFF_LIGHT: Network(0x1c16) is open for 180 seconds
I (15207) ESP_HA_ON_OFF_LIGHT: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (15207) ESP_HA_ON_OFF_LIGHT: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (15487) ESP_HA_ON_OFF_LIGHT: New device commissioned or rejoined (short: 0x0696)
I (30217) ESP_HA_ON_OFF_LIGHT: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (31907) ESP_HA_ON_OFF_LIGHT: Zigbee Node is leaving network: 00:15:8d:00:0a:f2:c2:8e)
I (31927) ESP_HA_ON_OFF_LIGHT: Zigbee Node is leaving network: 00:15:8d:00:0a:f2:c2:8e)
W (181137) ESP_HA_ON_OFF_LIGHT: Network(0x1c16) closed, devices joining not allowed.

Please check the Wireshark captured file for more detailed. It started from this line:
[image]

[aqara_water_leak_6.zip](https://github.com/user-attachments/files/17631054/aqara_water_leak_6.zip)

xieqinan, 2024-11-05:
@vanbuong ,

Could you please also attach the network key of `.pcap` file?

vanbuong, 2024-11-05:
It's using Zigbee Alliance 09 key: 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39

xieqinan, 2024-11-05:
I need the network key to parse the `,pcap` file, you can use the `esp_zb_secur_primary_network_key_get()` to print it.

vanbuong, 2024-11-05:
Oh, I though you can find the network key in the .pcap file. Here is the key: bc7874de8a3829a3709d55108c70b92c
Same value when I print it with function you provided
[image]

vanbuong, 2024-11-06:
I read other threads and saw that my Aqara devices do not support child aging. It causes devices to be kicked off the network.
Line 830 of https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/firmware.patch

But when I run with https://github.com/andryblack/esp-coordinator firmware and intergrate with Zigbee2mqtt, I didn't see that issue. Seems Aqara device needs special steps to initialize or ZBOSS have API to disable child aging??? I don't know actually

vanbuong, 2024-11-08:
I just found a related ticket that mentioned this issue. https://github.com/espressif/esp-zigbee-sdk/issues/21#issuecomment-1564291634

closed 2024-11-08 by vanbuong