espressif/esp-zigbee-sdk #470: Zigbee Aps API

SUHAN447, 2024-11-06:
### Question

Hello,

I am very new to Zigbee and not quite sure what to do. Honestly, what I need is a code that sends a string from a coordinator device to an end device and vice versa, from the end device to the coordinator. Based on my needs, I came across APS, but I couldn't find a general example of how it is used, and as a result, all my efforts have been fruitless. Could you possibly recommend another API or example that might suit my needs?

Apologies if I made any mistakes.

xieqinan, 2024-11-06:
Hi @SUHAN447 ,

>  I need is a code that sends a string from a coordinator device to an end device and vice versa, from the end device to the coordinator. 

If you plan to use the APS APIs to implement this feature, I recommend reviewing the [APS doc](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/aps.html) first.

This feature can also be implemented at the ZCL layer; please refer to the [custom cluster doc](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_custom.html) and [this comment](https://github.com/espressif/esp-zigbee-sdk/issues/202#issuecomment-2005690377) for guidance.

SUHAN447, 2024-11-07:
Hello @xieqinan 

I was previously experimenting with modifying the examples within ESP-IDF, but after our conversation yesterday, I realized that the Zigbee version within ESP-IDF is an older version (1.0.9). After that, I reorganized my experiments. After your suggestions, I completed my research on APS and ZLC custom clusters. Personally, APS API seems more suitable and useful for my project. For this reason, I updated the on/off light and switch codes to make them APS-based. Currently, the codes work in their updated state, but I’m not entirely sure about their efficiency. I'm sharing the codes below. I would be very grateful if you could review and provide feedback.

Light

`#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_check.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_light.h"
#include "aps/esp_zigbee_aps.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif

static const char *TAG = "ESP_ZB_ON_OFF_LIGHT";
/********************* Define functions **************************/
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type)
    {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Zigbee stack initialized");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK)
        {
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new())
            {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            }
            else
            {
                ESP_LOGI(TAG, "Device rebooted");
            }
        }
        else
        {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK)
        {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
        }
        else
        {
[... 250 lines cut ...]
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep = esp_zb_on_off_switch_ep_create(HA_ONOFF_SWITCH_ENDPOINT, &switch_cfg);
    esp_zb_device_register(esp_zb_on_off_switch_ep);
    esp_zb_aps_data_confirm_handler_register(zb_apsde_data_confirm_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), esp_zb_buttons_handler);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}`

xieqinan, 2024-11-07:
@SUHAN447 

The `esp_zb_get_short_address();` only can obtain the address of device itself. You should assign the address of remote device to `req.dst_addr.addr_short`; I think the above can work fine if the correct address is assigned.

You can try to get the the short address of remote device in `user_find_cb()` callback.

SUHAN447, 2024-11-07:
Thank you @xieqinan I changed req.dst_addr.addr_short but I send old version code so sory thank you so much again

closed 2024-11-07 by SUHAN447