espressif/esp-zigbee-sdk #471: First test examples

Elettronicosentino, 2024-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

last

### esp-zigbee-lib version.

Last

### esp-zboss-lib version.

Last

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I uploaded the sample codes for zigbee in 2 esp32 C6 (Light and switch code).

Purtroppo il devices con il codice Light riporta la seguente striga sulla seriale
(427) phy_init: saving new calibration data because of checksum failure, mode(0)
I (427) main_task: Returned from app_main()
I (487) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
I (497) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (497) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (497) ESP_ZB_ON_OFF_LIGHT: Device started up in  factory-reset mode
I (507) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (2897) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (6307) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)

And it can't unlock at all.
Can anyone help me?

### What is the actual behavior?

I expected the example to work

### Steps to reproduce.

1. Step
2. Step
3. Step
...

xieqinan, 2024-11-07:
@Elettronicosentino ，

Could you please share the logs from the switch side? When the switch device, acting as the coordinator, reboots, it will open the network for 180 seconds. If the network is closed, other devices will be unable to join.

Elettronicosentino, 2024-11-07:
As soon as I turn on the "switch" card the message appears and by pressing the boot button it appears
I (1586) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command

180 Seconds Don't Pass

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c110,len:0xe2c
load:0x4086e610,len:0x2e5c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (24) boot: compile time Nov  6 2024 15:50:39
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000e1000
I (79) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=42090020 size=1117ch ( 70012) map
I (121) esp_image: segment 1: paddr=000211a4 vaddr=40800000 size=06e74h ( 28276) load
I (129) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=8b26ch (569964) map
I (247) esp_image: segment 3: paddr=000b3294 vaddr=40806e74 size=08220h ( 33312) load
I (256) esp_image: segment 4: paddr=000bb4bc vaddr=4080f0a0 size=022a4h (  8868) load
I (262) boot: Loaded app from partition at offset 0x10000
I (263) boot: Disabling RNG early entropy source...
I (276) cpu_start: Unicore app
I (286) cpu_start: Pro cpu start user code
I (286) cpu_start: cpu freq: 160000000 Hz
I (286) app_init: Application information:
I (289) app_init: Project name:     light_switch
I (294) app_init: App version:      1
I (298) app_init: Compile time:     Nov  6 2024 15:49:56
I (304) app_init: ELF file SHA256:  eb102551c...
I (310) app_init: ESP-IDF:          v5.3.1
I (315) efuse_init: Min chip rev:     v0.0
I (319) efuse_init: Max chip rev:     v0.99
I (324) efuse_init: Chip rev:         v0.0
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 4081AF70 len 000616A0 (389 KiB): RAM
I (342) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (348) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (356) spi_flash: detected chip: generic
I (360) spi_flash: flash io: dio
W (363) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (377) sleep: Configure to isolate all GPIO pins in sleep state
I (383) sleep: Enable automatic switching of GPIO sleep configuration
I (390) coexist: coex firmware version: 4482466
I (405) coexist: coexist rom version 5b8dcfa
I (406) main_task: Started on CPU0
I (406) main_task: Calling app_main()
I (406) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (416) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11
W (456) phy_init: saving new calibration data because of checksum failure, mode(0)
I (456) main_task: Returned from app_main()
I (516) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
I (526) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (526) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
W (546) ESP_ZB_ON_OFF_SWITCH: Network(0xc5bc) closed, devices joining not allowed.
I (546) ESP_ZB_ON_OFF_SWITCH: Device started up in non factory-reset mode
I (546) ESP_ZB_ON_OFF_SWITCH: Device rebooted
I (1586) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (2346) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (3716) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (4346) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command

xieqinan, 2024-11-07:
Based on the above log, it indicates that the network has already closed. I recommend using the latest SDK, which will automatically open the network upon reboot. Please refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c#L120

chshu, 2024-12-20:
@Elettronicosentino Please reopen if any follow up questions.