espressif/esp-zigbee-sdk #472: Unable to force specific network (pan ID, extended pan ID).

Fienberber, 2024-11-06:
### Question

Hellpo, 
I'm facing a behavior which was partially addressed in some issues.
I try to make a Zigbee Gateway using the [esp_zigbee_gateway example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_gateway/README.md), and I need to force the the device on a specific zigbee network. 
I'd like to switch networks. 

For this I use : `esp_zb_set_extended_pan_id`, `esp_zb_set_pan_id` which don't work each time. 
I also use `esp_zb_set_primary_network_channel_set` and I also set the network keys, which seem to work just fine. 

I'm able to set the pan ID but not reliably. It work only once, after one forced network switch I have to restart the chip. 
- Before setting the network parameter I leave the network using `esp_zb_bdb_reset_via_local_action` 
- I'm waiting for the leave to take effect
- I set the network parameters 
- and then using `esp_zb_start` to create the network.

I'm probably missing something. I feel like there is one thing I didn't understand otherwise it would work 100% of the time. 

I tried to set the pan ID (extended and normal) between `esp_zb_init` and `esp_zb_start` during the startup sequence without success, this won't work at all. 
It would be nice if I can switch network during the normal operation of the device.

If you need more informations I'll share it. 
Thanks in advance for your help.

xieqinan, 2024-11-07:
@Fienberber ，

> it would be nice if I can switch network during the normal operation of the device.

Zigbee does not support switching networks once the network has been formed. Therefore, `esp_zb_set_extended_pan_id()` and `esp_zb_set_pan_id()` only take effect when the device is in a factory-new state.

Fienberber, 2024-11-07:
@xieqinan thank you for your quick reply. 

It seem quite understandable that a factory-new state is required. That is what I'm doing but I unable to do it multiple time in a row. 

is the use of `esp_zb_bdb_reset_via_local_action` to trigger a factory reset appropriate ? 
Do I need to do something else to keep the Zigbee stack in the same state ?

xieqinan, 2024-11-08:
Hi @Fienberber ,

The stack only allows the application to set the PAN ID once during runtime. Could you please provide more details about the application background? There might be a better solution for your needs.

Fienberber, 2024-11-12:
Hi @xieqinan, 

I understand what I want to do isn't standard. I'd like to switch networks for production validation, we have a sequence of tests to do and wanted to use pre-defines network for each one of them. 
Depending on the step the gateway would move to a network. It would be very efficient for us if we don't have to power cycle the device. 
Currently I use a workaround which reboots and set the pan ID which work but is quite a slow process.

When you say we can only set the PAN ID once during runtime, If I completely restart the Zigbee stack it could work ? Or the ESP32 has to power cycle ? 

Thanks for your support,

xieqinan, 2024-11-15:
Hi,

> When you say we can only set the PAN ID once during runtime, If I completely restart the Zigbee stack it could work ? Or the ESP32 has to power cycle ?

With the current SDK implementation, you may need to power cycle the device.

Fienberber, 2024-11-15:
Thank you for your clarification,

This is what I was wondering. I can close the issue.

closed 2024-11-15 by Fienberber