espressif/esp-zigbee-sdk #474: Permit 34 Zigbee devices simultaneously upon power-up cannot be completed.

wensenl, 2024-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.3.1

### esp-zigbee-lib version.

1.5.0

### esp-zboss-lib version.

1.5.0

### Espressif SoC revision.

esp_zigbee_gateway：ESP32S3 + ESP32H2 （rcp），Device：ESP32H2

### What is the expected behavior?

34 Zigbee devices can be permit simultaneously. esp_zigbee_gateway no error

### What is the actual behavior?

about 10 devices are unable to join the network.

### Steps to reproduce.

1. one esp_zigbee_gateway （TYPE_COORDINATOR），power on
2. 34 zigbee device （TYPE_ROUTER），power on simultaneously
3. about 10 devices are unable to join the network.
 
esp_zigbee_gateway log：

I (56020) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
ZBOSS: nwk/nwk_cr_mesh_routing.c:406   Call zb_nlde_data_confirm(), param 12
ZBOSS: common/zb_address.c:1138   zb_address_delete ref 46
ZBOSS: common/zb_address.c:1168   WARNING: pending_for_delete ref 46
I (56170) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (56220) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (56220) ESP_ZB_GATEWAY: ZDO Device (0x7ea7) Unavailable
I (56230) ESP_ZB_GATEWAY: node device_name: A9571B000375, short_addr: 0x7ea7 is already in the array, update status to 0, unavailable_count: 0
I (56240) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (56250) ESP_ZB_GATEWAY: ZDO Device (0x7ea7) Unavailable
I (56260) ESP_ZB_GATEWAY: node device_name: A9571B000375, short_addr: 0x7ea7 is already in the array, update status to 0, unavailable_count: 1
ZBOSS: common/zb_address.c:941   WARNING: remove ref 46 which was in pending_for_delete
ZBOSS: common/zb_address.c:1138   zb_address_delete ref 46
I (57010) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (57010) ESP_ZB_GATEWAY: ZDO Device (0x7ebc) Unavailable
I (57020) ESP_ZB_GATEWAY: node device_name: 2E544A010430, short_addr: 0x7ebc is already in the array, update status to 0, unavailable_count: 0
I (58070) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (58600) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (59160) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (59190) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (61480) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (61480) ESP_ZB_GATEWAY: ZDO Device (0x8430) Unavailable
I (61480) ESP_ZB_GATEWAY: node device_name: A9578700038D, short_addr: 0x8430 is already in the array, update status to 0, unavailable_count: 0
I (61600) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (61600) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0000, network_addr: 0x8430, unknown_command_id: 0
I (62600) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (62600) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0000, network_addr: 0x7ebc, unknown_command_id: 0
I (64600) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (64600) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0000, network_addr: 0xf28f, unknown_command_id: 0
I (66600) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (66600) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0000, network_addr: 0xcc95, unknown_command_id: 0
I (66670) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (66670) ESP_ZB_GATEWAY: ZDO Device (0xcc95) Unavailable
I (66670) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (66680) ESP_ZB_GATEWAY: ZDO Device (0x8430) Unavailable
I (66690) ESP_ZB_GATEWAY: node device_name: A9578700038D, short_addr: 0x8430 is already in the array, update status to 0, unavailable_count: 1
I (66700) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (66710) ESP_ZB_GATEWAY: ZDO Device (0xf28f) Unavailable
D/[1970-01-01 00:01:10] device_name D83BDA587737 is work, need to heartbeat
D/[1970-01-01 00:01:10] [70.222][LK-0309] pub: v1/gateway/D83BDA587737/event

xieqinan, 2024-11-12:
Hi @wensenl ,

The issue might be caused by the insufficient static buffers. Could you try using `esp_zb_overall_network_size_set(128)` and `esp_zb_io_buffer_size_set(100)` before calling `esp_zb_init()` and test again?

wensenl, 2024-11-13:
Hi,@xieqinan
I add code，but also can not Permit join 34 Zigbee devices simultaneously upon power-up.

modify code:
static void esp_zb_task(void *pvParameters)
{
#if CONFIG_OPENTHREAD_SPINEL_ONLY
    esp_radio_spinel_register_rcp_failure_handler(rcp_error_handler, ESP_RADIO_SPINEL_ZIGBEE);
#endif
    
    **//设置zb网络和缓冲大小
    esp_zb_overall_network_size_set(128);
    esp_zb_io_buffer_size_set(100);**

    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
#if CONFIG_OPENTHREAD_SPINEL_ONLY
    ESP_ERROR_CHECK(check_ot_rcp_version());
#endif
    
    #if 0
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    #else
    uint32_t zb_channel = 0, channel_mask = 0;
    //设置zb channel
    zb_channel = 11;
    channel_mask = (1l << zb_channel);
    esp_zb_set_primary_network_channel_set(channel_mask);
    //设置pannid
    esp_zb_set_pan_id(0xfedc);
    #endif
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = HA_ESP_GW_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_REMOTE_CONTROL_DEVICE_ID,
        .app_device_version = 0,
    };
    //basic cluster
    esp_zb_attribute_list_t *basic_cluster = esp_zb_basic_cluster_create(NULL);
    esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, ESP_MANUFACTURER_NAME);
    esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, ESP_MODEL_IDENTIFIER);
    esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    //identify cluster
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    //groups cluster
    esp_zb_groups_cluster_cfg_t group_cfg;
    group_cfg.groups_name_support_id = ESP_ZB_ZCL_GROUPS_NAME_SUPPORT_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_groups_cluster_create(&group_cfg);
    esp_zb_cluster_list_add_groups_cluster(cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    //custom cluster
    esp_zb_attribute_list_t *custom_attr = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_TRANSMIT_NODE);
    esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_attr, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    //
    esp_zb_ep_list_add_gateway_ep(ep_list, cluster_list, endpoint_config);
    esp_zb_device_register(ep_list);

    //
    esp_zb_core_action_handler_register(zb_action_handler);
    //
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
    esp_rcp_update_deinit();
    vTaskDelete(NULL);
}

error log：
D/[1970-01-01 00:00:35] smart light power device A9572A000366 other func
[... 49 lines cut ...]
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565   MAC callbacks rb overflow! param 59

wensenl, 2024-11-13:
Hi,@xieqinan
Additionally, if we need to provision 34 devices in batch, can all 34 devices be powered on simultaneously, or do they need to be powered on in batches? What is the maximum number of devices our Zigbee network can support for simultaneous power-on and provisioning?

chi-lee-almagest, 2024-11-14:
@wensenl We face the same problem and raised #480, it looks like they didn't increase the ZBOSS input static buffer.

wensenl, 2024-11-14:
Hi,@xieqinan，
I increased the ZBOSS input static buffer, but the same error still occurs. When 34 unprovisioned devices are powered on simultaneously for provisioning, the above error appears. How can this issue be resolved?

code：
//设置zb网络和缓冲大小
esp_zb_overall_network_size_set(128);
esp_zb_io_buffer_size_set(100);

xieqinan, 2024-11-15:
Hi @wensenl ,

The `esp_zb_io_buffer_size_set(100);` function allocates a total of 100 buffers for the stack, with 4 reserved for high-priority events. The remaining buffers are evenly divided between input and output buffers. The log message `OOM! bufs allocated {48 i-o 5} max 100` indicates that the input buffers are insufficient. Could you please set the `esp_zb_io_buffer_size_set(254)` and `esp_zb_scheduler_queue_size_set(254)` then test again?

wensenl, 2024-11-18:
Hi,@xieqinan，
set the esp_zb_io_buffer_size_set(254) and esp_zb_scheduler_queue_size_set(254)，the OOM! bufs allocated {48 i-o 5} max 10 error did not occur. 
However, the problem of 34 Zigbee devices being unable to successfully power on for permit join network remains. How to solve this problem? Can you set up an environment to test this issue on your end?

The error message printed by Zigbee during my testing is as follows:
I (1697997) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1697997) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0000, network_addr: 0xee46, unknown_command_id: 0
D/[1970-01-01 01:50:42] heartbeat response

ZBOSS: zdo/af_rx.c:494   zb_zdo_device_is_unreachable: addr_ref 56
ZBOSS: zdo/af_rx.c:494   zb_zdo_device_is_unreachable: addr_ref 56
ZBOSS: zdo/zdo_app_leave.c:360   no nbt for addr_ref 56, do nothing
ZBOSS: zdo/zdo_app_leave.c:383   zb_nwk_neighbor_delete addr_ref not found [-28]
ZBOSS: zdo/zdo_app_leave.c:360   no nbt for addr_ref 56, do nothing
ZBOSS: zdo/zdo_app_leave.c:383   zb_nwk_neighbor_delete addr_ref not found [-28]
I (1701987) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (1701997) ESP_ZB_GATEWAY: ZDO Device (0xee46) Unavailable
I (1702007) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (1702007) ESP_ZB_GATEWAY: ZDO Device (0xee46) Unavailable
E (1713147) ESP_RADIO_SPINEL: dropping radio frame: Parse
ZBOSS: nwk/nwk_main.c:3775   Put 2 into input queue (blocked by 49)
ZBOSS: common/zb_address.c:696   WARNING: clear pending_for_delete for ref 102
ZBOSS: mac/mac_cr_coordinator.c:250   Oops - out of MAC queue!
I (1713197) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
ZBOSS: nwk/nwk_main.c:3775   Put 17 into input queue (blocked by 49)
ZBOSS: secur/bdb_secur.c:398   ACHTUNG: schedule bdb_link_key_refresh_alarm, param 12
ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 2 from in queue
I (1713267) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
ZBOSS: secur/bdb_secur.c:398   ACHTUNG: schedule bdb_link_key_refresh_alarm, param 42
ZBOSS: nwk/nwk_main.c:2360   set input_blocked_by and process 17 from in queue
I (1713287) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1713287) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0006, network_addr: 0xb771, unknown_command_id: 0
I (1713297) ESP_ZB_GATEWAY: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (1713307) ESP_ZB_GATEWAY: NLME Status Indication status: 0x0006, network_addr: 0xb771, unknown_command_id: 0
I (1713317) ESP_ZB_GATEWAY: New device commissioned or rejoined (short: 0x9ffb)
I (1713327) ESP_ZB_GATEWAY: esp_zb_zcl_add_group add device 0x9ffb to group 0x0001 successfully
[111811227s3.txt](https://github.com/user-attachments/files/17796037/111811227s3.txt)

wensenl, 2024-11-18:
Hi,@xieqinan，
Do we have API interfaces to set up whitelists for permit join networks to control device permit join networks? If so, which APIs are they? How to use it?

xieqinan, 2024-11-19:
Hi @wensenl ,

> However, the problem of 34 Zigbee devices being unable to successfully power on for permit join network remains. How to solve this problem? Can you set up an environment to test this issue on your end?

We are setting up the test environment based on the above description on our side. Please be patient.

> Do we have API interfaces to set up whitelists for permit join networks to control device permit join networks? If so, which APIs are they? How to use it?

No, the whitelist API is not available in the current SDK (v1.6.0).

xieqinan, 2024-11-20:
Hi @wensenl ,

Could you please also share the logs generated by the devices that failed to join the network?

wensenl, 2024-11-21:
Hi,@xieqinan，
the devices log
[11211125h2.txt](https://github.com/user-attachments/files/17840505/11211125h2.txt)

wensenl, 2024-11-21:
Hi,@xieqinan，
the devices log

[11211125h2.txt](https://github.com/user-attachments/files/17840527/11211125h2.txt)

wensenl, 2024-11-21:
Hi,@xieqinan，
the devices log

[device esp32h2.txt](https://github.com/user-attachments/files/17840638/device.esp32h2.txt)
[gateway esp32s3.txt](https://github.com/user-attachments/files/17840639/gateway.esp32s3.txt)

wensenl, 2024-11-25:
Hi,@xieqinan，
Has there been any progress on the issue?

xieqinan, 2024-11-25:
> Has there been any progress on the issue?

We have reproduced this issue, although with some differences. The issue is triggered by an insufficient size of the TX queue on our side, causing an infinite loop. We have fixed it and successfully joined 40 devices. This fix will be included in the next release. If you would like to receive temporary libraries to address this issue, please contact our [technical inquiries](https://www.espressif.com/en/contact-us/technical-inquiries).

wensenl, 2024-11-26:
Hi,@xieqinan，
What is the new SDK version? When will the release version be released?

xieqinan, 2024-11-26:
I can't provide a specific time as it mainly depends on the development progress of the release plan.

xieqinan, 2024-12-13:
Hi,

We provided you with the temporary Zigbee libraries; were they helpful? The fixes included in those temporary libraries have already been incorporated into esp-zigbee-sdk v1.6.1. Please check it out.

chshu, 2024-12-20:
The issue should be fixed in the 1.6.1 release, feel free to reopen if any follow up questions.