espressif/esp-zigbee-sdk #476: Monitoring RSSI for Zigbee Gateway Devices

renatomotorline, 2024-11-12:
### Question

What is the best way to keep the RSSI of gateway devices updated? I noticed that `esp_zb_zcl_cmd_read_attr_resp_message_t` contains RSSI information, but `esp_zb_zcl_report_attr_message_t` does not. Shouldn't the RSSI information also be included in `esp_zb_zcl_report_attr_message_t` to facilitate continuous monitoring?

xieqinan, 2024-11-13:
> What is the best way to keep the RSSI of gateway devices updated?

The Zigbee specification does not define a specific application command to retrieve the RSSI of devices, so I think the good way depend on your application, there isn’t an ideal method for monitoring device RSSI directly. However, `esp_zb_zdo_mgmt_lqi_req()` provides a way to obtain basic information, including the LQI, from the neighbor table of a local or remote device—this may meet your requirements. Alternatively, you can use `esp_zb_nwk_get_next_neighbor()` to check the RSSI of devices listed in the gateway's neighbor table.

chshu, 2024-12-20:
Feel free to reopen is any follow up questions.

closed 2024-12-20 by chshu