espressif/esp-zigbee-sdk #477: Modifying pan ID, channel, and network key in zb_storage

chi-lee-almagest, 2024-11-12:
### Question

I know that the API has provided a function for setting the pan ID once during runtime and a function for changing the network key. But I would like to persist the change so that I don't have to call those functions during runtime.

The situation is that we are providing home automation solutions to residential projects. Our technicians will grab a new gateway, form a network and then join all required devices to the network at our warehouse. And then the installers will go to the site and install everything without configuration. 

One of the unit requires additional devices to be joined to the network but during the process we realized the pan ID and the network key the new devices using are different from the previous ones. So we use the functions to set it to the previous pan ID and network key as a temporary solution.

Now we want to eliminate those lines in the source code to prevent applying the parameters to other gateways. And also we don't want an OTA to accidentally switch our "special" gateway to use the wrong pan ID and network key some day.

chi-lee-almagest, 2024-11-13:
@xieqinan
The previously commissioned devices are installed above the false ceiling. And some of them are not easily reachable. That would be the last resort but we would like to see if there's a software solution to it.

chshu, 2024-12-20:
For any kind of software solution, you need to call the APIs to change the network parameters explicitly, right?

One possible solution is to push OTA to the installed devices, and switch to new network after reboot.

xieqinan, 2025-05-30:
@chi-lee-almagest 

Could you please share the current state about this issue?

chi-lee-almagest, 2025-07-18:
> [@chi-lee-almagest](https://github.com/chi-lee-almagest)
> 
> Could you please share the current state about this issue?

We use the zb_fct partition to set the channel & network key. However, we still have some gateways that the zb_storage might be corrupted and the gateways formed a new network. Can you advise us that if it is possible that after we set the pan ID, network key, and channel to the same as before, the network can be successfully restored? Are they anything critical to the network in zb_storage?

xieqinan, 2025-07-21:
@chi-lee-almagest ,

>  Can you advise us that if it is possible that after we set the pan ID, network key, and channel to the same as before, the network can be successfully restored?

I think it’s unsafe and unreliable. I recommend considering the OTA solution instead.

xieqinan, 2025-09-30:
Hi @chi-lee-almagest ,

Do you have any updates? Please let me know your status.