espressif/esp-zigbee-sdk #479: Can Coordinator send esp_zb_zdo_active_ep_req() and esp_zb_zdo_simple_desc_req() to multiple end device at the same time

vanbuong, 2024-11-13:
### Question

Hello, I'm implementing the ZigBee gateway, which will discover all the information about the newly joined device.
These two functions will be used for that purpose
**esp_zb_zdo_active_ep_req()**
**esp_zb_zdo_simple_desc_req()**
However, I saw that the callback function to handle the response does not input the device network address. As I understand, seems I can put the req_param in the req function and use that for checking the device address in the response callback later.

But I'm curious to know if it works correctly when many requests are sent for different devices before receiving a response. Or I have to wait the response to be received before sending the next request?

xieqinan, 2024-11-13:
Hi,

> I saw that the callback function to handle the response does not input the device network address. 

You could use the `user_ctx` parameter in `esp_zb_zdo_active_ep_req()` and `esp_zb_zdo_simple_desc_req()` to associate each request with the device's network address, allowing multiple responses to be recognized.

lpy4105, 2024-11-15:
> But I'm curious to know if it works correctly when many requests are sent for different devices before receiving a response. Or I have to wait the response to be received before sending the next request?

No worry. The callbacks and their `user_ctx` are identified by the `TSN` of the request, so you can do a second request without waiting for the first one to be responded.

vanbuong, 2024-11-19:
Thank @lpy4105 @xieqinan for clarification

closed 2024-11-19 by vanbuong