espressif/esp-zigbee-sdk #480: ZB_ESP_MAC: No buffer, handle it in the next iteration

chi-lee-almagest, 2024-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-638-ga0f798cfc4

### esp-zigbee-lib version.

1.6.0

### esp-zboss-lib version.

1.6.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I expected the stack to handle requests

### What is the actual behavior?

We log our free heap periodically and we still have free heap size about 305640 bytes.

We are controlling some curtain motors, they will keep reporting the curtain position while we're controlling it. If we try to control a few of them at the same time, we can first see:

```
ZBOSS: nwk/nwk_nlme.c:808   OOM state is detected. ACHTUNG: Out of memory!
ZBOSS: common/zb_bufpool_mult.c:1650   OOM! bufs allocated {23 i-o 38} max 80
ZBOSS: nwk/nwk_nlme.c:871   OOM state ends
```

And then we can see:
```
ZBOSS: nwk/nwk_cr_mesh_routing.c:330   Oops - out of memory, can't initiate route disc
ZBOSS: nwk/nwk_cr_mesh_routing.c:268   Call zb_nlde_data_confirm(), param 26
ZBOSS: nwk/nwk_cr_mesh_routing.c:330   Oops - out of memory, can't initiate route disc
ZBOSS: nwk/nwk_cr_mesh_routing.c:268   Call zb_nlde_data_confirm(), param 37
ZBOSS: common/zb_bufpool_mult.c:426   delayed buf q[0] is full
ZBOSS: zdo/af_rx.c:737   could not send ZB_ZDO_DEVICE_UNAVAILABLE signal
ZBOSS: common/zb_bufpool_mult.c:426   delayed buf q[0] is full
ZBOSS: zdo/af_rx.c:737   could not send ZB_ZDO_DEVICE_UNAVAILABLE signal
ZBOSS: nwk/nwk_nlme.c:850   ACHTUNG: Sending OOM status
```

```
E (84520) ESP_ZIGBEE_APS: esp_zb_aps_data_request(102): Failed to get buffer id
E (84520) ESP_ZIGBEE_APS: esp_zb_aps_data_request(102): Failed to get buffer id
```

### Steps to reproduce.

Control some curtain motors at the same time.

xieqinan, 2024-11-13:
Hi,

The issue appears to be caused by insufficient IO buffer resources, though I'm not certain whether there's a memory leak in the IO buffer. Please help verify the below question.

- How many curtain motors are being controlled simultaneously?
- What is the reporting period from the motors?
- If the reporting stops, does the Zigbee stack continue functioning properly?

chi-lee-almagest, 2024-11-13:
@xieqinan 

1. Only 2 motors are being controlled at the same time
2. When the curtain is moving, it keeps reporting the position of the curtain around every other 100ms
3. Once the error "ZB_ESP_MAC: No buffer, handle it in the next iteration" first pop up in the console. The stack will only continue to print this message nonstop.

wensenl, 2024-11-14:
@chi-lee-almagest 
Have you encountered the following issue: 
when dozens of unprovisioned devices（eg：42 devices） are powered on simultaneously for provisioning, they are unable to complete the network formation? How did you resolve this issue?

error log：
D/[1970-01-01 00:00:35] smart light power device A9572A000366 other func
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x5293, device 74:4d:bd:ff:fe:61:d9:b0, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0xf35b, device 74:4d:bd:ff:fe:61:d6:77, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0xc503, device 74:4d:bd:ff:fe:61:d9:b7, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x9e3f, device 74:4d:bd:ff:fe:61:d6:7c, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:7d, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:da:37, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0xe268, device 74:4d:bd:ff:fe:61:d6:8f, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:7d, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:86, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:91, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:7f, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x9489, device 74:4d:bd:ff:fe:61:d6:95, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d9:b5, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d9:ae, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:da:39, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:53, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:da:30, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:53, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x3629, device 74:4d:bd:ff:fe:61:d6:90, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0xbdef, device 74:4d:bd:ff:fe:61:d6:7b, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x32ca, device 74:4d:bd:ff:fe:61:d6:87, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 0, address 0x403, device 74:4d:bd:ff:fe:61:da:35, mac_iface_idx 0
ZBOSS: nwk/nwk_nlme.c:803 OOM state is detected. ACHTUNG: Out of memory!
ZBOSS: common/zb_bufpool_mult.c:1650 OOM! bufs allocated {48 i-o 5} max 100
ZBOSS: nwk/nwk_nlme.c:845 ACHTUNG: Sending OOM status
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:8e, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:43, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:80, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:41, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d6:8a, mac_iface_idx 0
ZBOSS: nwk/nwk_cr_join.c:1285 association: status 2, address 0xffff, device 74:4d:bd:ff:fe:61:d9:c4, mac_iface_idx 0
ZBOSS: nwk/nwk_nlme.c:866 OOM state ends
ZBOSS: nwk/nwk_nlme.c:803 OOM state is detected. ACHTUNG: Out of memory!
ZBOSS: common/zb_bufpool_mult.c:1650 OOM! bufs allocated {48 i-o 5} max 100
I (42319) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
ZBOSS: secur/bdb_secur.c:398 ACHTUNG: schedule bdb_link_key_refresh_alarm, param 60
I (42319) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (42369) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
ZBOSS: secur/bdb_secur.c:398 ACHTUNG: schedule bdb_link_key_refresh_alarm, param 62
I (42379) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (42419) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
ZBOSS: secur/bdb_secur.c:398 ACHTUNG: schedule bdb_link_key_refresh_alarm, param 64
I (42429) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12
ZBOSS: common/zb_scheduler.c:565 MAC callbacks rb overflow! param 12

chi-lee-almagest, 2024-11-14:
@wensenl basically we also have OOM issues. From the log, you can see a line ZBOSS: common/zb_bufpool_mult.c:1650 OOM! bufs allocated {48 i-o 5} max 100

Which means you have 48 input buffers and 5 output buffers allocated and the maximum number of buffers is 100. So you haven't hit the limit yet but I suspect that some static memory pool has run out.

chi-lee-almagest, 2024-11-15:
@xieqinan after we increase the io buffer and overall network size, things seem to be a lot smoother. Please let us know what is the maximum of those 2 parameters.

wensenl, 2024-11-15:
@chi-lee-almagest
How did you add 100 devices to the Zigbee network? Should one device be connected to the Zigbee network one by one, or should 100 devices be powered on simultaneously to join the network?

chi-lee-almagest, 2024-11-15:
@wensenl we joined them mostly one by one, but one time we joined a batch of 8 devices

xieqinan, 2024-11-15:
Hi @chi-lee-almagest ,

Based on the information provided in [this comment](https://github.com/espressif/esp-zigbee-sdk/issues/480#issuecomment-2473692533), we will attempt to reproduce this issue on our side. Meanwhile, could you please test with the settings `esp_zb_io_buffer_size_set(254)` and `esp_zb_scheduler_queue_size_set(254)`?

> We then face issue https://github.com/espressif/esp-zigbee-sdk/issues/342, then we have upgraded everything to 1.6.0. And now we see this. We have over 5000 ESP32 and ESP32-C6 devices manufactured at our warehouse and I feel like we keep running into stack issues when we deploy our products at scale. There will be more than 170 devices at one of our pending unit. Sometimes, I just couldn't believe ZBOSS would have such bugs given that it is used by so many SoCs manufacturers. I couldn't tell whether they are bugs from ZBOSS or just ESP adoption issues. We couldn't access the source code which is one big problem, otherwise we would fix it in no time. Please help iron out the issues.

Sorry for the inconvenience caused to your project. The core logic of the esp-zigbee-sdk is based on ZBOSS, with both ESP and ZBOSS striving to improve performance. We will do our best to resolve your issue.

For source code requests or additional feature support, you can contact our business team via [technical inquiries](https://www.espressif.com/en/contact-us/technical-inquiries).

chi-lee-almagest, 2024-11-18:
@wensenl we use our own code for interfacing between ncp and host

xieqinan, 2024-11-19:
@chi-lee-almagest ,

Does the `No buffer, handle it in the next iteration` still exist when the `esp_zb_io_buffer_size_set(254) and esp_zb_scheduler_queue_size_set(254)` configuration is added? 

> sometimes, there's still some non-tree link failure even when the router is so close to the gateway.

Please provide the related log for us to detect this issue.

chi-lee-almagest, 2024-11-26:
@xieqinan it doesn't seem to appear now. but we are running into all sorts of routing issues. the logs are really large and messy i don't think we can provide it at this moment.

btw, after we upgrade to 1.6.0, we found out that some of the zigbee routers and zigbee end devices will leave the network by themselves. Is it defined in the BDB that if the device can't connect to the parent it will leave? But when using 1.5.x, we didn't see this issue.

xieqinan, 2024-11-27:
@chi-lee-almagest 

> we found out that some of the zigbee routers and zigbee end devices will leave the network by themselves. Is it defined in the BDB that if the device can't connect to the parent it will leave? But when using 1.5.x, we didn't see this issue.

Based on the [SDK Release Notes](https://github.com/espressif/esp-zigbee-sdk/blob/main/RELEASE_NOTES.md), if you have not called `esp_zb_secur_network_security_enable(0)` to set the `nwkSecurityLevel`, I believe the network layer implementation in v1.6.0 is the same as in v1.5.x. Generally, routers do not leave the network on their own.  

If there is heavy network traffic and the end device cannot maintain communication with its parent, the parent may remove the timeout device from the network. Could you please call `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);` on the device that has left the network to allow it to rejoin?

chi-lee-almagest, 2024-11-29:
@xieqinan we found out that many routers have left the network since we upgrade their stack to 1.6.0. Also from our gateway, even if the router is in the gateway's neighbor table, the gateway will keep sending route request out. We have another unit which uses some light switches as routers and it works extremely well. However, the unit which is using ESP stack as the routers is a disaster with the new stack version.

chi-lee-almagest, 2024-11-29:
[20241129_44bd.zip](https://github.com/user-attachments/files/17955286/20241129_44bd.zip)
We were sitting near a curtain motor which has a network address of 0x44bd, and we controlled it using zcl using cluster 0xef00, command 0x04. The curtain motor is a few hops away from the coordinator, the hops in between are mostly ESP32-C6 routers with IEEE802.15.4 timing optimization enabled, buffer size set to maximum. As you can see in the log file, the network was very unstable. The routing record was useless, and there were a lot of routing request in the network. There were also bunch of non link tree failure. 

The whole user experience is bad. Sometimes the curtain motor can be controlled but without report back to the gateway. Sometimes the control commands were delayed by more than 10 seconds. Sometimes it was totally incontrollable.

xieqinan, 2024-11-29:
@chi-lee-almagest , Please also share us with the network key of wireshark.

chi-lee-almagest, 2024-11-29:
@xieqinan The network key is 02:71:85:4a:e6:a5:04:6d:96:d5:0f:79:df:2f:00:19.
We prefer Ethernet over Wi-Fi, so we have disabled Wi-Fi for this gateway.

chi-lee-almagest, 2024-12-02:
@xieqinan please let me know if there's any updates?

xieqinan, 2024-12-03:
I am reviewing the wireshark packet.

> We prefer Ethernet over Wi-Fi, so we have disabled Wi-Fi for this gateway.

I mean that if the ESP32-C6 is acting as the coordinator, please disable Wi-Fi, Ethernet, or other potentially interfering features to avoid any unknown effects.  

By the way, could you also provide the simple steps to reproduce the `route request too many` issue?

chi-lee-almagest, 2024-12-03:
@xieqinan we have disabled everything else already. Not only the coordinator, also the routers.
We just keep adding esp32-c6 routers and some other routers using silab to the network.

chi-lee-almagest, 2024-12-06:
@xieqinan please let us know if you can reproduce the issue which is shown in our log file.

xieqinan, 2024-12-12:
Hi @chi-lee-almagest,

We successfully reproduced this issue using 100 devices in a controlled test environment, and they worked as expected. Here’s how we conducted the test:

1. We used 1 ESP32C6 as the coordinator to form a Zigbee network.
2. We then used 99 ESP32C6 devices as routers to join the network, divided into 5 groups that joined sequentially.
3. All devices were placed in close proximity, ensuring they could communicate via a single hop.
4. During the commissioning process, these devices did not generate significant application traffic.
5. We observed the network convergence using a sniffer. Once stable, we sent various commands across the network, which performed as expected.

Given this, we have a few questions to clarify your scenario:

1. Is the topology of your network and the device types used similar to our test setup?  
2. When a device joins your network, what kind of application commands/packets are being sent?  
3. Have you tested sending application commands/packets after ensuring the network is fully converged and stable?  

Your insights on these questions would help us better understand and analyze the root cause of your issue.

chi-lee-almagest, 2024-12-12:
@xieqinan do you mean you have found the issues or you haven't found any problems at that scale?

1. Similar topology, but we join them one by one and they are not in close proximity. I can send you the floor map if you could provide me an email address
2. We first "interview" the device by send ZDO commands like node descriptor req and active eq req, and we read the ZCL basic cluster
3. We haven't been able to see a fully converged and stable network. Every time there are some non-tree link failure here and there.

xieqinan, 2024-12-12:
> do you mean you have found the issues or you haven't found any problems at that scale?

I have not reproduce your issue on my side, they work fine.

> Similar topology, but we join them one by one and they are not in close proximity. I can send you the floor map if you could provide me an email address

I believe your product is meanful and urgent for you. Why not contact ESP's [technical support](https://www.espressif.com/en/contact-us/technical-inquiries) for more official and prompt assistance?

> We haven't been able to see a fully converged and stable network. Every time there are some non-tree link failure here and there.

The unstable network might be caused by large data transmissions, heavy traffic, or a complex topology. We are using a simple topology to reproduce the issue and have observed that the network converges within a short time.

> Btw, we can indeed see some routers randomly leaving the gateway. Please let us know under which conditions a router will leave a coordinator

Yes, it also happens in my test. To establish a more efficient mesh network, the coordinator will assist a router in leaving. However, the rejoin operation needs to be implemented at the application layer, for example:
```
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Deferred driver initialization %s", deferred_driver_init() ? "failed" : "successful");
            ESP_LOGI(TAG, "Device started up in %s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : "non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network steering");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            ESP_LOGW(TAG, "%s failure with status: %s, retry it", esp_zb_zdo_signal_to_string(sig_type),
                     esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb,
                                   ESP_ZB_BDB_MODE_INITIALIZATION, 500);
        }
        break;
```

chshu, 2024-12-20:
@chi-lee-almagest Since the origin `No buffer` issue has been resolved, closing the issue. Please file new issues for follow up issues.

Basically, in a large scale and noisy network environment, the router request and leave commands don't indicate any issue, you can monitor the overall flow.

closed 2024-12-20 by chshu