espressif/esp-zigbee-sdk #483: How to renew the open network time on Zigbee coordinator?

nalves23, 2024-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.3.1

### esp-zigbee-lib version.

1.5.1

### esp-zboss-lib version.

1.5.1

### Espressif SoC revision.

ESP32-S3

### What is the expected behavior?

When a coordinator has network in open mode to accept new Zigbee devices, after device joins, timer of open network must be renew by another 180 seconds.

### What is the actual behavior?

I have a Zigbee coordinator with a physical button to open/close network for router or end devices joins the Zigbee network. 
My default open network time is defined to 180 seconds. To open and close the network, I use the functions esp_zb_bdb_open_network(ZB_ZGP_DEFAULT_COMMISSIONING_WINDOW) and esp_zb_bdb_close_network(), respectively, where ZB_ZGP_DEFAULT_COMMISSIONING_WINDOW is 180. 

I have a feature that after one device joins the network, I want to renew the open network time. After device joins I call the function esp_zb_bdb_open_network(ZB_ZGP_DEFAULT_COMMISSIONING_WINDOW)  but the open network time doesn't renew it. 
I have try it call first esp_zb_bdb_close_network() and after esp_zb_bdb_open_network(ZB_ZGP_DEFAULT_COMMISSIONING_WINDOW), but this procedure still doesn't work.

There are any function or callback that i can use to read open network timer or renew this timer when I need it?

### Steps to reproduce.

1. Open network during 180 seconds (pushed button)
2. Put Router device in joining mode to join Zigbee network, after 60 seconds of 1
3. Device joins and coordinator call again the function esp_zb_bdb_open_network(ZB_ZGP_DEFAULT_COMMISSIONING_WINDOW)
4. After 180 seconds of point 1, the network close associations.

xieqinan, 2024-11-18:
Hi @nalves23 ,

I think your requirement can be achieved using the [gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway) and the [dimmable light example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light) without any modifications.  

The gateway running on the ESP32-S3 would behave as follows: You can observe the `Network(0x1c84) is open for 180 seconds` log being printed twice.

```
I (5699) ESP_ZB_GATEWAY: Initialize Zigbee stack
I (5709) ESP_ZB_GATEWAY: Device started up in  factory-reset mode
I (5709) ESP_ZB_GATEWAY: Start network formation
W (5889) ESP_ZB_GATEWAY: Network(0x1c84) closed, devices joining not allowed.
I (5889) ESP_ZB_GATEWAY: Formed network successfully (Extended PAN ID: 60:55:f9:ff:fe:f7:a2:ad, PAN ID: 0x1c84, Channel:13, Short Address: 0x0000)
I (6499) ESP_ZB_GATEWAY: Network(0x1c84) is open for 180 seconds
I (6499) ESP_ZB_GATEWAY: Network steering started
I (71019) ESP_ZB_GATEWAY: ZDO signal: NWK Device Associated (0x12), status: ESP_OK
I (71019) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Update (0x30), status: ESP_OK
I (71119) ESP_ZB_GATEWAY: New device commissioned or rejoined (short: 0x9a67)
I (71729) ESP_ZB_GATEWAY: ZDO signal: ZDO Device Authorized (0x2f), status: ESP_OK
I (71819) ESP_ZB_GATEWAY: Network(0x1c84) is open for 180 seconds

```

nalves23, 2024-11-18:
Hi @xieqinan 

Thank you for your feedback.

I have test it the same example that you referee and It worked well. However, in my example it didn't work again!
What I did immediately was to bring my code closer to the example code, calling the network opening at the beginning of my gateway's operation (in the ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT event of the esp_zb_app_signal_handler function), it worked well:
1- Gateway starts;
2- When the even ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT occurs, the function esp_zb_bdb_open_network(180) is called;
3- After 60 seconds, I associated a device, and in the ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE event, I called the esp_zb_bdb_open_network(180) function again;
4- After 180 seconds from the previous point (3) the network is closed. So it worked as I expected.

The problem occurs if I don't call the network opening function at gateway start up but if I call the network opening event randomly, with the gateway running for some time via a press button, for example.
My question is if there is any procedure that changes these settings after the gateway is initialized?

Can you try an example like mine, or has anyone already done it and had the same result?

kind regards 
nalves23

xieqinan, 2024-11-19:
@nalves23 ,

I think this issue can be resolved more quickly, if you can provide a simple example to reproduce it.

xieqinan, 2025-05-30:
Hi @nalves23 ,

Is this issue still causing confusion? Could you please share any updates or progress regarding it?

xieqinan, 2025-08-12:
Hi @nalves23 ,

I believe the original issue can be resolved as mentioned in https://github.com/espressif/esp-zigbee-sdk/issues/483#issuecomment-2481842301. Since there has been no response for a while, I will close this issue. If you have any updates, please feel free to reopen it.