espressif/esp-zigbee-sdk #486: Zigbee reconnect range shorter than working range

Matthieu-Poulain, 2024-11-20:
### Question

When an esp32-C6(EndDevice) is paired with an esp32-H2(Coordinator), if both go to deep sleep (Same behavior if powered on from a shut down state), the range at which the reconnect occurs is much shorter than the operating range at which both device keep communicating when already connected.

My guess is the End device does not reconnect to the network if the rssi ou LQI is below a certain range. Is this correct and if so, is there a way to configure the right threshold?

### Additional context.

The ESP32-C6 is on a set position, while the H2 is battery powered and is mobile. If the reconnect occurs under a certain range and is successful, i can get further and still transmit/receive my data (At least twice the range). However, if one of the 2 device gets disconnected and the reconnect procedure restart below that range, they are unable to reconnect.

Note that this range changes with the parameter of esp_zb_set_tx_power(XX);

xieqinan, 2024-11-21:
@Matthieu-Poulain ,

Please refer to https://github.com/espressif/esp-zigbee-sdk/issues/352#issuecomment-2141402806.

Matthieu-Poulain, 2024-11-22:
This solved my issue, many thanks!

xieqinan, 2024-12-03:
@Matthieu-Poulain If the issue has been resolved, please consider closing it. Welcome to open other SDK issues.

closed 2024-12-09 by Matthieu-Poulain