espressif/esp-zigbee-sdk #487: OTA Example Question

script0803, 2024-11-21:
### Question

Hi,

I'm using esp32h2 as client and esp32c6 as server and I'm having trouble using the ota example.
```plaintext
I (3448) ESP_OTA_CLIENT: Joined network successfully (Extended PAN ID: f0:f5:bd:ff:fe:32:9c:30, PAN ID: 0x2e49, Channel:13, Short Address: 0x6eb7)
I (3988) ESP_OTA_CLIENT: Queried OTA image from address: 0x0000, endpoint: 5
I (3988) ESP_OTA_CLIENT: Image version: 0x1010110, manufacturer code: 0x1001, image size: 618536
I (3998) ESP_OTA_CLIENT: Approving OTA image upgrade
I (4008) ESP_OTA_CLIENT: -- OTA upgrade start
I (4578) ESP_OTA_CLIENT: -- OTA Client receives data: progress [167/618480]
E (4578) ESP_DELTA_OTA_OPS: Invalid magic word in patch
E (4578) ESP_DELTA_OTA_OPS: esp_delta_ota_write(154): Patch Header verification failed, status: ESP_ERR_INVALID_ARG
E (4588) ESP_OTA_CLIENT: zb_ota_upgrade_status_handler(155): Failed to write OTA data to partition, status: ESP_ERR_INVALID_ARG
I (4598) ESP_OTA_CLIENT: OTA status: 4
```
In addition, if I want to use other Zigbee3.0 dongles as servers for OTA on other platforms, such as Zigbee2MQTT or ZHA, what should I do? Does the OTA file need to be generated using Delta OTA?

xieqinan, 2024-11-21:
Please refer to the [README.md](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_ota/ota_client/README.md#delta-ota-upgrade-functions) first.

chshu, 2025-02-08:
An official tool is provided to generate the standard OTA file format: [image_builder_tool](https://github.com/espressif/esp-zigbee-sdk/tree/main/tools/image_builder_tool), please follow the instructions here: [ota docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_ota_upgrade.html#generate-an-ota-upgrade-image-for-the-esp-platform).

Please file new issues if any problem when using the new tool.

closed 2025-02-08 by chshu