espressif/esp-zigbee-sdk #489: Request device endpoints

Daniel-Craciun, 2024-11-24:
### Question

Hello everyone,

I'm trying to create a Zigbee device (acting as a coordinator) that can connect to a Zigbee end device. But I was wondering is there a way to list all the endpoints on the connected device so that the coordinator device can know what kind of endpoints are on the connected device?

xieqinan, 2024-11-25:
The [HA customized devices](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L167) example demonstrates this functionality. Please refer to the example first.

Daniel-Craciun, 2024-12-11:
Thank you this example works well!

closed 2024-12-11 by Daniel-Craciun