espressif/esp-zigbee-sdk #490: Where to find full source of the library

hatch01, 2024-11-24:
### Question

I want to create a custom device which will follow my water consumption and send it through zigbee.
I think that it should be very similar to a temperature sensor in the code, with just some cluster ID to change.
That's why I try to start from the ZigbeeTempSensor, where the function `esp_zb_temperature_sensor_clusters_create` is used.
I found the header for this function here : [link](https://github.com/espressif/esp-zigbee-sdk/blob/608f7c0b1f57d825d0af0ce471e30d53d0d02b2a/components/esp-zigbee-lib/include/ha/esp_zigbee_ha_standard.h)
But I am unable to find the implementation of the function.
Is this available .

xieqinan, 2024-11-25:
Hi @hatch01 ,

The source code of the esp-zigbee-sdk is in a closed state. If you wish to obtain it, please contact our business team via [technical inquiries](https://www.espressif.com/en/contact-us/technical-inquiries) to request access to the source code.

hatch01, 2024-11-25:
Okay, I will ask for it.
Is it planned to make the esp-zigbee-sdk open later, or will this stay in this state?

chshu, 2024-12-20:
The current esp-zigbee-sdk is based in [ZBOSS](https://dsr-zoi.com/), per the agreement of ZOI, we can't make it opensource, but it's possible to give private access to customers under below rule:

> Free annual source code access passes (to a vendor branch) — customer receiving Free Annual Pass shall have a contractual commitment to a minimal shipment of 5M copies of product with ZOI software in it — like chipsets, and etc.

closed 2024-12-20 by chshu