espressif/esp-zigbee-sdk #491: Problem with Reporting Multistate Value

mahdi1-kh, 2024-11-25:
### Question

Description: I am attempting to create a Zigbee to UART dongle similar to the one described [here](https://ptvo.info/zigbee-configurable-firmware-features/uart/).
Based on my research and observing the PTVO firmware clusters on Z2M, I concluded that I should use the MultistateValue cluster. I have read the the ZCL documentation and [this GitHub issue](https://github.com/espressif/esp-zigbee-sdk/issues/372).

I have also tried making code changes based on [this example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices/customized_client).

However, after connecting the device to my Z2M network and pressing the key, no report is sent to Z2M.

I experimented with different modes of using `esp_zb_zcl_report_attr_cmd_req` and `esp_zb_zcl_set_attribute_val`. I tried these together and separately, but no report was sent.

Could you help identify the issue in my code or provide an example of using a multistate value?

Here is the code I tested:

### Additional context.

```/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee customized client Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */
#include "esp_HA_customized_switch.h"
#include "esp_check.h"
#include "esp_err.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "string.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "zcl/esp_zigbee_zcl_common.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light switch (End Device) source code.
#endif

static const char *TAG = "ESP_HA_ON_OFF_SWITCH";

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    uint16_t event_value = 121;

    switch (button_func_pair->func) {
    case SWITCH_ONOFF_TOGGLE_CONTROL: {
        
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_set_attribute_val(HA_ONOFF_SWITCH_ENDPOINT,
                                 ESP_ZB_ZCL_CLUSTER_ID_MULTI_VALUE, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                                 ESP_ZB_ZCL_ATTR_MULTI_VALUE_PRESENT_VALUE_ID, &event_value, false);
        esp_zb_lock_release();
        ESP_LOGI(TAG, "Set attribute");

        esp_zb_zcl_report_attr_cmd_t report_attr_cmd = {0};
        report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
        report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_MULTI_VALUE_PRESENT_VALUE_ID;
        report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
        report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_MULTI_VALUE;
        report_attr_cmd.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
[... 162 lines cut ...]

    esp_zb_zcl_update_reporting_info(&reporting_info);

    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    esp_zb_set_secondary_network_channel_set(ESP_ZB_SECONDARY_CHANNEL_MASK);

    esp_zb_ieee_addr_t my_addr = {0x32, 0x31, 0x19, 0x93, 0x21, 0xfB, 0xCD, 0xEF};
    esp_zb_set_long_address(my_addr);

    ESP_ERROR_CHECK(esp_zb_start(true));
    esp_zb_stack_main_loop();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}
```

xieqinan, 2024-11-26:
Hi,

I believe the issue is caused by the incorrect address mode: `report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;`. When using this mode, a binding relationship between the `Z2M` and the current device must exist in the binding table of the current device.  

Could you please try using the `ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT` mode instead? Assign the destination endpoint to `zcl_basic_cmd.dst_endpoint` and the destination address to `zcl_basic_cmd.dst_address`. Please refer to [here](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L57)

mahdi1-kh, 2024-11-26:
@xieqinan
Hi 

Thank you so much for your help! Your assistance resolved my issue.

I have another question regarding ESP_ZB_ZCL_ATTR_MULTI_VALUE_STATE_TEXT. Do you know when this feature will be added? If not, do you have any other suggestions that could substitute this feature? I need to solve this problem as soon as possible.

Thanks again for your help!

xieqinan, 2024-11-27:
@mahdi1-kh ,

The `ESP_ZB_ZCL_ATTR_MULTI_VALUE_STATE_TEXT` attribute is intended to describe all possible states of a multistate `PresentValue`. I believe this is not a necessary attribute. Could you please share how you plan to use it?

mahdi1-kh, 2024-11-28:
I was looking for a replacement and found [this document](https://github.com/espressif/esp-zigbee-sdk/blob/main/docs/en/user-guide/zcl_custom.rst). I tested the code provided 
there and it worked, and I was able to get reports in Z2M, somehow solving my problem. However, when I try to change the `attr_type`, I can't get any reports in Z2M anymore. Here is the the changes I have made in the code:

Custom Cluster:
```
esp_zb_attribute_list_t *custom_cluster = esp_zb_zcl_attr_list_create(CUSTOM_CLUSTER_ID);
uint16_t custom_attr1_id = 0x01;
uint8_t custom_attr1_value[4] = {70,80,90,100};
uint16_t custom_attr2_id = 0x0000;
uint16_t custom_attr2_value = 0x1234;
esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr1_id, ESP_ZB_ZCL_ATTR_TYPE_ARRAY,
                                      ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, custom_attr1_value);
esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr2_id, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
                                      &custom_attr2_value);
```
Custom Command 
```
 esp_zb_zcl_custom_cluster_cmd_req_t req;
        uint8_t custom_value[5] = {20,30,40,50,60};
        req.zcl_basic_cmd.dst_addr_u.addr_short = 0x000; // Set it according to the actual destination short address.
        req.zcl_basic_cmd.dst_endpoint = 1;
        req.zcl_basic_cmd.src_endpoint = 1;
        req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        req.cluster_id = CUSTOM_CLUSTER_ID;
        req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
        req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
        req.custom_cmd_id = 0x01;
        req.data.type = ESP_ZB_ZCL_ATTR_TYPE_ARRAY;
        req.data.size = (sizeof(custom_value)+2);
        req.data.value = custom_value;
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_custom_cluster_cmd_req(&req);
        esp_zb_lock_release();
        ESP_EARLY_LOGI(TAG, "Send %d to custom device", custom_value);
```

xieqinan, 2024-11-29:
@mahdi1-kh 

> And that's how I know that I should use the stateTextattribute.

The attribute will still not be supported in the upcoming release. I would like to discuss the solution you provided.

> However, when I try to change the attr_type, I can't get any reports in Z2M anymore. Here is the the changes I have made in the code:

The first two bytes indicate the length of the array; refer to [here](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L281).
 
```
esp_zb_attribute_list_t *custom_cluster = esp_zb_zcl_attr_list_create(CUSTOM_CLUSTER_ID);
uint16_t custom_attr1_id = 0x01;
uint8_t custom_attr1_value[6] = {0, 0, 70,80,90,100};
uint16_t custom_attr2_id = 0x0000;
uint16_t custom_attr2_value = 0x1234;

uint16_t *attr1_array_len = custom_attr1_value;
*attr1_array_len = sizeof(custom_attr1_value) - 2;

esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr1_id, ESP_ZB_ZCL_ATTR_TYPE_ARRAY,
                                      ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, custom_attr1_value);
esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr2_id, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
                                      &custom_attr2_value);
```

```                              
        esp_zb_zcl_custom_cluster_cmd_req_t req;
        uint8_t custom_value[7] = {0, 0, 20,30,40,50,60};
        
        uint16_t *value_array_len = custom_value;
        *value_array_len = sizeof(custom_value) - 2;
        
        req.zcl_basic_cmd.dst_addr_u.addr_short = 0x000; // Set it according to the actual destination short address.
        req.zcl_basic_cmd.dst_endpoint = 1;
        req.zcl_basic_cmd.src_endpoint = 1;
        req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
        req.cluster_id = CUSTOM_CLUSTER_ID;
        req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
        req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
        req.custom_cmd_id = 0x01;
        req.data.type = ESP_ZB_ZCL_ATTR_TYPE_ARRAY;
        req.data.size = (sizeof(custom_value));
        req.data.value = custom_value;
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_custom_cluster_cmd_req(&req);
        esp_zb_lock_release();
        ESP_EARLY_LOGI(TAG, "Send %d to custom device", custom_value);
```

mahdi1-kh, 2024-11-30:
Thank you very much for your guidance. This code worked correctly, but I realized that the `esp_zb_zcl_custom_cluster_cmd_req `function causes a panic with sizes greater than 120. In my search, I encountered [this guidance](https://github.com/espressif/esp-zigbee-sdk/issues/202#issuecomment-2002767263) from you. I used the `esp_zb_zcl_write_attr_cmd_req` function, and it worked fine up to a data size of 154. From 155 onwards, it did not report back. I wanted to know if this limitation is from z2m, if it's a problem with my code, or if it is a limitation of the SDK

xieqinan, 2024-12-02:
Hi @mahdi1-kh ,

> I used the esp_zb_zcl_write_attr_cmd_req function, and it worked fine up to a data size of 154. From 155 onwards, I got no reports in Z2M.

This issue is similar with the [link](https://github.com/espressif/esp-zigbee-sdk/issues/423#issuecomment-2362587608), I think you can follow it and check whether your issue also can be resolved.

mahdi1-kh, 2024-12-02:
hi @xieqinan,

The way I understood [the issue](https://github.com/espressif/esp-zigbee-sdk/issues/423#issuecomment-2362587608) you mentioned is that the length of the list should be stored in the first byte of the first list. I did the same thing as the code you modified [here](https://github.com/espressif/esp-zigbee-sdk/issues/491#issuecomment-2507161986)

> uint16_t *attr1_array_len = custom_attr1_value;
> *attr1_array_len = sizeof(custom_attr1_value) - 2;

 and increased the size in the initial definition but it didn't change and I can still see the reports in z2m up to size 154. This is the code I modified:
 
 ```
 esp_zb_attribute_list_t *custom_cluster = esp_zb_zcl_attr_list_create(CUSTOM_CLUSTER_ID);
uint16_t custom_attr1_id = 0x01;
uint8_t custom_attr1_value[250] ;
uint16_t custom_attr2_id = 0x0000;
uint16_t custom_attr2_value = 0x1234;

for (uint16_t i = 0; i < 250; i++) { 
            custom_attr1_value[i+2] = i + 1;            
        
         }

uint16_t *attr1_array_len = custom_attr1_value;
*attr1_array_len = sizeof(custom_attr1_value) - 2;
for (size_t i = 0; i < 5; i++) { printf("custom_value[%zu] = %u\n", i, custom_attr1_value[i]); }

esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr1_id, ESP_ZB_ZCL_ATTR_TYPE_ARRAY,
                                      ESP_ZB_ZCL_ATTR_ACCESS_WRITE_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, custom_attr1_value);
esp_zb_custom_cluster_add_custom_attr(custom_cluster, custom_attr2_id, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
                                      &custom_attr2_value);
 ```

```
esp_zb_zcl_custom_cluster_cmd_req_t req;
        uint8_t custom_value[156] ;

        for (uint16_t i = 0; i < 154; i++) { custom_value[i+2] = i + 1;}    

        uint16_t *value_array_len = custom_value;
        *value_array_len = sizeof(custom_value) - 2;
        for (size_t i = 0; i < 5; i++) { printf("custom_value[%zu] = %u\n", i, custom_value[i]); }

esp_zb_zcl_attribute_t attr_field1[] = 
{
   {0x01, {ESP_ZB_ZCL_ATTR_TYPE_ARRAY, (sizeof(custom_value)), custom_value}}

};

esp_zb_zcl_write_attr_cmd_t test_req = {
                        .zcl_basic_cmd = {
                            .dst_addr_u.addr_short = 0x000,
                            .dst_endpoint = 1,
                            .src_endpoint = 1,
                        },
                        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
                        .clusterID = CUSTOM_CLUSTER_ID,
                        .attr_number = 1,
                        . attr_field = attr_field1,
                    };

esp_zb_lock_acquire(portMAX_DELAY);
esp_zb_zcl_write_attr_cmd_req(&test_req);
esp_zb_lock_release();
```
Although 150 meets my needs, I'm just curious to understand the reason behind this issue.

xieqinan, 2024-12-10:
>  but it didn't resolve the issue and I still don't get reports in Z2M for messages with a size bigger than 154.

The maximum fragment length depends on the destination device. If the `Max Incoming Transfer Size` of the destination device is smaller than 154, a `write attribute request` exceeding this size might still be sent. Could you please verify this behavior? It would be helpful if you could share the .pcap file captured by the sniffer for further analysis.
[image]

mahdi1-kh, 2025-01-04:
First of all, sorry for the delayed response, I've been really busy. And thank you for your support.

I have set up Wireshark and a sniffer and I can successfully capture Zigbee packets. But unfortunately I can't find `Node Descriptor` or `Max Incoming Transfer Size`. Here is what I see in Wireshark:

[image]

And here is the `.pcap` file (I had to zip it, github doesn't allow uploading .pcap):

[ESP32.zip](https://github.com/user-attachments/files/18307531/ESP32.zip)

closed 2025-02-11 by mahdi1-kh