espressif/esp-zigbee-sdk #492: Espressif Zigbee light verify fail in scene store when adding light in a market based Zigbee wireless (12-scene) switch.

mundevx, 2024-11-26:
### Question

Hi,

I am using a market based zigbee scene switch. When i added my espressif zigbee light in scene switch using app, then it's shows verify fail. But if i use market based zigbee light, and add to scene switch using app, then its shows verify ok. What could be the reason?

mundevx, 2024-11-27:
@xieqinan 

I build 4 light, 4 dimmer light, 1 thermostat and 1 color light using espressif zigbee sdk library. They are working as expected. I used Grouo and scene clusters also in these devices. I purchased a zigbee wireless scene switch from market, and i added all my zigbee 4 lights, 4 dimmer,  1 color light and 1 thermostat to scene switch using app. But app shows all the zigbee devices verify failed. But if i use others brand light and add it to scene switch, then it verify ok in scene switch and work as expected. But will the issue.

xieqinan, 2024-11-29:
@mundevx ,

> But app shows all the zigbee devices verify failed. 

Have the ESP lights joined the same Zigbee network as the `zigbee wireless scene switch`? If they have, do you know what is verification required for other devices to connect with the `zigbee wireless scene switch`? If you know verification process, could you please share which step failed during verification?

mundevx, 2024-11-30:
Yes, they are on same network.   

On my zigbee device-A, i got this callback when a scene switch is trying to add this device-A through smart life app.

esp_err_t zb_get_scene_store_resp_handler(const esp_zb_zcl_store_scene_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);

    return ESP_OK;
}

and after adding, when scene is recall from scene switch, my device-A is getting a callback, 

esp_err_t zb_get_scene_recall_resp_handler(const esp_zb_zcl_recall_scene_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);

    esp_zb_zcl_scenes_extension_field_t* ext_list =   message->field_set;
    //Here i am getting extension list NULL
    while (ext_list != NULL) {
		printf("NOT NULL\n");
		ext_list = ext_list->next;  // Move to the next node
	}
    return ESP_OK;
}
Then i am getting ext_list NULL. 

I also added scene and group clusters in my device-A.

Second Task i did that, i inserted a function "nuos_zb_scenes_store_scene_request" wjhen i get callback of zb_get_scene_store_resp_handler on my device-A when scene switch trying to add it. But this will give message in smartlife when adding scene that Verification fail mesaage while adding device to scene switch. 

esp_err_t zb_get_scene_store_resp_handler(const esp_zb_zcl_store_scene_message_t *message)
{
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);

    nuos_zb_scenes_store_scene_request(message->group_id, message->scene_id, message->info.dst_endpoint, message->info.dst_endpoint, 0);
    return ESP_OK;
}   
Then what happened, i am receiving callback zb_attribute_handler instead of zb_get_scene_store_resp_handler,which is our requirement.

static esp_err_t zb_attribute_handler(const esp_zb_zcl_set_attr_value_message_t *message)
{
    esp_err_t ret = ESP_OK;
    ESP_RETURN_ON_FALSE(message, ESP_FAIL, TAG, "Empty message");
    ESP_RETURN_ON_FALSE(message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS, ESP_ERR_INVALID_ARG, TAG, "Received message: error status(%d)",
                        message->info.status);

    
    return ret;
}

While adding device-A in scene created by another device scene switch through smart life app, i found following issues.
1. Smart Life app shows verification success when sending no response back to scene switch on callback function.
2. On Recall scene by scene switch, extension filed is NULL. (No info on zigbee device attributes value).

1. Smart Life app shows verification fail when sending response back to scene switch on callback function.
2. On Recall scene by scene switch, zb_attribute_handler is called, but verification is failed on smart life app by scene switch while adding device-A in scene.
3. then scene switch is driving device-A successfully.

Question is that, Why i am getting extension filed NULL after successfull verification(storing scene) by scene switch.

I hope you can understand.

xieqinan, 2024-12-10:
@mundevx ,

The https://github.com/espressif/esp-zigbee-sdk/issues/492#issuecomment-2508860656 is helpful. In the esp-zigbee-sdk, scenes are not automatically stored in the `ESP_ZB_CORE_SCENES_STORE_SCENE_CB_ID` callback. You need to manually call `esp_zb_zcl_scenes_table_store()` to store the scene. Once stored, the `ESP_ZB_CORE_SCENES_RECALL_SCENE_CB_ID` callback will retrieve the related scene from the `esp_zb_zcl_scenes_extension_field_t`.

mundevx, 2024-12-11:
Hi @xieqinan ,

Yes, but getting crash on calling esp_zb_zcl_scenes_table_store() inside callback esp_err_t zb_get_scene_store_resp_handler(const esp_zb_zcl_store_scene_message_t *message){ }
I am passing  message->info.dst_endpoint, message->group_id, message->scene_id, message->info.cluster in  esp_zb_zcl_scenes_table_store().

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4001837a  RA      : 0x420118ac  SP      : 0x4081ab40  GP      : 0x4080e3f0
0x4001837a: memcpy in ROM
0x420118ac: esp_zb_zcl_scenes_table_store at ??:?

TP      : 0x408072bc  T0      : 0x4000af8c  T1      : 0xffffffe0  T2      : 0xcf0c8861
0x408072bc: vRingbufferReturnItem at C:/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1219
0x4000af8c: multi_heap_internal_unlock in ROM

S0/FP   : 0x40810536  S1      : 0x40810188  A0      : 0x408207c8  A1      : 0xa1b72c3b
A2      : 0x00000016  A3      : 0x00000001  A4      : 0x408207de  A5      : 0x408207c8
A6      : 0x000000ff  A7      : 0x0000ffff  S2      : 0x40821410  S3      : 0x408211e0
S4      : 0x00000000  S5      : 0x4081b11c  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x0000002b  S9      : 0x00000000  S10     : 0x00000005  S11     : 0x00000007
T3      : 0x000000ff  T4      : 0x4081f434  T5      : 0x0000000b  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0xa1b72c3b
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4081ab40: 0x82cd6cf1 0x00000000 0x4080dc30 0x00000016 0x00000000 0x4081b11c 0x00000001 0x40815000
4081ab60: 0x4081abd0 0x00000000 0x4080ff28 0x4200bb48 0x82cdfe01 0x00020005 0x4081ab70 0x40810536
0x4200bb48: zb_get_scene_store_resp_handler at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/nuos_zigbee_scene_commands.c:325

4081ab80: 0x0000000e 0x4081abe8 0x00000001 0x4081ac4f 0x4081abd0 0x00000001 0xffffffff 0x4200bd7c
0x4200bd7c: nuos_zb_scene_action_handler at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/nuos_zigbee_scene_commands.c:542

4081aba0: 0x1b391a88 0x4bae8e41 0xaffe5dcf 0x42009286 0x00000043 0x00000000 0x00000043 0x42007ab6
0x42009286: nuos_zigbee_action_handler at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/app_zigbee_clusters.c:1102
0x42007ab6: zb_action_handler at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/esp_zb_controller_main.c:427

4081abc0: 0x4081aca4 0x4081ac9c 0x00000005 0x42011388 0x00000000 0x00050001 0x00000004 0x42430043
0x42011388: zcl_scenes_store_scene_handler at esp_zigbee_zcl_scenes.c.obj:?

4081abe0: 0x00000000 0x4203782a 0x7fffffff 0x00000001 0x42011290 0x4081b11c 0x00000005 0x4200e1c4
0x4203782a: zb_zcl_process_device_command at ??:?
0x42011290: zcl_scenes_store_scene_handler at esp_zigbee_zcl_scenes.c.obj:?
0x4200e1c4: zb_zcl_device_callback at esp_zigbee_core.c.obj:?

4081ac00: 0x4200e600 0x42037008 0x4081ac80 0x4080e3f0 0x408072bc 0x400184be 0x0000000f 0xcf0c8861
0x4200e600: zb_zcl_endpoint_handler_callback at esp_zigbee_core.c.obj:?
0x42037008: ep_process_zcl_cmd at zcl_common.c.obj:?
0x408072bc: vRingbufferReturnItem at C:/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1219
0x400184be: memset in ROM

4081ac20: 0x00000005 0x4203782a 0x7fffffff 0x00000001 0x4081ac9c 0x4081aca4 0x00000005 0x420425c0
0x4203782a: zb_zcl_process_device_command at ??:?
0x420425c0: zb_zcl_scenes_process_store_scene at ??:?

4081ac40: 0x00000006 0x00000006 0x00000001 0x000000ff 0x7fffffff 0x4203782a 0x00000001 0x000000ff
0x4203782a: zb_zcl_process_device_command at ??:?

4081ac60: 0x4081b116 0x40821638 0x00000005 0x42042b56 0x0000005a 0x00000005 0x00000005 0x4200ec26
0x42042b56: zb_zcl_process_scenes_specific_commands at ??:?
0x4200ec26: zb_zcl_endpoint_handler_callback at esp_zigbee_core.c.obj:?

4081ac80: 0x00000008 0x4081ad08 0x00000000 0x4081ace8 0x00000000 0x4081ace8 0x00000008 0x42430043
4081aca0: 0x408102e8 0x00000000 0x00000000 0x01049900 0x05e24001 0x04010400 0x01002200 0x00000000
4081acc0: 0x00000005 0x4081b138 0x00000005 0x000000ff 0x00000005 0x4081b138 0x00000000 0x4203c160
0x4203c160: zb_zcl_handle at ??:?

4081ace0: 0x00000001 0x00000005 0x4082112c 0x42036ff6 0x0000000f 0x00000005 0x00000001 0x4081b138
0x42036ff6: ep_process_zcl_cmd at zcl_common.c.obj:?

4081ad00: 0x00000005 0x00000000 0x00000000 0x01049900 0x05e24001 0x04010400 0x01002200 0x42000000
4081ad20: 0x0000000f 0x00000000 0x00000005 0x4203789a 0x00000000 0x00000000 0x00000000 0x01049900
0x4203789a: zb_zcl_process_device_command at ??:?

4081ad40: 0x05e24001 0x04010400 0x01002200 0x42000000 0x0000000f 0x00000000 0x408102e8 0x42020256
0x42020256: zb_sched_loop_iteration at ??:?

4081ad60: 0x00000000 0x00000001 0x00000300 0x00000000 0x4207345a 0x420731d4 0x00000000 0x00000000
0x4207345a: event_read at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:284
0x420731d4: event_start_select at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:99

4081ad80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081ada0: 0x00000000 0x00000000 0x00000000 0x4200ef16 0x00000000 0x00000000 0x00000000 0x4200765c
0x4200ef16: esp_zb_main_loop_iteration at ??:?
0x4200765c: esp_zb_task at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/esp_zb_controller_main.c:494 (discriminator 2)

4081adc0: 0x00000001 0x00000000 0x0000000a 0x00000000 0x00000000 0x00000000 0x00000000 0x40809ef0
0x40809ef0: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

xieqinan, 2024-12-12:
I would like to know the implementation of `zb_get_scene_store_resp_handler()` in your code.

mundevx, 2024-12-12:
@xieqinan ,

I have a wireless scene switch, and i am coding for espressif zigbee device to interact with that device. Coordinator is some manufacturer zigbee gateway. 
when ever i add my espressif zigbee device to scene switch via app through gateway, then app send the scene store request to my zigbee device. As per in code, this request is received by scene store callback.

xieqinan, 2024-12-12:
Sorry for the misunderstanding. I suspect there may be a mistake in the implementation of the `zb_get_scene_store_resp_handler()` function causing the assert mentioned in  https://github.com/espressif/esp-zigbee-sdk/issues/492#issuecomment-2533712060 . I would like to review it. Could you please share the code details with me?

xieqinan, 2025-09-30:
Hi,

Do you have any updates about this issue? Please let's me know your status.