espressif/esp-zigbee-sdk #493: Request for Future Support of the Discover Attributes Extended Command

renatomotorline, 2024-11-27:
### Question

I’d like to inquire if there are any plans to add support for the [Discover Attributes Extended Command](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf#page=92) in the future.

luar123, 2026-10-31:
@xieqinan this is still missing in 2.0.5. Any chance to add it soon?

xieqinan, 2026-11-04:
Hi @luar123  @renatomotorline,

This feature is supported starting from [v2.1.0](https://github.com/espressif/esp-zigbee-sdk/commit/16fb735e12014a80173dd2c4b32f5bbc5322fbf2#diff-68d24fa81558aae3d8c59e2aa57a4fa719ea3b04d7fa14beff45f16f00858f50). Please update to this version and test again.

Let's close this issue for now. any questions can reopen it.

closed 2026-11-04 by xieqinan