espressif/esp-zigbee-sdk #494: Can zigbee support devices to join network using install code and global link key dynamically

syam-ari, 2024-11-27:
### Question

Can Gateway support devices to join network using install code and Global linke key. In Example code, Install code policy is compile time constant. How to allow devices to join network with and without install code dynamically?

#define ESP_ZB_ZC_CONFIG()                                                  \
    {                                                                       \
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,                      \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,                   \
        .nwk_cfg.zczr_cfg = {                                               \
            .max_children = MAX_CHILDREN,                                   \
        },                                                                  \
    }

xieqinan, 2024-11-29:
Hi @syam-ari ,

We will provide an API to enable or disable the install code to meet your requirements. In the meantime, you can test it by adding `extern void zb_set_installcode_policy(bool allow_ic_only);`. If you set `zb_set_installcode_policy(true)`, the coordinator will only allow devices to join using an install code. However, please note that this temporary API will no longer be supported once the public API is available.

closed 2024-12-06 by espressif-bot

xieqinan, 2024-12-06:
Hi, 

The [esp_zb_secur_ic_only_enable()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_secur.h#L34) provide a better way to support this issue. Please use the new API to test it again.